MichaIng/DietPi #334: DietPi-Software | Rethink user data directories

Fourdee, 2016-05-13:
Just brainstorming.

Currently its:
```
/mnt/usb_1 #dedicated usb install
/root #SD card only
```

We need to change where we store files for ```/root```. A static global location in /mnt would be ideal.
eg:
```
/mnt/DietPi-UserData
```
Then we can use symlinks to point usb drive.
Using a single location will make installation code simpler, and the online documentation. eg: where is my files ? ```/mnt/DietPi-UserData```

k-plan, 2016-05-13:
@Fourdee 

a go from my side, if allow for me to say this,
> Using a single location will make installation code simpler,

:100: 

Idea;
- if done, a option via menu to move the file to usb drive or a usb drive partitoin (eg. sda2) and point back to `/mnt/DietPi-UserData` 
(for running system, not only clean installations)

Future;
- support for UUID
-  moving rootfs from sd-card to usb devices, like armbian  `nand-sata-install.sh`  
(for running system, not only clean installations)

xenfomation, 2016-05-13:
I agree and say, sorry for the delay on the disk by uuid as I am finalizing code for dual USB mounts. 

Thanks for your patience.  Crazy week and every inch I can get into DietPi had been a warm reprieve.

Fourdee, 2016-05-30:
Ok, so the new Userdata location is ```/mnt/dietpi_userdata```.

If the user has a USB drive, a symlink will be created. All data that exists in ```/mnt/dietpi_userdata``` will be copied to USB drive beforehand:
```
                        USERDATA_DIRECTORY="/mnt/dietpi_userdata"
			# - Copy any existing data to usb drive, if it is not a symlink already.
			if [ -d "$USERDATA_DIRECTORY"] && [ ! -L "$USERDATA_DIRECTORY" ]; then

				# - stop all running services. As this is done in apply_configs sections, its probably unessacary, but lets be safe.
				/DietPi/dietpi/dietpi-services stop
				cp -R "$USERDATA_DIRECTORY"/* /mnt/usb_1/

			fi

			# - Create symlink
			rm -R "$USERDATA_DIRECTORY"
			ln -sf /mnt/usb_1 "$USERDATA_DIRECTORY"

```

If user has a custom directory set in dietpi.txt (or via menu), that will be used instead:
```
		local dietpi_userdata_basedirectory=$(cat /DietPi/dietpi.txt | grep -m1 'dietpi_userdata_basedirectory=' | sed 's/.*=//')
		if [ "$dietpi_userdata_basedirectory" != "Auto" ]; then
			USERDATA_DIRECTORY="$dietpi_userdata_basedirectory"

```

Just need to test it

Fourdee, 2016-05-30:
Tests fine.

> note to self: When moving data from flash to usb drive, add a check to ensure all data is copied over before removing the folder and creating the symlink.

Fourdee, 2016-05-31:
Completed.

closed 2016-05-31 by Fourdee