MichaIng/DietPi #335: DietPi-Software | Add Calibre2opds installation option

arv187, 2016-05-13:
Calibre2opds is an open source tool that can generate OPDS and HTML based online catalogs from the metadata database of the Calibre eBook library manager.

Fourdee, 2016-05-13:
@arv187 

My head = o

> Calibre2opds is an open source tool that can generate OPDS and HTML based online catalogs from the metadata database of the Calibre eBook library manager.

Went over my head /'o'\  lol

Regardless, thanks for the software request and we will look into it when possible :+1:

arv187, 2016-06-19:
Update with this info http://blog.slucas.fr/en/oss/calibre-opds-php-server

loro102, 2016-10-28:
Use calibre server here the info

http://www.linux-magazine.com/Online/Blogs/Productivity-Sauce/Turn-Raspberry-Pi-into-an-Ebook-Server-with-Calibre

Invictaz, 2017-04-07:
@arv187 Why would you need a ODPS server? It serves the metadata yes but what does it more than FTP or HTTP?

arv187, 2017-04-07:
Info about autor, genre, sinopsis, image, search function, a catalog.
An Example:
https://lh3.googleusercontent.com/_lG58t3XWd3c/TcN0Bpytj3I/AAAAAAAAAEw/VAfVlHr8L3U/s800/cs_list.jpeg

Invictaz, 2017-04-08:
@arv187 

But most e-readers with wifi option don't have the high specced webbrowser to view this?
At least last time when I setup my ODPS in Calibre and browsed to it from my e-reader it looked like a mobile site, a very barebones look.

arv187, 2017-08-01:
https://opensource.com/article/17/6/raspberrypi-ebook-server

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

Atavic, 2018-09-23:
- Software title | cops
- Short description | web-based light alternative to Calibre content server
- GitHub URL (if available) | https://github.com/seblucas/cops
- External URL (if available) | http://blog.slucas.fr/en/oss/calibre-opds-php-server
- (Official) install documentation | https://github.com/seblucas/cops#install-a-release-easiest-way

https://github.com/seblucas/cops/issues/398