MichaIng/DietPi #337: Wifi cant connect: Edimax EW-7811UTC (AC600) Realtek RTL8812au

arv187, 2016-05-13:
https://www.raspberrypi.org/forums/viewtopic.php?p=849631#p849631

Driver not is detected by dietpi, then I install following the above steps:
```
root@DietPi:~# uname -a
Linux DietPi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux
root@DietPi:~# sudo wget https://dl.dropboxusercontent.com/u/80256631/8812au-4.1.19-v7-858.tar.gz
--2016-05-13 18:27:30--  https://dl.dropboxusercontent.com/u/80256631/8812au-4.1.19-v7-858.tar.gz
Resolving dl.dropboxusercontent.com (dl.dropboxusercontent.com)... 108.160.173.69
Connecting to dl.dropboxusercontent.com (dl.dropboxusercontent.com)|108.160.173.69|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 718104 (701K) [application/octet-stream]
Saving to: ‘8812au-4.1.19-v7-858.tar.gz’

8812au-4.1.19-v7-85 100%[=====================>] 701.27K  1.10MB/s   in 0.6s

2016-05-13 18:27:32 (1.10 MB/s) - ‘8812au-4.1.19-v7-858.tar.gz’ saved [718104/718104]

root@DietPi:~# sudo tar xzf 8812au-4.1.19-v7-858.tar.gz
root@DietPi:~# ./install.sh
sudo cp 8812au.conf /etc/modprobe.d/.
sudo install -p -m 644 8812au.ko /lib/modules/4.1.19-v7+/kernel/drivers/net/wireless
sudo depmod 4.1.19-v7+

Reboot to run the driver.

If you have already configured your wifi it should start up and connect to your
wireless network.

If you have not configured your wifi you will need to do that to enable the wifi.

Bus 001 Device 005: ID 7392:a812 Edimax Technology Co., Ltd
[    2.257287] usb 1-1.4: new high-speed USB device number 5 using dwc_otg
[    2.365958] usb 1-1.4: New USB device found, idVendor=7392, idProduct=a812
[    2.369953] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.369964] usb 1-1.4: Product: Edimax AC600 USB
[    2.369977] usb 1-1.4: Manufacturer: Realtek
[    2.369985] usb 1-1.4: SerialNumber: 00e04c000001
[    5.275050] usbcore: registered new interface driver rtl8821au
```
Scan & connect:

```
Connecting to ESM_EF2000_T3B
 , please wait

 DietPi-Services
 Mode: stop
 Please wait...
 Completed

 - Dropping connections..

 - Restarting networking services...
ifup: interface eth0 already configured
run-parts: /etc/network/if-pre-up.d/wireless-tools exited with return code 2
Failed to bring up wlan0.

 DietPi-Services
 Mode: start
 Please wait...
 Completed

 DietPi-Process Tool
 Applying settings, please wait...

Cron (26239)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

DHCP Client (26811)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

MiniDLNA (26755)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

NetData (26930)
Ok | Nice      19
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
NetData (26933)
Ok | Nice      19
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
NetData (26941)
Ok | Nice      19
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
NetData (26944)
Ok | Nice      19
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
NetData (26945)
Failed | Nice      19
Failed | Affinity  0-3
Failed | Scheduler SCHED_OTHER 0

OpenSSH Server (409)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
OpenSSH Server (3026)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

ProFTP (26249)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Webserver Nginx (26258)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Webserver Nginx (26259)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Webserver Nginx (26260)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Webserver Nginx (26261)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Webserver Nginx (26262)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Webserver MySql (26308)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Webserver MySql (26656)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Webserver Php-fpm (26272)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Webserver Php-fpm (26275)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Webserver Php-fpm (26276)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Webserver Php-fpm (26277)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Webserver Php-fpm (26278)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

 Completed

```

Fourdee, 2016-05-13:
@arv187 
I've got a note to purchase a AC600 so I can test 5ghz band.

Can you provide the kernel version, and, have you tried updating the RPi firmware?
```
uname -a
```

Install rpi-update. Either via ```dietpi-config``` advanced, or :
```
apt-get install -y rpi-update
rpi-update
```

arv187, 2016-05-13:
I updated firmware and I downloaded the driver corresponding to new firmware but i have same error:
```
- Dropping connections..
 
 - Restarting networking services...
ifup: interface eth0 already configured
run-parts: /etc/network/if-pre-up.d/wireless-tools exited with return code 2
Failed to bring up wlan0.
```
http://pastebin.com/gw3rvtLX

I only update firmware and reinstall wifi driver to 4.4.10-v7+ #885

k-plan, 2016-05-14:
@arv187 

have a look in all your posted output.
In your second installation you have loaded and installed the wrong driver:

>
```
uname -a
Linux DietPi 4.4.10-v7+ #885 SMP Fri May 13 15:44:27 BST 2016 armv7l GNU/Linux
```
```
root@DietPi:~# sudo wget https://dl.dropboxusercontent.com/u/80256631/8812au-4.1
.19-v7-858.tar.gz
```

If I follow your [posted link](https://www.raspberrypi.org/forums/viewtopic.php?p=849631#p849631) and read it correct, it should have to be:

https://dl.dropboxusercontent.com/u/80256631/8812au-4.4.10-v7-885.tar.gz

> But in var/run not there wpa_supplicant folder

It's okay.Your WiFi credentials will be set via "dietpi-config" menu.

Have tested it for you on my RPi with : 

```
──────────────
 DietPi     | 18:29 | Sat 14/05/16
──────────────
 V117       | Raspberry Pi B+
──────────────

```
```
root@RPi-B-Plus:~# lsusb
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
root@RPi-B-Plus:~# lsmod
Module                  Size  Used by
evdev                  10534  0
sg                     20575  0
bcm2835_gpiomem         3703  0
bcm2835_rng             2207  0
uio_pdrv_genirq         3526  0
uio                    10078  1 uio_pdrv_genirq
fuse                   86348  3
ipv6                  358811  12
```

```
root@RPi-B-Plus:~# uname -a
Linux RPi-B-Plus 4.1.19+ #851 Sat Mar 5 22:26:57 GMT 2016 armv6l GNU/Linux
```

```
root@RPi-B-Plus:~# wget https://dl.dropboxusercontent.com/u/80256631/8812au-4.1.19-851.tar.gz

root@RPi-B-Plus:~# tar xzf 8812au-4.1.19-851.tar.gz

root@RPi-B-Plus:~# ./install.sh

root@RPi-B-Plus:~# poweroff
```
- Plug in your USB WiFi adapter, power on device and wait until it have completely boot up, to login.

```
root@RPi-B-Plus:~# lsusb
Bus 001 Device 004: ID 0bda:8812 Realtek Semiconductor Corp.
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
root@RPi-B-Plus:~# lsmod
Module                  Size  Used by
8812au               1759630  0
evdev                  10534  0
sg                     20575  0
cfg80211              479279  1 8812au
rfkill                 21508  1 cfg80211
bcm2835_gpiomem         3703  0
bcm2835_rng             2207  0
uio_pdrv_genirq         3526  0
uio                    10078  1 uio_pdrv_genirq
fuse                   86348  3
ipv6                  358811  12
```

`root@RPi-B-Plus:~# dietpi-config`

- enable WiFi adapter
[image]

- It will happen, that the first `Restarting networking services ...` fail, because the default WiFi credentials don't fit to your network. But it is no problem, be patient!

- Set up your Wireless Network Options via menu (Change Mode:  STATIC or DHCP) and Scan your WLAN.
- Choose wanted AP in the list and "Enter the Access Key (WPA/WEP)."
[image]
- After automatically restarting networking and services by dietpi-config, you should be connected.
[image]
- Test your internet connection and your ip addresses `root@RPi-B-Plus:~# ip a`

k-plan, 2016-05-14:
@Fourdee 

Note: no such fiddling around with Odroid C1. Driver for Realtek 8812au in here:

```
root@oDroid-C1-TEST:~# cat /sys/module/8812au/version
v4.2.5_10143.20140103_ASUS

root@oDroid-C1-TEST:~# uname -a
Linux oDroid-C1-TEST 3.10.80 #1 SMP PREEMPT Sun Mar 13 15:50:53 CET 2016 armv7l GNU/Linux
```

But can't find it on C2 ...

arv187, 2016-05-14:
You are right respect my wrong download, but one minute after (line n15) I installed the right versión. After try again and comment here.

k-plan, 2016-05-14:
@arv187 

hmmm ... line nr. 31

> root@DietPi:~# sudo tar xzf 8812au-4.1.19-v7-858.tar.gz

----

> Petición HTTP enviada, esperando respuesta... 200 OK

:es: ?   :smiley:

arv187, 2016-05-14:
Yes, im from spain xD.
Still not working, in test connection its ok because eth0 is plugin, if not wifi dongle power led on, but connect.
 - Restarting networking services...
ifup: interface eth0 already configured
run-parts: /etc/network/if-pre-up.d/wireless-tools exited with return code 2
Failed to bring up wlan0.

log: [http://pastebin.com/6w6SrfNL](url)

k-plan, 2016-05-14:
@arv187 

- You are on DietPi v117?
- Can you please post `root@RPi-B-Plus:~# cat /etc/network/interfaces` ?
- Your access point is  visible on the scan list ?
- Access Point is on 2.4 or 5 GHz? What channel no.?
- `Max Usb Current  : Enabled` ?

> ```
root@DietPi:~# iwlist wlan0 channel
wlan0     17 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Current Frequency:2.412 GHz (Channel 1)
```

- Output in `dmesg` with `cfg80211`, ` usbcore`, ` RTL871X` ?

arv187, 2016-05-14:
I reinstalled image in sd card, because not boot, i record this video in slowmotion [https://drive.google.com/open?id=0B8EMglGxQawZbWNZUTNyWTRIbWs](url)
Not is the first time.

Well, @k-plan @Fourdee here are the data:

**1 You are on DietPi v117?
   2 Can you please post root@RPi-B-Plus:~# cat /etc/network/interfaces ?
   3 Your access point is visible on the scan list ?
   4 Access Point is on 2.4 or 5 GHz? What channel no.?
   5 Max Usb Current : Enabled ?
   6 Output in dmesg with cfg80211, usbcore, RTL871X ?
    7 Have you manually edit config files, to get WLAN on work?
    8 Have install packets via apt-get, to become WLAN on work?**

**1.**
 ───────────────────────────────────────
 DietPi     | 23:21 | Sat 14/05/16
 ───────────────────────────────────────
 V117       | Raspberry Pi 2 Model B

**2.**

```
root@DietPi:~# cat /etc/network/interfaces
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet dhcp
metric 0
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 8.8.8.8 8.8.4.4

# Wifi
allow-hotplug wlan0
iface wlan0 inet dhcp
metric 1
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
wireless-essid ESM_EF2000_T3B
wireless-key xxxxx
wireless-mode Managed
wireless-power off
wpa-ssid ESM_EF2000_T3B
wpa-psk xxxxx
#dns-nameservers 8.8.8.8 8.8.4.4
```

**3.**
Yes.

**4.**

5Ghz, my router have configured channel nº 36

```
root@DietPi:~# iwlist wlan0 channel
wlan0     17 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Current Frequency:2.412 GHz (Channel 1)
```

**5.**
Yes.
          ┌──────────────┤ DietPi - Advanced Options ├───────────────┐
          │ Choose an option                                         │
          │                                                          │
          │             1 Swapfile Size    : 100 MB                  │
          │             2 Time sync (ntpd) : Boot + Daily            │
          │             3 Max Usb Current  : Enabled                 │
          │             4 Update Firmware                            │
          │             5 RPI i2c          : Disabled                │
          │             6 RPI i2c freq     : 100 kHz                 │
          │                                                          │
          │                                                          │
          │                                                          │
          │              <Ok>                  <Back>                │
          │                                                          │
        └─────────────────────────────────────────────────┘

**6.**

```
root@DietPi:~# dmesg | grep -i cfg80211
[    5.339882] cfg80211: World regulatory domain updated:
[    5.339919] cfg80211:  DFS Master region: unset
[    5.339932] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    5.339951] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    5.339967] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    5.339982] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[    5.340001] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[    5.340019] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[    5.340035] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[    5.340050] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[    5.340066] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)

root@DietPi:~# dmesg | grep -i usbcore
[    0.122209] usbcore: registered new interface driver usbfs
[    0.122301] usbcore: registered new interface driver hub
[    0.122432] usbcore: registered new device driver usb
[    1.051233] usbcore: registered new interface driver smsc95xx
[    1.520438] usbcore: registered new interface driver usb-storage
[    1.630923] usbcore: registered new interface driver usbhid
[    5.433036] usbcore: registered new interface driver rtl8821au

root@DietPi:~# dmesg | grep -i 8812au
root@DietPi:~# dmesg | grep -i rtl8812au
root@DietPi:~# dmesg | grep -i RTL8812au
root@DietPi:~# dmesg | grep -i RTL871X
[    5.173096] RTL871X: module init start
[    5.173135] RTL871X: rtl8821au v4.3.14_13455.20150212_BTCOEX20150128-51
[    5.173146] RTL871X: build time: May 13 2016 16:55:31
[    5.173156] RTL871X: rtl8821au BT-Coex version = BTCOEX20150128-51
[    5.430576] RTL871X: rtw_ndev_init(wlan0)
[    5.433062] RTL871X: module init ret=0
[   10.760750] RTL871X: nolinked power save enter
```

**7.**

No I dont.

**8.**

Only wget to wifi driver, apt-get update & apt-get upgrade.

I try unplugin ethernet cable, but not connect, some times the boot crash doing this.
My wifi dongle work well in openelec, in raspbian I install driver wifi from here https://www.raspberrypi.org/forums/viewtopic.php?p=849631#p849631 to work.

k-plan, 2016-05-15:
@arv187 

thanks for sharing your boot-up video. Cool idea! 👍 

Video:

- in the boot-up screen output,  there is running a file check (00:00:20) `/dev/mmcblk0p2 `
- same at  `/dev/mmcblk0p1 ` FAT  `Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.`
Not a good sign. I hate to said this, problem with sd-card or worst PSU ?

[image]

- show a `Flash_Drive_Fit Samsung_USB` ? Have you a USB drive attached?
`Fail to mount /dev/usb_1`
- one of the `NetData` processes fail (with `Nginx`) but this is a incidental defect.
- BUT there is NO IP ADDRESS at login prompt.
[image]

So I have test (on my RPi B+) if it is a non-english system locale issue.
```
root@RPi-B-Plus:~# locale
LANG=es_ES.UTF-8
```
But it works here. It will show me the correct IP Address at login.

Posted output:
- your `/etc/network/interfaces` looks fine. Both interfaces use DHCP

If it is possible, can you test eth0 = STATIC and wlan0 = DHCP?
* Please do this via `dietpi-config`. Best practice, disable WiFi via menu first.
* Then set ethernet to static (incl. DNS), save and restart network via menu.
* Test internet connection via menu.
* Exit menu and logout ( `exit` ). If you now login again, is a IP Address printed in login screen?
* If so, do a `reboot`.
* After fully boot-up,  is a IP Address printed in login screen?

arv187, 2016-05-15:
Thanks K-plan,
Forget the video hehe, I restored imagen in Sony class 10 sd card. worst psu, may be, its a apokin 2.1A, and I have a samsung usb memory, but system is installed in sd card only.

Sometimes in reboot no wifi or lan work, then I unplug PSU to reboot because I haven't usb keyboard or hdmi screen, may be that data is corrupted because this.
In video raspberry its attach to my hdmi tv that is in my living room, and not IP address because wifi isn't working and my router with ethernet cable is in my room, then not there wan or eth0 up, is headless. Was only to know because not boot in my room with ethernet cable plug and ssh putty.

Appart of video, continuing with a fresh working system where all work via putty (except wifi)....
Yes both interfaces are in dhcp. I have doubt about if I can get eth0 and wifi working together or I need disable eth0 to wifi work, in that case if wifi not work I couldn't connect via putty.

I go to follow your instructions and write here again.

k-plan, 2016-05-15:
@arv187 

> Appart of video, continuing with a fresh working system where all work via putty (except wifi)....

Best idea! 💯  Please fully format sd-card  (no Quick-Format) eg. with "SDFormatter"
In your case, I would start with a fresh install with default kernel (no rpi Firmware Update), update to v117 and no extra software. Install wifi firmware like describe above and get your network eth0 = static / wlan0 = dhcp on work. It should work for you.

> Yes both interfaces are in dhcp. I have doubt about if I can get eth0 and wifi working together or I need disable eth0 to wifi work, in that case if wifi not work I couldn't connect via putty.

Yes in v117, we have now the possibility to have two active network interface.

[cangelog](https://raw.githubusercontent.com/Fourdee/DietPi/master/CHANGELOG.txt)

arv187, 2016-05-15:
Please do this via dietpi-config. Best practice, disable WiFi via menu first.
    Then set ethernet to static (incl. DNS), save and restart network via menu.
    Test internet connection via menu. online
    Exit menu and logout ( ssh = exit ). If you now login (ssh) again, is a IP Address printed in login screen? Yes, ever its showed ethernet IP,
For first time wifi led still alive, before only start blink at boot or scan.

    If so, do a reboot.
    After fully boot-up, is a IP Address printed in login screen (e.g. ssh)?
Yes ip address (lan) its printed in login screen but In reboot led only blink at start, not still alive.

Go in again to config menu, re-fixing static address to eth0 not work, but when I disable and after enable wifi interface led blink still in the time but config menu show disconnected.
If I use copy current address to static values are wrong, 

```
 - Mode    = DHCP                                             │
       │  - Ip      = 0.0.0.0                                          │
       │                                                               │
       │        Scan           Scan and Connect                        │
       │        Manual         Manually Set Wireless Details           │
       │        Change Mode    DHCP                                    │
       │        Copy           Copy Current address to Static          │
       │        Static Ip      192.168.0.100                           │
       │        Static Mask    255.255.255.0                           │
       │        Static Gateway 192.168.0.1                             │
       │        Static DNS     8.8.8.8 8.8.4.4                         │
       │        Disable        Disable WiFi adapter                    │
       │        Apply          Save Changes (Restart Networking)    
```
Fix manually values. but not connect. Ever this error: 
ifup: interface eth0 already configured
run-parts: /etc/network/if-pre-up.d/wireless-tools exited with return code 2
Failed to bring up wlan0.

 ```
DietPi-Services
 Mode: stop
 Please wait...
 Completed

 - Dropping connections..

 - Restarting networking services...
ifup: interface eth0 already configured
run-parts: /etc/network/if-pre-up.d/wireless-tools exited with return code 2
Failed to bring up wlan0.

 DietPi-Services
 Mode: start
 Please wait...
 Completed

 DietPi-Process Tool
 Applying settings, please wait...

Cron (3952)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

OpenSSH Server (426)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
OpenSSH Server (1311)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Samba Server (Daemon) (858)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Samba Server (Daemon) (867)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Samba Server (NetBios) (848)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

 Completed
```

arv187, 2016-05-15:
```
root@DietPi:~# sudo ifdown wlan0 && sudo ifup -v wlan0
ifdown: interface wlan0 not configured
Configuring interface wlan0=wlan0 (inet)
run-parts --exit-on-error --verbose /etc/network/if-pre-up.d
run-parts: executing /etc/network/if-pre-up.d/wireless-tools
run-parts: /etc/network/if-pre-up.d/wireless-tools exited with return code 2
Failed to bring up wlan0.
root@DietPi:~#
```

k-plan, 2016-05-15:
@arv187 

`ifup: interface eth0 already configured`
- your ethernet is working, you can login via ssh, have a static ip address, which is print correct on login prompt? 
Right?

- You start form the scratch with a fresh install? before writing image on sd-card, you have format it with SDFormatter?

BTW: you don't need sudo, you are root. You don't need manually `ifup` or `ifdown` if you use `dietpi-config` menu. If you do this or edit `/etc/network/interfaces`, `dietpi-config` can not work proper.
either all doing by hand (if you know, what you are doing) or  let `dietpi-config` do it for you.

Which way you like to do?

k-plan, 2016-05-15:
Note for other readers here: 
- interface is configured via `dietpi-config` menu with DHCP and correct WiFi credentials and saved.
[image]

- exit menu and executing by hand for testing:

```
root@RPi-B-Plus:~# sudo ifdown wlan0 && sudo ifup -v wlan0
ifdown: interface wlan0 not configured
Configuring interface wlan0=wlan0 (inet)
run-parts --exit-on-error --verbose /etc/network/if-pre-up.d
run-parts: executing /etc/network/if-pre-up.d/wireless-tools
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Error for wireless request "Set Encode" (8B2A) :
    invalid argument "wifi-password".
run-parts: executing /etc/network/if-pre-up.d/wpasupplicant
wpa_supplicant: wpa-driver nl80211,wext (default)
wpa_supplicant: /sbin/wpa_supplicant -s -B -P /run/wpa_supplicant.wlan0.pid -i wlan0 -D nl80211,wext -C /run/wpa_supplicant
Starting /sbin/wpa_supplicant...
wpa_supplicant: waiting for "/run/wpa_supplicant.wlan0.pid":  0 (max. 5)
wpa_supplicant: creating sendsigs omission pidfile: /run/sendsigs.omit.d/wpasupplicant.wpa_supplicant.wlan0.pid
wpa_supplicant: ctrl_interface socket located at /run/wpa_supplicant/wlan0
wpa_supplicant: configuring network block -- 0
wpa_supplicant: wpa-ssid "wifi-ssid" -- OK
wpa_supplicant: wpa-psk ***** -- OK
wpa_supplicant: enabling network block 0 -- OK

dhclient -v -pf /run/dhclient.wlan0.pid -lf /var/lib/dhcp/dhclient.wlan0.leases wlan0   -e IF_METRIC=1
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/00:c0:ca:xx:xx:xx
Sending on   LPF/wlan0/00:c0:ca:xx:xx:xx
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 4
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 12
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 13
DHCPREQUEST on wlan0 to 255.255.255.255 port 67
DHCPOFFER from 192.168.0.1
DHCPACK from 192.168.0.1
bound to 192.168.0.204 -- renewal in 2812 seconds.
run-parts --exit-on-error --verbose /etc/network/if-up.d
run-parts: executing /etc/network/if-up.d/000resolvconf
run-parts: executing /etc/network/if-up.d/ifmetric
run-parts: executing /etc/network/if-up.d/mountnfs
run-parts: executing /etc/network/if-up.d/upstart
run-parts: executing /etc/network/if-up.d/wpasupplicant
```

Most of the error messages caused by  dietpi use only on file (/etc/network/interfaces) to configure network like "old debian style networking"  (legacy)

arv187, 2016-05-15:
ifup: interface eth0 already configured

your ethernet is working, you can login via ssh, have a static ip address, which is print correct on login prompt?
    Right? **Yes.**

You start form the scratch with a fresh install? before writing image on sd-card, you have format it with SDFormatter?
 **Yes, SDFormatter & win32 disk imager to write image. My first post was with kernel 4.1.19 & dietpi V117 with same error message code 2.
Im using a fresh install since post that include video (video was maked before of fresh installation) all results from that post was with fresh install, only have kernel updated at 4.4.10-v7+, dietpi V117, this modules:
 FFmpeg
Java 
Node.js
Openssh
samba**

BTW: you don't need sudo, you are root. You don't need manually ifup or ifdown if you use dietpi-config menu. If you do this or edit /etc/network/interfaces, dietpi-config can not work proper.
either all doing by hand (if you know, what you are doing) or let dietpi-config do it for you.

Which way you like to do?

**Im using dietpi-config, but Im tired then try others options although I dont know what Im doing. But I only did by hand what you saw**

┌──────────┤ DietPi - Ethernet Network Options ├───────────┐
          │  Current Details:                                        │
          │  - Mode    = STATIC                                      │
          │  - Ip      = 192.168.1.100                               │
          │  - Mask    = 255.255.255.0                               │
          │  - Gateway = 192.168.1.1                                 │
          │  - DNS     = 192.168.1.1                                 │
          │                                                          │
          │     Change Mode    STATIC                                │
          │     Copy           Copy Current address to Static        │
          │     Static Ip      192.168.1.100                         │
          │     Static Mask    255.255.255.0                         │
          │     Static Gateway 192.168.1.1                           │
          │     Static DNS     192.168.1.1                           │
          │     Disable        Disable Ethernet adapter              │
          │     Apply          Save Changes (Restart Networking)     │
          │                                                          │
          │                                                          │
          │              <Ok>                  <Back>                │
          │                                                          

 Current Details:                                             │
       │  - SSID    = xxxxx | 0 Mbit | Strength: 0%           │
       │  - KEY     =            xxxxxx                                      │
       │ xxxxx                                                        │
       │  - Mode    = STATIC                                           │
       │  - Ip      = 0.0.0.0                                          │
       │                                                               │
       │        Scan           Scan and Connect                        │
       │        Manual         Manually Set Wireless Details           │
       │        Change Mode    STATIC                                  │
       │        Copy           Copy Current address to Static          │
       │        Static Ip      192.168.1.105                           │
       │        Static Mask    255.255.255.0                           │
       │        Static Gateway 192.168.1.1                             │
       │        Static DNS     192.168.1.1                             │
       │        Disable        Disable WiFi adapter                    │
       │        Apply          Save Changes (Restart Networking)       │
       │                                                               │
       │                                                               │
       │                <Ok>                    <Back>        

**My wlan IP still on 0.0.0.0**

k-plan, 2016-05-15:
@arv187 

my problem is,  you have now a different os software status (kernel 4.4.10-v7, java, node.js, samba, ssh) to my test system here. 
but anyway ...

- disable wifi
- enable wifi
- set wifi to dhcp
- scan for your wifi access point
- set wifi password

and then screen shot from menu please (you can delete this later)

arv187, 2016-05-15:
I enable guest access wifi connection with wpa/wpa2 with a key with 3 letters+1 number+3 letters+1 number. After in dietpi-config set dhcp, push in scan & connect, select my guest access ssid, this output only happen with this guest access wifi:

```
Connecting to myssid_inv
 , please wait

 DietPi-Services
 Mode: stop
 Please wait...
 Completed

 - Dropping connections..

 - Restarting networking services...
ifup: interface eth0 already configured
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Error for wireless request "Set Encode" (8B2A) :
    invalid argument "my passkey".
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/xx:xx:xx:xx:xx:xx <-- my dongle mac
Sending on   LPF/wlan0/xx:xx:xx:xx:xx:xx
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 15
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 15
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 14
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 10
No DHCPOFFERS received.
No working leases in persistent database - sleeping.

 DietPi-Services
 Mode: start
 Please wait...
 Completed

 DietPi-Process Tool
 Applying settings, please wait...

Cron (3925)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

DHCP Client (3834)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

OpenSSH Server (434)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
OpenSSH Server (1274)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Samba Server (Daemon) (823)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Samba Server (Daemon) (832)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Samba Server (NetBios) (813)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

 Completed
```

Following your instructions in my normal 5g ssid (disabling wifi):

```
 DietPi-Services
 Mode: stop
 Please wait...
 Completed

 - Dropping connections..

 - Restarting networking services...
ifup: interface eth0 already configured

 DietPi-Services
 Mode: start
 Please wait...
 Completed

 DietPi-Process Tool
 Applying settings, please wait...

Cron (4937)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

OpenSSH Server (434)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
OpenSSH Server (1274)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Samba Server (Daemon) (823)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Samba Server (Daemon) (832)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Samba Server (NetBios) (813)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

 Completed
```
Enabling, set to dhcp, scan, set pass:

```
Scanning SSIDS, please wait....
-------------------------------
xxxxxxxxxxx

Connecting to myssid , please wait

 DietPi-Services
 Mode: stop
 Please wait...
 Completed

 - Dropping connections..

 - Restarting networking services...
ifup: interface eth0 already configured
run-parts: /etc/network/if-pre-up.d/wireless-tools exited with return code 2
Failed to bring up wlan0.

 DietPi-Services
 Mode: start
 Please wait...
 Completed

 DietPi-Process Tool
 Applying settings, please wait...

Cron (4210)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

OpenSSH Server (427)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
OpenSSH Server (1342)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Samba Server (Daemon) (881)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Samba Server (Daemon) (890)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Samba Server (NetBios) (871)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

 Completed
```
[image]

With a 2,4ghz ssid wpa/wpa2 and numeric key of 8 digit:

```
Connecting to myssid
 , please wait

 DietPi-Services
 Mode: stop
 Please wait...
 Completed

 - Dropping connections..

 - Restarting networking services...
ifup: interface eth0 already configured
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/xx:xx:xx:xx:xx:xx
Sending on   LPF/wlan0/xx:xx:xx:xx:xx:xx
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 15
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 14
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 14
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 4
No DHCPOFFERS received.
No working leases in persistent database - sleeping.

 DietPi-Services
 Mode: start
 Please wait...
 Completed

 DietPi-Process Tool
 Applying settings, please wait...

Cron (4882)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

DHCP Client (4791)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

OpenSSH Server (439)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
OpenSSH Server (1273)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Samba Server (Daemon) (821)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0
Samba Server (Daemon) (830)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

Samba Server (NetBios) (811)
Ok | Nice      0
Ok | Affinity  0-3
Ok | Scheduler SCHED_OTHER 0

 Completed
```

k-plan, 2016-05-16:
@arv187 

**_Output:_**
your first output looks good, but your device can't get a ip address via dhcp, because it can't login in AP/WiFi-Router (wpa/wpa2) or DHCP server on AP/WiFi-Router don't answer or give your device any address. Have a look in AP/WiFi-Router log files and `dmesg`
( e.g. see below: `RTL871X: start auth` , `RTL871X: auth success, start assoc` )

---

**_WiFi general_**
Have AP/WiFi-Router here configured only to use "WPA2 PSK" with only support encryption  "aes ccm" and group encryption  "aes ccm". 
Regulatory domain is set to support my country. 
AP/WiFi-Router Channels:  "channel-auto"
Have 2.4 and 5 GHz support and guest WLAN.
Passphrase have:
- minimum 10 digit
- alphabetic character (only a-z, A-Z, NO umlaut )
- digit (0-9)
- ONLY symbol ( - / . _ )
- NO special character 

---

**_Linux Kernel 4.4.10+ #885_**
Now as well tested with 4.4.10  kernel here. After installing the new 8812au-4.4.10-885 driver and reboot, it works out of the box.

Changing connected WLAN always via `dietpi-config` and `Scan and Connect `
To be sure, you have to do this (in some rare cases), **_better twice_**!

Tested with:
- 2.4 GHz (work)
-  5GHz (work)
- Guest-WLAN (work)
- DHCP (work)
-  static IP (work)

So I will suggest, you have a look on  AP/WiFi-Router settings (see above) and log files.

Some output (working connections):

```
root@RPi-B-Plus:~# uname -a
Linux RPi-B-Plus 4.4.10+ #885 Fri May 13 15:30:07 BST 2016 armv6l GNU/Linux

root@RPi-B-Plus:~# wget https://dl.dropboxusercontent.com/u/80256631/8812au-4.4.10-885.tar.gz

root@RPi-B-Plus:~# tar xzf 8812au-4.4.10-885.tar.gz

root@RPi-B-Plus:~# ./install.sh

root@RPi-B-Plus:~# dmesg
 . . .
[   20.573231] RTL871X: module init start
[   20.573275] RTL871X: rtl8821au v4.3.14_13455.20150212_BTCOEX20150128-51
[   20.573292] RTL871X: build time: May 13 2016 16:54:22
[   20.573306] RTL871X: rtl8821au BT-Coex version = BTCOEX20150128-51
[   22.104695] cfg80211: World regulatory domain updated:
[   22.104741] cfg80211:  DFS Master region: unset
[   22.104758] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   22.104783] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   22.104806] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   22.104825] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[   22.104846] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   22.104869] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   22.104889] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[   22.104906] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[   22.104925] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   22.650510] RTL871X: rtw_ndev_init(wlan0)
[   22.690073] usbcore: registered new interface driver rtl8821au
[   22.690113] RTL871X: module init ret=0
[   25.509321] RTL871X: set ssid [wifi-ssid] fw_state=0x00000008
[... 63 lines cut ...]
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Channel 140 : 5.7 GHz
          Current Frequency:5.22 GHz (Channel 44)
```

arv187, 2016-05-16:
Thanks @k-plan I will check that.
But I dont understand why in openelec, raspbian, lubuntu or windows work fine.
Im using wpa/wpa2 - type of key: password (8 digits minimun) and a key similar to this:
3%J2t;uZ`@PpXYp,_b3m{||@bL=+iU z}=H[f So55vJ9H4Oj[d>Z0$iihYn]'E
WPA 504 bits 63 characters I think,

k-plan, 2016-05-16:
@arv187 

> But I dont understand why in openelec, raspbian, lubuntu or windows work fine.

yes all based on debian sources but ...

- openelec => own optimised mini linux image only to run Kodi (and different kernel)
- raspbian => run desktop environment with a network manager
- lubuntu => run desktop environment with a network manager (different kernel - based on ubuntu)
- window => (not really, or?) different only graphical OS on different architecture (not ARM)

 . . .  and look, your linked 8812au-4.4.10-885 driver had to by compiled for every rasbian and especially for every kernel!!

> 3%J2t;uZ`@PpXYp,_b3m{||@bL=+iU z}=H[f So55vJ9H4Oj[d>Z0$iihYn]'E

In this string are more problematical characters!

http://help.avm.de/fritzbox.php?oem=avm&hardware=173&language=de&set=300E001&topic=hilfe_wlan_wpa

> Besondere Vorsicht ist bei Sonderzeichen aus dem internationalen Sprachraum (z. B. ü, ö, ä,Ä, Ö, Ü, ß, `, ´, °, ^, oder auch §) geboten. Je nach Betriebssystem (Microsoft Windows, Mac OS X, Unix) werden diese völlig verschieden kodiert und sind dann nicht kompatibel zueinander.

... and fu.. for "WPA (TKIP)" - it's dead like WEP

arv187, 2016-05-16:
There not option to select tkip or aes, so I think is aes only in my router.
My guest wlan show same error message and is set type: only wpa2, key: 123456789

- Dropping connections..

 - Restarting networking services...
ifup: interface eth0 already configured
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/74:da:38:1d:bd:97
Sending on   LPF/wlan0/74:da:38:1d:bd:97
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 4
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 9
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 16
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 12
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 17
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
No DHCPOFFERS received.
No working leases in persistent database - sleeping.

But now I set dietpi-config menu to static an is connect.

arv187, 2016-05-16:
root@DietPi:~# dmesg
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.10-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #885 SMP Fri May 13 15:44:27 BST 2016
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3d800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 253952
[    0.000000] free_area_init_node: node 0, pgdat 808c0d80, node_mem_map bcf3a000
[    0.000000]   Normal zone: 2232 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 253952 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9520->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @bcef7000 s22592 r8192 d22464 u53248
[    0.000000] pcpu-alloc: s22592 r8192 d22464 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 251720
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=720 bcm2709.boardrev=0xa01041 bcm2709.serial=0xbca767ab smsc95xx.macaddr=B8:27:EB:A7:67:AB bcm2708_fb.fbswap=1 bcm2709.uart_clock=3000000 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 vc_mem.mem_base=0x3ea00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 987808K/1015808K available (6340K kernel code, 432K rwdata, 1712K rodata, 480K init, 764K bss, 19808K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
    lowmem  : 0x80000000 - 0xbe000000   ( 992 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x807e5330   (8053 kB)
      .init : 0x807e6000 - 0x8085e000   ( 480 kB)
      .data : 0x8085e000 - 0x808ca088   ( 433 kB)
       .bss : 0x808cd000 - 0x8098c1ac   ( 765 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000267] Console: colour dummy device 80x30
[    0.001047] console [tty1] enabled
[    0.001085] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001138] pid_max: default: 32768 minimum: 301
[    0.001437] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001474] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002430] Disabling cpuset control group subsystem
[    0.002500] Initializing cgroup subsys io
[    0.002545] Initializing cgroup subsys memory
[    0.002603] Initializing cgroup subsys devices
[    0.002643] Initializing cgroup subsys freezer
[    0.002680] Initializing cgroup subsys net_cls
[    0.002751] CPU: Testing write buffer coherency: ok
[    0.002836] ftrace: allocating 21193 entries in 63 pages
[    0.039779] CPU0: update cpu_capacity 1024
[    0.039840] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.039868] [bcm2709_smp_prepare_cpus] enter
[    0.039993] Setting up static identity map for 0x8240 - 0x8274
[    0.041808] [bcm2709_boot_secondary] cpu:1 started (0) 18
[    0.042237] [bcm2709_secondary_init] enter cpu:1
[    0.042286] CPU1: update cpu_capacity 1024
[    0.042291] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.042706] [bcm2709_boot_secondary] cpu:2 started (0) 18
[    0.043034] [bcm2709_secondary_init] enter cpu:2
[    0.043056] CPU2: update cpu_capacity 1024
[... 299 lines cut ...]
[    9.333701] Process RTW_CMD_THREAD (pid: 408, stack limit = 0xbba2a210)
[    9.337229] Stack: (0xbba2bed8 to 0xbba2c000)
[    9.340474] bec0:                                                       805b7d0c 800d9594
[    9.344091] bee0: bc20ef5c bc20ef40 bba2bf14 bc20ef40 bf469000 bf46a958 bf46a978 ffff8e6e
[    9.347712] bf00: bf46a958 000048a4 bba2bf5c bba2bf18 7f156b5c 7f179ea8 bba2bf3c bba2a000
[    9.351127] bf20: bb988400 bf46a968 bf46a9a4 b9fe2a00 00000000 00000000 bb988400 bf469000
[    9.354584] bf40: 7f1568cc 00000000 00000000 00000000 bba2bfac bba2bf60 80042660 7f1568d8
[    9.358147] bf60: 3c6c7000 00000000 bba2bf94 bf469000 00000000 00000000 bba2bf78 bba2bf78
[... cut ...]
[   97.661316] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   97.730579] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   98.482352] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   98.549963] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   98.830277] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   98.891294] RTL871X: set ssid [ESM_EF2000_inv] fw_state=0x00000008
[  100.009526] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC1E1

arv187, 2016-05-16:
[image]
Note that don give me info about speed or strench

Fourdee, 2016-05-16:
@arv187 

Can you post results of the following commands please?:
```
iwconfig wlan0
```

> note to self
```
WIFI_BITRATE=$(iwconfig wlan$WIFI_INDEX | grep -m1 'Bit Rate:' | awk '{ print $2 }' | sed 's/Rate://g')
WIFI_SIGNALSTRENGTH=$(iwconfig wlan$WIFI_INDEX | grep -m1 'Signal level=' | awk '{ print $4 }' | sed 's/level=//g' | cut -f1 -d "/")
```

arv187, 2016-05-16:
```
root@DietPi:~# iwconfig wlan0
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

k-plan, 2016-05-16:
@Fourdee 

```
root@RPi-B-Plus:~# iwconfig wlan0
wlan0     IEEE 802.11AC  ESSID:"wifi-ssid_5GHz"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.22 GHz  Access Point: D6:CA:6D:xx:xx:xx
          Bit Rate:780 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=95/100  Signal level=100/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

```
root@RPi-B-Plus:~# iwconfig wlan0
wlan0     IEEE 802.11bgn  ESSID:"wifi-ssid_2.4GHz"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.412 GHz  Access Point: D6:CA:6D:xx:xx:xx
          Bit Rate:270 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=100/100  Signal level=100/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

Fourdee, 2016-05-16:
@k-plan 

> (@Fourdee: I think, his device is not connected ...)

Yep, i parsed your results with the current code, working fine:
```
cat test | grep -m1 'Bit Rate:' | awk '{ print $2 }' | sed 's/Rate://g'
780

cat test | grep -m1 'Signal level=' | awk '{ print $4 }' | sed 's/level=//g' | cut -f1 -d "/"
100
```

arv187, 2016-05-16:
@Fourdee Yes, dont worry, i post you link to dhcp - static test.
https://github.com/Fourdee/DietPi/issues/337#issuecomment-219305372

Basically in dhcp mode I get this error with my guest wlan connection (easy numeric key): 

starting networking services...
ifup: interface eth0 already configured
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/xx:xx:xx:xx:xx:xx
Sending on   LPF/wlan0/xx:xx:xx:xx:xx:xx
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 4
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 9
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 16
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 12
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 17
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
No DHCPOFFERS received.
No working leases in persistent database - sleeping.

And this with my 2,4ghz - 5ghz and a key similar to this:
3%J2t;uZ`@PpXYp,_b3m{||@bL=+iU z}=H[f So55vJ9H4Oj[d>Z0$iihYn]'E WPA 504 bits 63 characters.

Connecting to myssid , please wait

 DietPi-Services
 Mode: stop
 Please wait...
 Completed

 - Dropping connections..

 - Restarting networking services...
ifup: interface eth0 already configured
run-parts: /etc/network/if-pre-up.d/wireless-tools exited with return code 2
Failed to bring up wlan0.

 DietPi-Services
 Mode: start
 Please wait...
 Completed

arv187, 2016-06-18:
With the last update my device isn't detected.

dietpi v119 firmware updated.

`root@DietPi:~# wget https://dl.dropboxusercontent.com/u/80256631/8812au-4.4.13-894.tar.gz
--2016-06-18 20:59:54--  https://dl.dropboxusercontent.com/u/80256631/8812au-4.4.13-894.tar.gz
Resolving dl.dropboxusercontent.com (dl.dropboxusercontent.com)... 45.58.74.5
Connecting to dl.dropboxusercontent.com (dl.dropboxusercontent.com)|45.58.74.5|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 736700 (719K) [application/octet-stream]
Saving to: ‘8812au-4.4.13-894.tar.gz’

8812au-4.4.13-894.t 100%[=====================>] 719.43K  1.29MB/s   in 0.5s

2016-06-18 20:59:56 (1.29 MB/s) - ‘8812au-4.4.13-894.tar.gz’ saved [736700/736700]

root@DietPi:~# tar xzf 8812au-4.4.13-894.tar.gz
root@DietPi:~# ./install.sh
sudo cp 8812au.conf /etc/modprobe.d/.
sudo install -p -m 644 8812au.ko /lib/modules/4.4.13+/kernel/drivers/net/wireless
sudo depmod 4.4.13+

Reboot to run the driver.

If you have already configured your wifi it should start up and connect to your
wireless network.

If you have not configured your wifi you will need to do that to enable the wifi.
root@DietPi:~# uname -a
Linux DietPi 4.4.13-v7+ #894 SMP Mon Jun 13 13:13:27 BST 2016 armv7l GNU/Linux
root@DietPi:~# lsusb
Bus 001 Device 005: ID 7392:a812 Edimax Technology Co., Ltd
Bus 001 Device 004: ID 090c:1000 Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) Flash Drive
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@DietPi:~# lsmod
Module                  Size  Used by
cfg80211              427855  0
rfkill                 16037  1 cfg80211
sg                     18319  0
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
fuse                   83461  1
ipv6                  347530  22
root@DietPi:~#
`

arv187, 2016-06-18:
Ok, its working now configuring wifi with wicd:
[https://blog.bartbania.com/raspberry_pi/easy-wireless-configuration-for-raspberry-pi/](url)

I think that there something not work with key type "phrasse" or dhcp in dietpi network manager.

Fourdee, 2016-06-19:
@arv187 

> I think that there something not work with key type "phrasse"

Possibly. 
Are you using a wifi key with non-letter/number characters? I'd be grateful if you could rearrange the order, then paste your key here (eg: mykey, i'd paste kmyye)

Also, what encryption type is your router set to? eg wpa-psk/wep etc.

arv187, 2016-06-19:
I don't understand this "if you could rearrange the order"

My key its similar to this: bAZkp&\D7XfqnrcQpJvLIgG]},c>P'XCEL$#!.CQRHaN??KywO2YIn!AXM:(o)J WPA/WPA2 504 bits 63 characters

My router options:
Encryption:  ***WPA/WPA2** -   Only WPA2  -  WEP  -  Disabled 
Type: ***key(min 8 characters** - Hex(64digits)
My selection with *****

With a easy numeric key: 123456789 the error is different:

```
starting networking services...
ifup: interface eth0 already configured
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/xx:xx:xx:xx:xx:xx
Sending on   LPF/wlan0/xx:xx:xx:xx:xx:xx
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 4
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 9
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 16
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 12
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 17
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
```

Fourdee, 2016-06-19:
@arv187 

> bAZkp&\D7XfqnrcQpJvLIgG]},c>P'XCEL$#!.CQRHaN??KywO2YIn!AXM:(o)J 

Try putting the key manually into /etc/network/interfaces, replace the following lines:
```
wireless-key bAZkp&\D7XfqnrcQpJvLIgG]},c>P'XCEL$#!.CQRHaN??KywO2YIn!AXM:(o)J 
wpa-ssid bAZkp&\D7XfqnrcQpJvLIgG]},c>P'XCEL$#!.CQRHaN??KywO2YIn!AXM:(o)J 
```Then reboot.

arv187, 2016-06-19:
With " say me invalid argument

Ok, I think that i found the issue.

With wicd was working even rebooting, but when I shutdown yesterday at night and power on today, wifi device is on, but not connect.

I go to my config router, in LAN/dhcp static when i have a static ip to my wifi dongle, never fail, only with diet-pi, I dont know why... ¿May be that take host name not only mac wifi dongle, and have a incompatibility with new hostname?

Well, after delete my wifi dongle from dhcp static list and reboot wifi connect. Now I set my wifi dongle to dhcp list and continue working. shutdown and power on and continue working.
Its very random issue... but seems related with dhcp.

```
*LAN Settings  

You can enable DHCP to dynamically allocate IP addresses to your client PCs, or configure filtering functions based on specific clients or protocols. In addition to that you can assign static addresses for certain clients. The DSL-Easy Box must have an IP address for the local network.

VLAN Binding is to define the port-based VLAN belonging of the physical ports. Each physical port can be assigned to any configured VLAN profile.

Static DHCP
PC List  
    
ID	Name	IP	MAC Address

*WIFI Security settings
Allowed Client Type:  **WPA/WPA2**    WPA2 Only    WEP    Disabled  
Pre-shared key type:	 **Passphrase** (8~63 characters)    Hex (64 digits)
```

Fourdee, 2016-06-20:
> Well, after delete my wifi dongle from dhcp static list and reboot wifi connect. Now I set my wifi dongle to dhcp list and continue working. shutdown and power on and continue working.
Its very random issue... but seems related with dhcp and diet-pi.

In ```dietpi-config``` and whilst in DHCP mode, switch to STATIC, then select "Copy current IP to static" and "apply". 
This will copy your working DHCP IP addresses and use them for STATIC IP. 
Would be interesting to see if this works.

Fourdee, 2016-06-20:
> 
```
No DHCPOFFERS received.

--- 192.168.1.1 ping statistics ---
1 packets transmitted, 1 received
```

Looks like router is failing to completed the DHCP request for IP. Check your router settings, most specifically conflicting address reservation for DHCP.

> When I talked about static dhcp list I was talking about a config page in my router.

Yep, Its probably easier to set reserved IP's in your router and use DHCP on the devices. I do this for all my devices (i have about 30 entries lol). Either way, has the same effect as using a static IP on the device.

arv187, 2016-06-20:
Error is normal, not ethernet cable there. The strange was see connected status and static hehe.

Fourdee, 2016-07-23:
Outdated ticket, marking as closed. Please reopen if required.

closed 2016-07-23 by Fourdee

arv187, 2016-07-24:
Sorry to reopen this issue, today I have had again this issue, 
DNS server was set in 0.0.0.0.0 by default and set to google 8.8.8.8 8.8.4.4 not worked.
But simply **I set manually DNS server with mi router IP and work** again, after unplug ethernet set wifi in dhcp, reboot and wifi working.