MichaIng/DietPi #339: DietPi-Software | Add optimized installation for Couchpotato

Fourdee, 2016-05-15:
>

rhkean, 2016-05-15:
I have automated installs for CouchPotato, HeadPhones, Sonarr, and NzbGet.

After I'm done with the PineA64 image, I will look into generalizing to code so that it can be bundled into DietPi.  

It also streamlines the python and mono installs, so I'll break it out into individual "packages/bundles" for dietpi-software.

Fourdee, 2016-05-16:
@rhkean Excellent, sounds good :+1:

xenfomation, 2016-05-27:
Nice!  And @Fourdee -s pun, too!!

moogu12, 2016-08-02:
> I have automated installs for CouchPotato, HeadPhones, Sonarr, and NzbGet.

After I'm done with the PineA64 image, I will look into generalizing to code so that it can be bundled into DietPi.

It also streamlines the python and mono installs, so I'll break it out into individual "packages/bundles" for dietpi-software.

I am very excited to hear you are going to do this.  I have had a heck of a time getting nzbget working properly on my pine64.  I am new to linux in general and any automated installs is greatly appreciated.

Gravesy83, 2016-08-16:
I'm also fairly interested in this, since i've been slowly moving all my mediacenter and filesharing related software off of my desktop and onto my Raspberry Pi servers, and compiling CouchPotato and NZBGet has been a royal pain in the backside.

bobokun, 2016-08-16:
Also very interested in this. I have compiled CouchPotato and Sonarr along with rutorrent/rtorrent on my odroid c2 but it took a lot of time and effort to do. Automating this along will be very useful!

rhkean, 2016-08-16:
I'll start working on this over the weekend.

(sorry for the delays, guys)

blindpet, 2016-08-16:
I have some code that should help you out - my installer for media servers automates the installation for a lot of the popular automation software. CouchPotato, Sonarr and NZBGet are not compiled as they do not need to be unless outdated guides are being followed - some extensions for CP to speed things up are compiled (libxml but are not necessary). I do not use rtorrent so that is not included.

https://github.com/blindpet/MediaServerInstaller

http://www.htpcguides.com/raspberry-pi-2-home-media-server-installer-image/

thehijacker, 2016-08-26:
I too am very interested in having automated install of:

- NZBGet
- SickRage
- Headphones
- CouchPotato

Please let me know if I can help with any kind of testing. I am using Orange Pi PC image.

thehijacker, 2016-09-02:
Couldn't wait. I have all four apps running on my DietPi server. If you need any tips on how to run them, let me know.

Gravesy83, 2016-09-02:
Personally it's more about not having to tinker with it on my own minimal install images than anything else, so i'll gladly wait for the day this gets implemented on DietPi, in the meantime i'll just use these on a seperate server.

@blindpet : I always compile from source when not using a plug and play setup, i'm simply too used to stuff breaking whenever i didn't (back when Linux was still the new kid on the block and i wasn't a middle-aged curmudgeon).

rhkean, 2016-09-02:
sorry guys, work and family obligations keep consuming me.  I'll get to this as quickly as I can

blindpet, 2016-09-02:
@Gravesy83 my installer pretty much runs all stuff from source or uses the official repos (in the case of sabnzbd and nzbget). I've spent a lot of time on similar issues you have described so the installer practically eliminates these issues. For example you can update everything from the github source from within the respective service's web interfaces.

@rhkean if you do use my installer's code I'd appreciate a shoutout or link somewhere

rhkean, 2016-09-02:
> @rhkean if you do use my installer's code I'd appreciate a shoutout or link somewhere

I'll try to take a look at your stuff, but I already have combined scripts to install everything (sonarr, nzbget, couchpotato, & headphones) on top of DietPi.  I just have to separate them out for individual installs and remove my personal customizations (like different port numbers).  They also take steps to safely install the python 2.7 packages without stepping on python 3 stuff.

bobokun, 2016-10-18:
Any update on this? Is there any way we can help? I haven't contributed to github before but I would like to help out in whatever way I can to get this working :) I'm sure it would benefit the community a lot for dietpi users.

Fourdee, 2017-05-15:
Online doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=7212#p7212

Tested with Transmission

Completed.

closed 2017-05-15 by Fourdee

Invictaz, 2017-05-16:
@rhkean Maybe you can open a new issue for the others (Headphones / Sonarr and others)