MichaIng/DietPi #341: Add HW_ARCH to dietpi-obtain_hw_model and .hw_model

rhkean, 2016-05-17:
We are starting to see more arm64 vs armhf installation situations.

by adding a line 6:  HW_ARCH to the dietpi-obtain_hw_model script, we can change the lines that look like this:
`    if (( ( $HW_MODEL == 12 ) || ( $HW_MODEL >= 40 && $HW_MODEL < 50 ) )); then
`
to this:
`    if (( $HW_ARCH == 3 )); then
`

proposed:
```
# - Line 6 -
# HW_ARCH 4    i386 (or VM)??
# HW_ARCH 3    arm64
# HW_ARCH 2    armhf
# HW_ARCH 1    armel
# HW_ARCH 0    unknown

```

rhkean, 2016-05-17:
Seeking @Fourdee input, so I didn't automatically assign this to me....

Fourdee, 2016-05-17:
@rhkean Yep, was thinking about this the other day. Would be handy to have, rather than the current ```HW_MODEL``` we use now.

Dont think we need armel, all SBC's I know of these days have hardfloat support.

I'd probably suggest something like the following:
```
# - Line 6 -
# HW_ARCH 21    x86_64 (VM's) 
# HW_ARCH 20    x86_32 (you never know :))
# HW_ARCH 10    arm64 (c2) (10 gives us room for any new arm64 types that may come out, and, allows for arm9 arm10 below)
# HW_ARCH 2    arm8 hf (RPi 3)
# HW_ARCH 2    arm7 hf (RPi 2)
# HW_ARCH 1    arm6 hf (RPi 1)
# HW_ARCH 0    unknown
```

rhkean, 2016-05-17:
I knew that you'd have better insight!!!  LOL

Fourdee, 2016-05-18:
Scrape/parsing notes:
```
root@DietPi:~# cat /proc/cpuinfo | grep -m1 'architecture' | awk '{print $3}'
AArch64

root@DietPi:~# arch
aarch64

root@DietPi:~# dpkg --print-architecture
arm64
```

Fourdee, 2016-05-22:
Ok lets go with ```arch```. We need to get a list going:

arm64 = aarch64
amd64 = ?
armv8 = ?
armv7 = ?
armv6 = ?

rhkean, 2016-05-22:
RPi A+ :  armv6l
Odroid Xu4:  armv7l
pineA64+:  aarch64
virtualbox image:  x86_64

rhkean, 2016-05-23:
although, we should probably use

`uname -m
`
for portability

according to http://www.gnu.org/software/coreutils/arch , arch is not installed by default

Fourdee, 2016-05-23:
@rhkean 

> ```uname -m ``` according to http://www.gnu.org/software/coreutils/arch , arch is not installed by default

Excellent :+1:

Fourdee, 2016-05-26:
Ok, this is done, wanted to get it in so we can start using HW_ARCH in our code:  :+1: https://github.com/Fourdee/DietPi/commit/e44ac7e31d757191c4dc6a628f0ed51503b2c734

Marking as closed.

closed 2016-05-26 by Fourdee

rhkean, 2016-05-26:
@Fourdee 
NICE!!!
sorry for my lack of contributions for the last week.... swamped at work.

Fourdee, 2016-05-26:
@rhkean 
Yep, ```HW_ARCH``` should help clean up the code, and, make it easier in the long run, so was a good idea :+1: 

No worries Rob :+1:  Do what we can, when we can :)