MichaIng/DietPi #343: General | Enable serial console by default on all DietPi images

Fourdee, 2016-05-17:
>

Fourdee, 2016-05-20:
:u6307: BPi M2+

Fourdee, 2016-05-25:
Ok so the initial coding is done: https://github.com/Fourdee/DietPi/commit/2922cb8cce7399c2737d42b7e9b895dd19e483c0

Serial console will now be enabled by default for all future images we create (eg: when finalize is run). The current images will remain disabled by default.

Fourdee, 2016-06-18:
Most likely going to annoy a few people, doing a u-turn on serial consoles, but here goes:

#### Serial console will now be Disabled for all DietPi images by default. Heres why:
- DietPi is supposed to be minimal and optimized. Serial console uses up resources.
- DietPi requires a network connection to run through the initial setup. So generally speaking, the ease of remote access over SSH should always be preferred, when compared to serial.
- 99.9% of pre boot issues that may require a serial console to debug, will always be caused by a faulty/unstable PSU and/or corrupt filesystem on flash device.
- Very few people know what a serial console is, let alone use one.

If users still require a serial console, they can set the following in dietpi.txt prior to boot. DietPi will then enable the serial console during 1st run:
```
serial_console_enabled=1
``` 
Users can also toggle the serial console via ```dietpi-config``` > advanced.

Open for discussion, so if you have feedback, let me know.

Fourdee, 2016-06-22:
v120 SBC images completed.

closed 2016-06-22 by Fourdee

axel-h, 2016-06-23:
I agree that there is really not need to run a login prompt there. But aside from this, dies the console really eat this much resources?
But can we get a message "serial console disabled" at least then? Seems I'm coming from a completely different background and I've see too many cases where the serial console prints "loading kernel..." and then nothing - which usually means the kernel has crashed. Having a nice message there that says things are fine and now please use SSH or the network to connect would really help. 
Thinking of this, printing the DietPi version and IP might also be nice. Or an option to get the full dmsg output printed there.