MichaIng/DietPi #345: BPi M2+ | Add support and create DietPi image

Fourdee, 2016-05-19:


Fourdee, 2016-05-19:
ARM build tools failing (for reference):
https://github.com/igorpecovnik/lib/issues/310#issue-155720159

k-plan, 2016-05-19:
@Fourdee 

hmm... why BPi M2+ now? Do you get a developer sample device?
And why have to compile a new u-boot or kernel? Support for serial console?

xenfomation, 2016-05-19:
Cool!

Fourdee, 2016-05-19:
@k-plan 

> hmm... why BPi M2+ now? Do you get a developer sample device?

Yep dev sample board came today. Using ARMbian build tools to create a base Jessie image. Left a compile running when I took misses hospital, its failed again, so will look tomorrow.

Fourdee, 2016-05-19:
Looks like ARMbian build tools fails if ```lirc``` (IR receiver software) and ```hostapd``` (wifi hotspot) isnt enabled in the package list of ```lib/configuration.sh```.
[image]

Will try a default ARMbian compile tomorrow to verify. I'd use the ARMbian premade image, but its only 1 partition. DietPi needs 2 (FAT16/32 + EXT4).

rhkean, 2016-05-19:
The kernel shouldn't be dependent on the partition layout unless it's an android image build (where the linux kernel is bundled up into a fake android image).  Some of the BSP's (like the Pine64 ) get released like that b/c the chip manufacturer got android booting on the chip and left it at that.

but the number/layout of partitions is probably more of a initrd thing.

send me some links... I'll take a look tonight.

Fourdee, 2016-05-19:
@rhkean 
Interesting and awesome at the same time :)  :+1: 

Not sure what links your after, hopefully these should help:
- ARMbian premade (Jessie server) image: http://www.armbian.com/banana-pi-m2-plus/ (1 partition)
- ARMbian build tools: https://github.com/igorpecovnik/lib#how-to-build-my-own-image-or-kernel . Add ```BOOTSIZE=64``` to ```lib/configuration.sh``` to create a 2 partition image. Theres also a bug in the first run script of ARMbian when building with 2 partitions (https://github.com/igorpecovnik/lib/issues/290#issuecomment-217435551). You'll need to do the following before powering on:
```
#After compile
- write image.
- mount sd/image 2nd parition to modify files
- Disable do_expand_rootfs() from /etc/init.d/firstrun
- Disable ln -sf /boot/bin/*.bin /boot/script.bin from /etc/init.d/firstrun
- Copy the correct .bin if required: eg: cp /boot/bin/orangepipc.bin /boot/script.bin

Power on
```

If your doing the ARMbian build tools, I've got a 14.04 x64 Vmware VM ready to go if you need it? Just let me know and i'll upload it.

rhkean, 2016-05-20:
>    . Add BOOTSIZE=64 to lib/configuration.sh before you run ./compile.sh
>    ARMbian when building with 2 partitions (igorpecovnik/lib#290 (comment)
> - Disable (comment) do_expand_rootfs() from /etc/init.d/firstrun
> - Disable (comment) ln -sf /boot/bin/*.bin /boot/script.bin from /etc/init.d/firstrun

that's it, the jessie image.

yes, I'm fairly certain that I can reconfigure it as 2 partitions.  But I
don't have a board to test with...

here is what I believe will do it (kinda winging it, though)
looks like there's some u-boot stuff at the beginning of the image, so
you'll need to copy that across

> dd if=Armbian_5.10_Bananapim2plus_Debian_jessie_3.4.112.raw of=/dev/sdX
bs=1048576 count=1
> partprobe /dev/sdX

then create your partitions (I prefer parted.)
delete old partition
> parted /dev/sdX rm 1

create new boot partition
> parted /dev/sdX mkpart pri fat16 1M 75.5M
> mkfs.fat -n boot /dev/sdX1

create new root partition
> parted /dev/sdX mkpart pri ext4 75.5M 2048M
> mkfs.ext4 -L / /dev/sdc2

mount the image and new partitions
> mkdir /mnt/src_armbian
> mount -o loop,offset=1M Armbian_5.10_Bananapim2plus_Debian_jessie_3.4.112.raw
/mnt/src_armbian

> mkdir /mnt/new_armbian
> mount /dev/sdX2 /mnt/new_armbian
> mkdir /mnt/new_armbian/boot
> mount /dev/sdX1 /mnt/new_armbian/boot

copy everything
> cp -av /mnt/src_armbian/* /mnt/new_armbian

edit boot.cmd
> cd /mnt/new_armbian
> sed -i 's/mmcblk0p1/mmcblk0p2/' boot.cmd

rebuild boot.scr
> mkimage -C none -A arm -T script -d boot.cmd boot.scr

rhkean, 2016-05-20:
that's it, the jessie image.

yes, I'm fairly certain that I can reconfigure it as 2 partitions.  But I
don't have a board to test with...

here is what I believe will do it (kinda winging it, though)
looks like there's some u-boot stuff at the beginning of the image, so
you'll need to copy that across
```
dd if=Armbian_5.10_Bananapim2plus_Debian_jessie_3.4.112.raw of=/dev/sdX bs=1048576 count=1
partprobe /dev/sdX

```
then create your partitions (I prefer parted.)
delete old partition
`parted /dev/sdX rm 1
`

create new boot partition
```
parted /dev/sdX mkpart pri fat16 1M 75.5M
mkfs.fat -n boot /dev/sdX1

```
create new root partition
```
parted /dev/sdX mkpart pri ext4 75.5M 2048M
mkfs.ext4 -L / /dev/sdc2
```
mount the image and new partitions
```
mkdir /mnt/src_armbian
mount -o loop,offset=1M Armbian_5.10_Bananapim2plus_Debian_jessie_3.4.112.raw
/mnt/src_armbian
```
mkdir /mnt/new_armbian
> mount /dev/sdX2 /mnt/new_armbian
> mkdir /mnt/new_armbian/boot
> mount /dev/sdX1 /mnt/new_armbian/boot

copy everything
> cp -av /mnt/src_armbian/* /mnt/new_armbian

edit boot.cmd
> cd /mnt/new_armbian
> sed -i 's/mmcblk0p1/mmcblk0p2/' boot.cmd

rebuild boot.scr
> mkimage -C none -A arm -T script -d boot.cmd boot.scr

Fourdee, 2016-05-20:
@rhkean 
Thanks for this, I managed to get ARMbian to build with default packages.

Had to manually strip packages afterwards:
```HW_MODEL=50```
[image]

rhkean, 2016-05-20:
cool.

If you get a chance to test my instructions, I'd appreciate it.... It'll let me know if I'm grasping some of the u-boot processes correctly.

Fourdee, 2016-05-20:
Cant promise it will be anytime soon, but I will give it a spin.

Fourdee, 2016-05-20:
#### So issues:

#### eth0 doesnt always come up after boot 
```
root@DietPi:~# dmesg | grep eth0
[    9.360333] gmac0 gmac0: eth0: eth0: PHY ID 001cc915 at 0 IRQ poll (gmac0-0:00)
[   11.803256] gmac0 gmac0: eth0: No PHY found!
```

#### No onboard wlan support

#### CPU max freq is sometimes 912mhz, instead of 1.2ghz
```
root@DietPi:~# cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     armv7l
 Temp         |     0'c | Who put me in the freezer!
 Governor     |     interactive

                 Current Freq    Min Freq   Max Freq
 CPU0         |      648 Mhz      648 Mhz    912 Mhz
 CPU1         |      648 Mhz      648 Mhz    912 Mhz
 CPU2         |      648 Mhz      648 Mhz    912 Mhz
 CPU3         |      648 Mhz      648 Mhz    912 Mhz

root@DietPi:~# dmesg | grep cpu
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] PERCPU: Embedded 8 pages/cpu @c1473000 s11904 r8192 d12672 u32768
[    0.000000] pcpu-alloc: s11904 r8192 d12672 u32768 alloc=8*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.001988] Initializing cgroup subsys cpuacct
[    0.040333] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.040354] [sunxi_smp_prepare_cpus] enter
[    0.041544] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.041682] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.041682] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.927932] calibrat: max_cpufreq 1200Mhz Type 0!
[    0.927948] [cpu_freq] ERR:get cpu extremity frequency from sysconfig failed,                                                                  use max_freq
```

k-plan, 2016-05-20:
@Fourdee 

Wow, sinovoip kept his word and really quick. Great news!
Congratulations and wish you less hitch as with Orange-Pi PC. 

@rhkean 

> If you get a chance to test my instructions, I'd appreciate it.... 

I like to test your instructions, if it is okay for you. Can not promise, whether I have the time and I get it on work for Allwinner A20 (cubietruck) based on armbian image. Zero/null knowledge and experience.
It not more that a fixed idea: [DietPi for Cubietruck](http://dietpi.com/phpbb/viewtopic.php?f=12&t=421)

rhkean, 2016-05-20:
@k-plan 
send me a link to the image.  I will verify that the image is laid out the same way or not.  If it is, then the instructions should work as written.  If not, I'll try to modify them for the cubietruck image

Fourdee, 2016-05-20:
@rhkean 

Just uploaded: http://dietpi.com/downloads/images/DietPi_BananaPi-M2plus-(Jessie).7z

This is roughly the the 10th compile of this image (including the failed attempts due to errors) using ARMbian build tools.

k-plan, 2016-05-21:
@rhkean 

thank you very much for your offer!  😃 

Image: [Cubietruck Jessie Server with Legacy Kernel 3.4.112  ](http://mirror.igorpecovnik.com/Armbian_5.10_Cubietruck_Debian_jessie_3.4.112.7z)

Description: http://www.armbian.com/cubietruck/

Fourdee, 2016-05-21:
Onboard EMMC R/W performance:
[image]

Fourdee, 2016-05-21:
Released under Beta: http://dietpi.com/phpbb/viewtopic.php?f=8&t=431#p1891

rhkean, 2016-05-21:
@k-plan 
same process, but the Cubietruck image has a 2MB block at the beginning... so here's the modified instructions...

1) copy 2MB BOOT1/ATF/DTB binary blob section from image to SD card
```
dd if=Armbian_5.10_Cubietruck_Debian_jessie_3.4.112.raw of=/dev/sdX bs=2097152 count=1
partprobe /dev/sdX
```
2) then create your partitions (I prefer parted.)
delete old partition
```
parted /dev/sdX rm 1
```
3) create new boot partition
```
parted /dev/sdX mkpart pri fat16 2M 75.5M
mkfs.fat -n boot /dev/sdX1
```
4) create new root partition
```
parted /dev/sdX mkpart pri ext4 75.5M 2048M
mkfs.ext4 -L / /dev/sdc2
```
5) mount the image and new partitions
```
mkdir /mnt/src_armbian
mount -o loop,offset=2M Armbian_5.10_Cubietruck_Debian_jessie_3.4.112.raw /mnt/src_armbian

mkdir /mnt/new_armbian
mount /dev/sdX2 /mnt/new_armbian
mkdir /mnt/new_armbian/boot
mount /dev/sdX1 /mnt/new_armbian/boot
```
6) copy everything
```
cp -av /mnt/src_armbian/* /mnt/new_armbian
```
7) edit boot.cmd
```
cd /mnt/new_armbian
sed -i 's/mmcblk0p1/mmcblk0p2/' boot.cmd
```
8) rebuild boot.scr
```
mkimage -C none -A arm -T script -d boot.cmd boot.scr
```

Fourdee, 2016-05-23:
Marking this as closed:
I will continue to support the M2+ and offer improvements/fixes during dietpi-updates.

closed 2016-05-23 by Fourdee