MichaIng/DietPi #347: Diet-Pi Software | No Machine

xenfomation, 2016-05-19:
This secure, omni OS supported software acts as a remote desktop server or client. It is light weight and easy to install and configure. 

Will provide more insight to these steps as for consideration to be included under Linux Software. 

It is a secure alternative to xrdp - just hoping to diversify the offerings.

k-plan, 2016-05-20:
@xenfomation 

I personally like multi-platform-supporting-software. :+1: 

https://www.nomachine.com/download

xenfomation, 2016-05-21:
Thanks!  It allows me to access many of my machines - virtual or real, windows or linux-based - easily.

While the original install generates a very generic configuration, specifically for ports, could be augmented via a whiptail menu post installation.

I am wondering if items such as XRDP and NoMachine should be tied with tightnc, etc under header "---Remote Desktop Access----" or something that indicates Graphical Interfaces?

xenfomation, 2016-05-29:
NoMachine tested and working.

[image]

xenfomation, 2016-05-29:
@Fourdee --

So as part of my check-in, I realized I had no mapping of HW_MODELS & Descriptions as to add an IF/THEN/ELSE condition so that the right DEB package is pulled from NoMachine's downloads.

I can add those in as they support x64, ARMv6, and even ARMHF has a lot of support for tons of boards.

Fourdee, 2016-06-04:
Additional implementation and testing for reference: https://github.com/Fourdee/DietPi/pull/364

Fourdee, 2016-06-08:
Documentation to finish: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=60#p2071

Fourdee, 2016-06-08:
Done.

closed 2016-06-08 by Fourdee