MichaIng/DietPi #350: Transmission | no web interface

Fourdee, 2016-05-21:
Due to Transmission-daemon service starting before network, and, ```dietpi-service```:

```
root@DietPi:~# systemctl status transmission-daemon.service
Process: 470 ExecStart=/etc/init.d/transmission-daemon start (code=exited, status=0/SUCCESS)
root@DietPi:~# systemctl status dietpi-service.service
Process: 483 ExecStart=/etc/init.d/dietpi-service start (code=exited, status=0/SUCCESS)
```

Even though ```dietpi-services``` successfully disables the transmission-daemon service during install (so dietpi-services can take control), its still being launched by Systemd.

Fourdee, 2016-05-21:
notes:
```
root@DietPi:~# find / -type f -name 'transmission-daemon.service'
/var/lib/systemd/deb-systemd-helper-enabled/multi-user.target.wants/transmission-daemon.service
/run/systemd/generator.late/transmission-daemon.service
```

xenfomation, 2016-05-21:
Woah.... that's nasty!  .... research time or sysctl.conf time?

Fourdee, 2016-05-22:
@xenfomation 
Good idea, ```ENABLE_DAEMON=0``` prevented the service from starting, but also when running ```systemctl start transmission-daemon```

Ok, made my own systemD service for this:
```
rm /etc/init.d/transmission-daemon
rm /etc/systemd/system/transmission-daemon.service
cat << _EOF_ > /etc/systemd/system/transmission-daemon.service
[Unit]
Description=Barebones transmission-daemon service
DefaultDependencies=no

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/usr/bin/transmission-daemon --config-dir /var/lib/transmission-daemon/info
ExecStop=/usr/bin/killall -w transmission-daemon
StandardOutput=tty

[Install]
WantedBy=multi-user.target
_EOF_
systemctl enable transmission-daemon.service
systemctl daemon-reload

dietpi-services disable
```

Problem solved :)

Fourdee, 2016-05-22:
Completed, will also patch existing systems during update.
Commit: https://github.com/Fourdee/DietPi/commit/09af08c53e9354d9b69e40ebe8470a73ac4b63ee

Fourdee, 2016-05-30:
Ok, this is still occurring on my C2, but only when using DHCP instead of STATIC IP.

Attempted to obtain logfile data. Generates file but doesn't write to it:
```
/usr/bin/transmission-daemon --config-dir /var/lib/transmission-daemon/info --logfile /var/log/transmission.log
```

studioai, 2016-06-02:
I solved this problem by edit my transmission-daemon.service like below

```
[Unit]
Description=Barebones transmission-daemon service
DefaultDependencies=no
Requires=network.target
After=dhcpcd.service

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/usr/bin/transmission-daemon --config-dir /var/lib/transmission-daemon/info
ExecStop=/usr/bin/killall -w transmission-daemon
StandardOutput=tty

[Install]
WantedBy=multi-user.target
```

Fourdee, 2016-06-02:
@studioai 
Nice fix :+1: 

I'll look into making our ```dietpi-service``` start after dhcpd aswell. Since our service starts transmission service during boot, this should fix transmission:9091 and "missing IP" from banner when in dhcp mode.

Fourdee, 2016-06-02:
Fixed: https://github.com/Fourdee/DietPi/commit/b0ca6c7f33f06d6b105ec284d473233d1832a72b

This also resolves:
- https://github.com/Fourdee/DietPi/issues/359
- https://github.com/Fourdee/DietPi/issues/284#issuecomment-217544170

Fourdee, 2016-06-02:
Did some testing with STATIC IP. IP address not showing in dietpi-banner again. lol...

debugging:
- ```while (( ! $(route | awk '{print $4}' | grep -ci -m1 'UG') ));``` doesn't get triggered. Route is providing us with a valid UG result (up, has gateway).
- Removed netplug package. Now the route command gets triggered correctly.

So netplug is doing something to the adapter/kernel, whilst route provides us with "false" information. Fun!

Fourdee, 2016-06-14:
Ok, workaround:
Netplug is now installed on demand in DietPi-Config when both adapters are enabled (we need netplug to allow for this feature to work correctly). Netplug is uninstalled when only 1 adapter is active.

MichaIng, 2018-09-25:
@Fourdee
Solution should be similar to how we run Sonarr/mpd etc now, with own user but dietpi group to allow access to e.g. samba mounts and the other way round have file, created by transmission as dietpi group to allow other members access.

I am thinking, as editing systemd units always breaks updates for those files, maybe we should add a separate grouping task/function to dietpi-software (besides files/folder permissions), and add each of our software run user to dietpi group as well as to each others softwares run user group (at least for those where cross access is required)? Still we need to assure that software creates files always with full group permissions 77X.

MichaIng, 2018-10-02:
Related: https://github.com/Fourdee/DietPi/issues/2067#issuecomment-426461043
- Plex has same issue than transmission.

MichaIng, 2018-10-06:
PR up for changing Plex + Transmission run as group "dietpi": https://github.com/Fourdee/DietPi/pull/2117

Fourdee, 2018-10-12:
Test:
- 🈯️ Sonarr > Transmission > Plex playback

@MichaIng 

Works a charm, completed 👍

closed 2018-10-12 by Fourdee

joaofl, 2018-11-25:
I know you have closed this, but I wanted to ask you what you think about the approach I gave to the problem you mentioned above.
First is, instead of running the daemon with the transmission user, I changed it to dietpi, which makes things much easier later.
And I also made my own copy of the settings file, to which I'm pointing when running the service. This avoids those catchy issues with the config files being copied from one place to another and being overwritten in some cases.
From `/etc/systemd/system/transmission-daemon.service` I have:
`ExecStart=/usr/bin/transmission-daemon --config-dir /mnt/dietpi_userdata/transmission-config`

If you think it makes sense, I can pull this changes.

joaofl, 2018-11-25:
The complete service config is below:
```
[Unit]
Description=Barebones transmission-daemon service
DefaultDependencies=no

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/usr/bin/transmission-daemon --config-dir /mnt/dietpi_userdata/transmission-config
ExecStop=/usr/bin/killall -w transmission-daemon
StandardOutput=tty
User=dietpi

[Install]
WantedBy=multi-user.target
```

MichaIng, 2018-11-25:
systemd unit is APT install default now, dietpi group added via separate file: `echo -e '[Service]\nGroup=dietpi' > /etc/systemd/system/transmission-daemon.service.d/dietpi-group.conf`
Config file thus is left in APT install default location: `/etc/transmission-daemon/settings.json`
- As long as we install via APT, I would touch as few things as possible, to allow APT updating everything as required?

One thing is, to have everything in one place, to symlink the config dir to `/mnt/dietpi_userdata/transmission_config`?