MichaIng/DietPi #351: DietPi-Software | Porting on other SBC

k-plan, 2016-05-22:
@rhkean 

to not disturb #345 any more,  my testing results form your [instructions](https://github.com/Fourdee/DietPi/issues/345#issuecomment-220789385)

```
## syspreps
apt-get install -y p7zip u-boot-tools

## Download image / extract
wget 'http://mirror.igorpecovnik.com/Armbian_5.10_Cubietruck_Debian_jessie_3.4.112.7z' -O image.7z
p7zip -d image.7z

## get informations
lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    1   15G  0 disk
└─sda1        8:1    1   15G  0 part
mmcblk0     179:0    0  7.4G  0 disk
├─mmcblk0p1 179:1    0   60M  0 part /boot
└─mmcblk0p2 179:2    0  7.4G  0 part /

df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       7.3G  2.0G  5.1G  28% /
devtmpfs        483M     0  483M   0% /dev
tmpfs           487M     0  487M   0% /dev/shm
tmpfs           487M  6.5M  481M   2% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           487M     0  487M   0% /sys/fs/cgroup
tmpfs            10M  1.1M  9.0M  11% /DietPi
tmpfs            20M   24K   20M   1% /var/log
tmpfs           487M     0  487M   0% /tmp
/dev/mmcblk0p1   60M   22M   39M  36% /boot

## copy 2MB BOOT1/ATF/DTB binary blob section from image to SD card (sdX or mmcblk0)
dd if=Armbian_5.10_Cubietruck_Debian_jessie_3.4.112.raw of=/dev/sda bs=2097152 count=1
partprobe /dev/sda

## then create your partitions (I prefer parted.) - delete old partition
parted /dev/sda rm 1

## create new boot partition
parted /dev/sda mkpart pri fat16 2M 75.5M
mkfs.fat -n BOOT /dev/sda1

## create new root partition
parted /dev/sda mkpart pri ext4 75.5M 2048M
mkfs.ext4 -L / /dev/sda2

## mount the image and new partitions
mkdir /mnt/src_armbian
mount -o loop,offset=2M Armbian_5.10_Cubietruck_Debian_jessie_3.4.112.raw /mnt/src_armbian

mkdir /mnt/new_armbian
mount /dev/sda2 /mnt/new_armbian
mkdir /mnt/new_armbian/boot
mount /dev/sda1 /mnt/new_armbian/boot

## copy everything
cp -av /mnt/src_armbian/* /mnt/new_armbian

## edit boot.cmd
cd /mnt/new_armbian/boot/
sed -i 's/mmcblk0p1/mmcblk0p2/' boot.cmd

## rebuild boot.scr
mkimage -C none -A arm -T script -d boot.cmd boot.scr

```

- serial output cubietruck first run with new prepared SD-Card:

````
U-Boot SPL 2016.01-armbian-sun4i (Apr 29 2016 - 02:30:12)
DRAM: 2048 MiB
CPU: 912000000Hz, AXI/AHB/APB: 3/2/2
Trying to boot from MMC

U-Boot 2016.01-armbian-sun4i (Apr 29 2016 - 02:30:12 +0200) Allwinner Technology

CPU:   Allwinner A20 (SUN7I)

I2C:   ready

DRAM:  2 GiB

MMC:   SUNXI SD/MMC: 0

*** Warning - bad CRC, using default environment

HDMI connected: Setting up a 1280x1024 dvi console (overscan 0x0)

In:    serial

Out:   vga

Err:   vga

SCSI:  SUNXI SCSI INIT

SATA link 0 timeout.

AHCI 0001.0100 32 slots 1 ports 3 Gbps 0x1 impl SATA mode

flags: ncq stag pm led clo only pmp pio slum part ccc apst 

Net:   eth0: ethernet@01c50000

starting USB...

USB0:   USB EHCI 1.00

USB1:   USB OHCI 1.0

USB2:   USB EHCI 1.00

USB3:   USB OHCI 1.0

scanning bus 0 for devices... 1 USB Device(s) found

scanning bus 2 for devices... 1 USB Device(s) found

Hit any key to stop autoboot:  2  1  0 

switch to partitions #0, OK

mmc0 is current device

Scanning mmc 0:1...

Found U-Boot script /boot.scr

reading /boot.scr

2426 bytes read in 27 ms (86.9 KiB/s)

## Executing script at 43100000

Failed to mount ext2 filesystem...

** Unrecognized filesystem type **

Failed to mount ext2 filesystem...

** Unrecognized filesystem type **

reading .next

** Unable to read file .next **

Failed to mount ext2 filesystem...

** Unrecognized filesystem type **

Failed to mount ext2 filesystem...

** Unrecognized filesystem type **

reading script.bin

46404 bytes read in 36 ms (1.2 MiB/s)

Failed to mount ext2 filesystem...

** Unrecognized filesystem type **

reading uInitrd

3646110 bytes read in 202 ms (17.2 MiB/s)

Failed to mount ext2 filesystem...

** Unrecognized filesystem type **

reading zImage

** Unable to read file zImage **

Failed to mount ext2 filesystem...

** Unrecognized filesystem type **

Bad Linux ARM zImage magic!

SCRIPT FAILED: continuing...

scanning bus for devices...

Found 0 device(s).

SCSI device 0: 

    Device 0: not available

USB device 0: unknown device

ethernet@01c50000 Waiting for PHY auto negotiation to complete..... done

Speed: 1000, full duplex

BOOTP broadcast 1

DHCP client bound to address 192.168.0.203 (15 ms)

*** Warning: no boot file name; using 'AC100ACB.img'

Using ethernet@01c50000 device

TFTP from server 192.168.0.1; our IP address is 192.168.0.203

Filename 'AC100ACB.img'.

Load address: 0x42000000

Loading: *T 
```

Sorry, no success.

rhkean, 2016-05-22:
@k-plan , bummer, and thank you....

I'll try to dive into this log and figure out what I missed.  Looks like a uboot issue, but I'm not sure.

k-plan, 2016-05-22:
@rhkean 

thanks, and only do this, if you have really spare time. It's not essential, more a trial in order to understand, what going on during image building and what all you coding guys doing out here.
There's no harm if it don't work at the end. But it is possible, we all are able to learn a lot of thing.

closed 2016-08-21 by Fourdee