MichaIng/DietPi #352: DietPi-Software | WiringPi will not install

Fourdee, 2016-05-23:
> https://twitter.com/PiTutorials/status/734720128307367936

Fails on URL connection test. Most likely the download has been removed.

Fourdee, 2016-05-23:
Notes: So two issues:

#### Cert fails:
```
root@DietPi:~# wget https://git.drogon.net/?p=wiringPi;a=snapshot;h=HEAD;sf=tgz --2016-05-23 16:37:09--  https://git.drogon.net/?p=wiringPi
Resolving git.drogon.net (git.drogon.net)... 195.10.226.169, 2a00:ce0:2:feed:beef:cafe:0:3
Connecting to git.drogon.net (git.drogon.net)|195.10.226.169|:443... connected.
ERROR: The certificate of ‘git.drogon.net’ is not trusted.
ERROR: The certificate of ‘git.drogon.net’ hasn't got a known issuer.
```

Add ```--no-check-certificate``` to all wget code in DietPi?

#### Wget not following redirects
```
root@DietPi:~# wget --no-check-certificate https://git.drogon.net/?p=wiringPi;a=snapshot;h=HEAD;sf=tgz
--2016-05-23 16:38:23--  https://git.drogon.net/?p=wiringPi
Resolving git.drogon.net (git.drogon.net)... 195.10.226.169, 2a00:ce0:2:feed:beef:cafe:0:3
Connecting to git.drogon.net (git.drogon.net)|195.10.226.169|:443... connected.
WARNING: The certificate of ‘git.drogon.net’ is not trusted.
WARNING: The certificate of ‘git.drogon.net’ hasn't got a known issuer.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘index.html?p=wiringPi.1’

index.html?p=wiring     [ <=>                  ]  20.53K  --.-KB/s   in 0.03s

2016-05-23 16:38:24 (685 KB/s) - ‘index.html?p=wiringPi.1’ saved [21020]
```

Switch to curl?

rhkean, 2016-05-23:
both of these worked for me:
```
rhkean@keanbean:~/test$ wget -v "http://git.drogon.net/?p=wiringPi;a=snapshot;h=HEAD;sf=tgz" -O package.tgz
--2016-05-23 12:02:32--  http://git.drogon.net/?p=wiringPi;a=snapshot;h=HEAD;sf=tgz
Resolving git.drogon.net (git.drogon.net)... 195.10.226.169, 2a00:ce0:2:feed:beef:cafe:0:3
Connecting to git.drogon.net (git.drogon.net)|195.10.226.169|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘package.tgz’

package.tgz                                                     [   <=>                                                                                                                                     ] 131.16K   284KB/s   in 0.5s

2016-05-23 12:02:33 (284 KB/s) - ‘package.tgz’ saved [134304]
```
and 
```
rhkean@keanbean:~/test$ wget -v --no-check-certificate  "https://git.drogon.net/?p=wiringPi;a=snapshot;h=HEAD;sf=tgz" -O package.tgz
--2016-05-23 12:02:52--  https://git.drogon.net/?p=wiringPi;a=snapshot;h=HEAD;sf=tgz
Resolving git.drogon.net (git.drogon.net)... 195.10.226.169, 2a00:ce0:2:feed:beef:cafe:0:3
Connecting to git.drogon.net (git.drogon.net)|195.10.226.169|:443... connected.
WARNING: The certificate of ‘git.drogon.net’ is not trusted.
WARNING: The certificate of ‘git.drogon.net’ hasn't got a known issuer.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘package.tgz’

package.tgz                                                     [    <=>                                                                                                                                    ] 131.16K   207KB/s   in 0.6s

2016-05-23 12:02:53 (207 KB/s) - ‘package.tgz’ saved [134304]

```
it did not get a redirect... I think your URL needs to be wrapped in quotes
I'm personally not a big fan of --no-check-certificate, but I understand its use these days with LetsEncrypt and the like out there....

but, in this case, I would recommend changing the URL to non-https

Fourdee, 2016-05-23:
@rhkean

> but, in this case, I would recommend changing the URL to non-https

Yep, i agree :1:. Love the simple resolution by just taking out a ```s``` from url heh.

For now, i've changed https to http as per your suggestion. 
I think we will need to revisit certificates wget/curl at a later date. Theres even an option in LetsEncrpt to force https access, so in this case, simply changing to http, may not be a solution?

Fourdee, 2016-05-23:
Done: https://github.com/Fourdee/DietPi/commit/c009c06716602cf0feaf324a5a056fb54e870ffc
Will mark this as closed.

closed 2016-05-23 by Fourdee