MichaIng/DietPi #353: haproxy fails to start

crazyquark, 2016-05-24:
Looks like the default config for haproxy is broken as the service fails to start:
-- Logs begin at Tue 2016-05-24 08:20:23 EEST, end at Tue 2016-05-24 08:44:06 EEST. --
May 24 08:44:06 DietPi haproxy[14414]: [WARNING] 144/084406 (14416) : config : missing timeouts for proxy 'stats'.
May 24 08:44:06 DietPi haproxy[14414]: | While not properly invalid, you will certainly encounter various problems
May 24 08:44:06 DietPi haproxy[14414]: | with such a configuration. To fix this, please ensure that all following
May 24 08:44:06 DietPi haproxy[14414]: | timeouts are set to a non-zero value: 'client', 'connect', 'server'.
May 24 08:44:06 DietPi haproxy[14414]: [WARNING] 144/084406 (14416) : config : 'stats' statement ignored for proxy 'stats' as it requires HTTP
May 24 08:44:06 DietPi haproxy[14414]: [ALERT] 144/084406 (14416) : Fatal errors found in configuration.
May 24 08:44:06 DietPi systemd[1]: haproxy.service: control process exited, code=exited status=2
May 24 08:44:06 DietPi systemd[1]: Failed to start LSB: fast and reliable load balancing reverse proxy.
-- Subject: Unit haproxy.service has failed
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit haproxy.service has failed.
--
-- The result is failed.
May 24 08:44:06 DietPi systemd[1]: Unit haproxy.service entered failed state.
May 24 08:44:06 DietPi haproxy[14414]: failed!

Running on Odroid C1/latest dietpi.

crazyquark, 2016-05-24:
It would appear that the line causing issues is this:

listen stats *:1338
[ALERT] 144/220546 (12162) : parsing [/etc/haproxy/haproxy.cfg:60] : 'listen' cannot handle unexpected argument '*:1338'.

crazyquark, 2016-05-24:
Looks like a haproxy backwards compat issue:
http://www.serverphorums.com/read.php?10,1326393

needed to change that line to:
 listen stats
        bind *:1338

xenfomation, 2016-05-25:
Investigating -- thanks!

xenfomation, 2016-05-25:
Well, I found the code -- will check out, correct, and test before reporting to @Fourdee ....
```

#Admin web page
        # listen stats *:1338
        listen stats
        bind *:1338
        stats enable
        stats uri /
```

Will keep all posted on top of issue #309

crazyquark, 2016-05-25:
haproxy is now version 1.6.2
When I first started dietPi it updated my system to the latest DietPi version(from 115 to 118 I think).

xenfomation, 2016-05-25:
Thanks, @crazyquark  -- this will allow me to ensure a condition is set to check for haproxy < = > 1.6.2

xenfomation, 2016-05-28:
This is the highest item on my backlog list and I will have a commit today for this.

#353 - haproxy fails to start #353
#361 - cli grouping of software?
#357 - DietPi-Software | GNUStep Desktop Environment #357 
#347 - No Machine & stock configs
#325 - DietPi Idea | Sourceable script with pre-defined color codes #325 
#309 - xenserver docs
#247 - DietPi-Software | Add Friendica: A Personal Social Networking Platform #247 
#246 - NFS
#245 - Full Disk Encryption #245
#271 - General | Add support for 2+ USB drives #271 
#195 - DietPi-Software | Add QuiteRSS installation option #195 
#191 - DietPi-Software | Add xrdp installation option #191 
#151 - DietPi-Software | Add MATE desktop installation option #151 
#41  - [Wish List] Convert DietPi-Software into SQL database #41

Fourdee, 2016-07-23:
Outdated ticket, making as closed. Please reopen if required.

attzonko, 2016-11-10:
Has this been fixed? I hit this same issue when I installed haproxy today using dietpi-software

I am running DietPi version 136

Fourdee, 2016-11-10:
@attzonko 

Thanks for letting us know. 👍 
I'll take a look.

attzonko, 2016-11-10:
By the way the workaround suggested by @crazyquark on May 24 corrected the issue for me.

Fourdee, 2016-11-15:
@attzonko 

> By the way the workaround suggested by @crazyquark on May 24 corrected the issue for me.

Yep, same for me. Excellent 👍 
I'll update sourcecode for those changes.

Fourdee, 2016-11-15:
Completed.

closed 2016-11-15 by Fourdee