MichaIng/DietPi #354: DietPi-Software | Add LMS + SqueezeLite to ARM64 (c2/pine)

Fourdee, 2016-05-25:
Currently RPi is the only device that have these installation options available.

Fourdee, 2016-05-25:
Worth a shot:
```
wget https://github.com/Logitech/slimserver/archive/public/7.9.zip -O package.zip
unzip package.zip
rm package.zip

cp -R slimserver-public-7.9/CPAN/* /usr/share/perl/5.20.2/CPAN/
```

Cant seem to find the perl modules in repo either: https://packages.debian.org/stable/perl/

Fourdee, 2016-08-17:
> http://dietpi.com/phpbb/viewtopic.php?f=12&t=440&p=2597#p2597

Gabba35, 2016-08-20:
I am also very interested in this and I have made a comment in that thread. I have made a small donation.

I recall reading something about not being able to use Perl 5.22 on the Odroid C2.

Fourdee, 2016-08-20:
#### Notes: Attempt 1:
apt-get install missing perl modules, using this package list: https://packages.debian.org/jessie/perl/

:u6307: XML::Parser::Expat
```
apt-get install -y libxml-parser-perl
```

:u55b6: HTML::Parser | Already installed
```
apt-get install -y libhtml-parser-perl
```

:u55b6: DBI |  installs, no effect
```
apt-get install -y libdbi-perl
```

:u6307: JSON::XS
```
apt-get install -y libjson-xs-perl 
```

:u55b6: Sub::Name | Installs, no effect
```
apt-get install -y libsub-name-perl
```

:u5408: YAML::XS | No package match for yaml-xs. Also no effect with
```
apt-get install -y libyaml-perl
```

:u5408: Digest::SHA1 | No package match, no effect with below
```
apt-get install -y libdigest-perl libdigest-perl-md5-perl
```

:u55b6: EV | Installs, no effect
```
apt-get install -y libev-perl
```

Leaves us with:
```
root@DietPi:~# squeezeboxserver
The following modules failed to load: DBI EV HTML::Parser Digest::SHA1 YAML::XS Sub::Name
```

Fourdee, 2016-08-20:
#### Notes:
Following instructions from binary:
```
If you're running some unsupported Linux/Unix platform, please use the buildme.sh
script located here:

https://github.com/Logitech/slimserver-vendor/tree/public/7.9/CPAN
```

Fourdee, 2016-08-20:
#### Notes:
Build CPAN modules:
```
apt-get install -y cpanminus build-essential yasm nasm
cpanm DBI EV XML::Parser::Expat HTML::Parser JSON::XS Digest::SHA1 YAML::XS Sub::Name

24 distributions installed

root@DietPi:~# squeezeboxserver
The following modules failed to load: DBI EV XML::Parser::Expat HTML::Parser JSON::XS Sub::Name
```

nts, check paths

Fourdee, 2016-08-22:
Finally getting somewhere:
[image]

Fourdee, 2016-08-22:
Current issues:
- Music scan does not work, scan does not return any results.
```
[16-08-22 13:02:04.3764] Slim::Utils::Misc::msg (1233) Warning: [13:02:04.3758] Can't locate auto/IO/Interface/autosplit.ix:   ./auto/IO/Interface/autosplit.ix: Permission denied at /usr/share/perl/5.20/AutoLoader.pm line 181.
 at /usr/share/squeezeboxserver/CPAN/IO/Interface.pm line 12.
JSON::XS object version 3.02 does not match bootstrap parameter 2.34 at /usr/lib/aarch64-linux-gnu/perl/5.20/DynaLoader.pm line 210.
Compilation failed in require at /usr/share/squeezeboxserver/CPAN/JSON/XS/VersionOneAndTwo.pm line 5.
BEGIN failed--compilation aborted at /usr/share/squeezeboxserver/CPAN/JSON/XS/VersionOneAndTwo.pm line 5.
Compilation failed in require at /usr/share/perl5/Slim/Utils/Prefs/Base.pm line 23.
BEGIN failed--compilation aborted at /usr/share/perl5/Slim/Utils/Prefs/Base.pm line 23.
Compilation failed in require at /usr/share/perl/5.20/base.pm line 100.
        ...propagated at /usr/share/perl/5.20/base.pm line 110.
BEGIN failed--compilation aborted at /usr/share/perl5/Slim/Utils/Prefs/Namespace.pm line 23.
Compilation failed in require at /usr/share/perl5/Slim/Utils/Prefs.pm line 79.
BEGIN failed--compilation aborted at /usr/share/perl5/Slim/Utils/Prefs.pm line 79.
Compilation failed in require at /usr/sbin/squeezeboxserver-scanner line 77.
BEGIN failed--compilation aborted at /usr/sbin/squeezeboxserver-scanner line 77.
JSON::XS object version 3.02 does not match bootstrap parameter 2.34 at /usr/lib/aarch64-linux-gnu/perl/5.20/DynaLoader.pm line 210.
Compilation failed in require at /usr/share/squeezeboxserver/CPAN/JSON/XS/VersionOneAndTwo.pm line 5.
BEGIN failed--compilation aborted at /usr/share/squeezeboxserver/CPAN/JSON/XS/VersionOneAndTwo.pm line 5.
Compilation failed in require at /usr/share/perl5/Slim/Utils/Prefs/Base.pm line 23.
BEGIN failed--compilation aborted at /usr/share/perl5/Slim/Utils/Prefs/Base.pm line 23.
Compilation failed in require at /usr/share/perl/5.20/base.pm line 100.
        ...propagated at /usr/share/perl/5.20/base.pm line 110.
BEGIN failed--compilation aborted at /usr/share/perl5/Slim/Utils/Prefs/Namespace.pm line 23.
Compilation failed in require at /usr/share/perl5/Slim/Utils/Prefs.pm line 79.
BEGIN failed--compilation aborted at /usr/share/perl5/Slim/Utils/Prefs.pm line 79.
Compilation failed in require at /usr/sbin/squeezeboxserver-scanner line 77.
BEGIN failed--compilation aborted at /usr/sbin/squeezeboxserver-scanner line 77.
```

Fourdee, 2016-08-22:
@Gabba35 
Looking good, should make v130 release:
[image]

Gabba35, 2016-08-23:
Thanks a lot Fourdee! Looks like you've cracked it! Seems like it wasn't too straight forward. What were the main issues (in fairly lay language!)?

Fourdee, 2016-08-23:
@Gabba35 

C2 LMS+Squeezelite working perfectly. Will be available in DietPi v130.

> What were the main issues

LMS uses Perl CPAN modules. As far as I can tell, LMS do not bundle the ARM64 binaries for these in their 7.9 deb package. Notice the ARM-hf (ARM32), i386 (32bit PC), x86_64 (64bit PC) but no ARM64.
[image]

So basically I needed to get those ARM64 binaries made.

LMS have a build script that should do this automatically (https://github.com/Logitech/slimserver-vendor/blob/public/7.9/CPAN/buildme.sh), however, it fails for ARM64: https://github.com/Fourdee/DietPi/issues/354#issuecomment-241210059

So, I ran ```squeezeboxserver``` and followed the errors to manually build each Perl CPAN module. Took a few days on and off (LMS likes specific versions of the CPAN modules). I've zipped up the binaries and hosted on DietPi. So when users install LMS through DietPi in v130, no cpan complies are needed.

Fourdee, 2016-08-23:
Note to self:
Test Pine A64

Fourdee, 2016-08-24:
Completed. LMS+Squeezelite Available in v130 for ARM64 devices.

closed 2016-08-24 by Fourdee

Gabba35, 2016-08-25:
Awesome, thanks for the explanation. Putting it all together would have been way beyond what I am capable of. Thanks so much for getting it working!

Fourdee, 2016-08-25:
@Gabba35 
We still have a bit to do on v130 before its released: https://github.com/Fourdee/DietPi/milestone/31

In the mean time, you can do a install of LMS by using the testing branch: https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md#steps-to-use-the-dietpi-testing-branch
I'd recommend using a different SDcard if you use the testing branch.