MichaIng/DietPi #356: DietPi-System | Default ARM Temp Limit - Thermal Throttling RPI-3

k-plan, 2016-05-27:
> http://dietpi.com/phpbb/viewtopic.php?f=11&t=439#p1923
[image]

Fourdee, 2016-05-27:
@k-plan 
Thanks :+1: 

I seen the results from your 70'c limit test. I'll run some today at 80'c (the official broadcom limit) and see if the cpu freqs are higher than 70'c. Just wondering if there is any benefit of a higher thermal limit, or if the clocks are roughly the same.

k-plan, 2016-05-27:
@Fourdee 

> While the BCM2837 is tested to 125°C, operating continuously at 100°C is out of specification for the chip. The official operating temperature limit is 85°C, and as a result the Raspberry Pi should start to thermally throttle performance around 82°C.
http://makezine.com/2016/03/02/raspberry-pi-3-not-halt-catch-fire/

Fourdee, 2016-05-27:
@k-plan 
Thanks :+1: 

i created a simple stress test, based on your temp check code:
```
#CPU stress
# - start
for (( i=0; i<$(nproc --all); i++ )); do yes > /dev/null & done
# - monitor and kill on CTRL+C
while true; do vcgencmd measure_temp && vcgencmd measure_clock arm; sleep 1; done; killall yes
```

k-plan, 2016-05-27:
@Fourdee 

[image]

```
frequency(45)=600000000
temp=70.9'C
frequency(45)=600000000
temp=70.9'C
frequency(45)=597000000
temp=71.4'C
frequency(45)=600000000
temp=70.9'C
frequency(45)=599998000
temp=70.9'C
frequency(45)=600000000
temp=70.9'C
frequency(45)=600000000
```

💯

Fourdee, 2016-05-27:
#### 80'c thermal limit test:
- Start temp = 49'c
- throttles down after 16 seconds
- Levels off at: 870mhz~ @ 80'c after 39 seconds.

```
temp=49.4'C
frequency(45)=1200002000
 - Duration 1 Seconds
temp=54.8'C
frequency(45)=1200000000
 - Duration 2 Seconds
temp=58.5'C
frequency(45)=1200000000
 - Duration 3 Seconds
temp=61.2'C
frequency(45)=1200000000
 - Duration 4 Seconds
temp=63.4'C
frequency(45)=1200002000
 - Duration 5 Seconds
temp=65.5'C
frequency(45)=1200000000
 - Duration 6 Seconds
temp=66.6'C
frequency(45)=1200000000
 - Duration 7 Seconds
temp=68.2'C
frequency(45)=1200000000
 - Duration 8 Seconds
temp=69.8'C
frequency(45)=1200000000
 - Duration 9 Seconds
temp=70.9'C
frequency(45)=1200000000
 - Duration 10 Seconds
temp=72.0'C
frequency(45)=1200000000
 - Duration 11 Seconds
temp=73.1'C
frequency(45)=1200000000
 - Duration 12 Seconds
temp=73.1'C
frequency(45)=1200000000
 - Duration 13 Seconds
temp=74.1'C
frequency(45)=1200000000
 - Duration 14 Seconds
temp=75.2'C
frequency(45)=1200000000
 - Duration 15 Seconds
temp=75.2'C
frequency(45)=1199000000
 - Duration 16 Seconds
temp=76.3'C
frequency(45)=1114000000
 - Duration 17 Seconds
temp=76.3'C
frequency(45)=1070998000
 - Duration 18 Seconds
temp=76.3'C
frequency(45)=1072000000
 - Duration 19 Seconds
temp=76.8'C
frequency(45)=1034000000
 - Duration 20 Seconds
temp=76.8'C
frequency(45)=1022000000
 - Duration 21 Seconds
temp=76.8'C
frequency(45)=988000000
 - Duration 22 Seconds
temp=77.4'C
frequency(45)=974000000
 - Duration 23 Seconds
temp=77.4'C
frequency(45)=967000000
 - Duration 24 Seconds
temp=77.4'C
frequency(45)=965000000
 - Duration 25 Seconds
temp=77.4'C
frequency(45)=965000000
 - Duration 26 Seconds
temp=77.9'C
frequency(45)=946998000
 - Duration 27 Seconds
temp=77.4'C
frequency(45)=941000000
 - Duration 28 Seconds
temp=77.9'C
frequency(45)=929998000
 - Duration 29 Seconds
temp=77.9'C
frequency(45)=932000000
 - Duration 30 Seconds
temp=77.9'C
frequency(45)=940000000
 - Duration 31 Seconds
temp=78.4'C
frequency(45)=930000000
 - Duration 32 Seconds
temp=78.4'C
frequency(45)=905000000
 - Duration 33 Seconds
temp=77.9'C
frequency(45)=907002000
 - Duration 34 Seconds
temp=77.9'C
frequency(45)=891002000
 - Duration 35 Seconds
temp=77.9'C
frequency(45)=901000000
 - Duration 36 Seconds
temp=77.4'C
frequency(45)=893000000
 - Duration 37 Seconds
temp=78.4'C
frequency(45)=877998000
 - Duration 38 Seconds
temp=77.9'C
frequency(45)=878000000
 - Duration 39 Seconds
temp=78.4'C
frequency(45)=878000000
```

Fourdee, 2016-05-27:
#### 80'c thermal limit test:

Start temp = 49'c
throttles down after 10 seconds @ 71'c~
Levels off at: 787mhz~ @ 75'c~ after 33 seconds.

`temp=53.8'C
frequency(45)=1200126000
 - Duration 1 Seconds
temp=58.0'C
frequency(45)=1200000000
 - Duration 2 Seconds
temp=61.2'C
frequency(45)=1200000000
 - Duration 3 Seconds
temp=62.8'C
frequency(45)=1200000000
 - Duration 4 Seconds
temp=65.0'C
frequency(45)=1200000000
 - Duration 5 Seconds
temp=66.1'C
frequency(45)=1200000000
 - Duration 6 Seconds
temp=67.7'C
frequency(45)=1200000000
 - Duration 7 Seconds
temp=68.8'C
frequency(45)=1200000000
 - Duration 8 Seconds
temp=69.8'C
frequency(45)=1200000000
 - Duration 9 Seconds
temp=70.9'C
frequency(45)=1147000000
 - Duration 10 Seconds
temp=72.0'C
frequency(45)=1074000000
 - Duration 11 Seconds
temp=72.0'C
frequency(45)=1012000000
 - Duration 12 Seconds
temp=72.5'C
frequency(45)=973000000
 - Duration 13 Seconds
temp=73.1'C
frequency(45)=959000000
 - Duration 14 Seconds
temp=73.1'C
frequency(45)=920000000
 - Duration 15 Seconds
temp=73.1'C
frequency(45)=911002000
 - Duration 16 Seconds
temp=73.1'C
frequency(45)=899000000
 - Duration 17 Seconds
temp=73.1'C
frequency(45)=881000000
 - Duration 18 Seconds
temp=73.6'C
frequency(45)=880000000
 - Duration 19 Seconds
temp=73.6'C
frequency(45)=878000000
 - Duration 20 Seconds
temp=74.1'C
frequency(45)=870000000
 - Duration 21 Seconds
temp=73.6'C
frequency(45)=869000000
 - Duration 22 Seconds
temp=73.6'C
frequency(45)=836000000
 - Duration 23 Seconds
temp=74.1'C
frequency(45)=843000000
 - Duration 24 Seconds
temp=73.1'C
frequency(45)=841000000
 - Duration 25 Seconds
temp=73.6'C
frequency(45)=815000000
 - Duration 26 Seconds
temp=73.6'C
frequency(45)=820000000
 - Duration 27 Seconds
temp=74.1'C
frequency(45)=817000000
 - Duration 28 Seconds
temp=74.1'C
frequency(45)=806000000
 - Duration 29 Seconds
temp=74.1'C
frequency(45)=802000000
 - Duration 30 Seconds
temp=74.1'C
frequency(45)=815000000
 - Duration 31 Seconds
temp=74.1'C
frequency(45)=799000000
 - Duration 32 Seconds
temp=74.1'C
frequency(45)=787000000
 - Duration 33 Seconds
temp=75.2'C
frequency(45)=787000000
`

k-plan, 2016-05-27:
### 80°C thermal limit test:

- compress backup directory with Midnight Commander as tar.bz (set cpu cores on heavy load)

[image]

```
temp=76.8'C
frequency(45)=980000000
temp=77.4'C
frequency(45)=1001000000
temp=77.4'C
frequency(45)=977000000
temp=77.4'C
frequency(45)=985000000
temp=77.4'C
frequency(45)=974000000
temp=77.4'C
frequency(45)=958000000
temp=77.4'C
frequency(45)=962000000
temp=77.4'C
frequency(45)=957000000
temp=77.4'C
frequency(45)=962000000
temp=77.4'C
frequency(45)=960000000
```

> Just wondering if there is any benefit of a higher thermal limit, or if the clocks are roughly the same.

- you push it to the limit of a **RPi-2**
- you push it to the limit of the 5,1 V / 2,5A PSU
- more heat and more energy consume
- you can use it as heater ...
- you reducing the life of your device
- nice yellow box appears at the top right corner of your screen

Fourdee, 2016-05-27:
@k-plan 

> you can use it as heater ...

Lol. Yep, its one hot SBC, in a bad way hehe

Loving the smell of burning components coming from my RPi 3.

I think 75'c is a good value. Considering the internal temp sensor is off by upto 17'c (http://makezine.com/2016/03/02/raspberry-pi-3-not-halt-catch-fire/)

> not only was Gareth measuring temperatures in excess of 100°C (212°F) under load for the BCM2837, but also measuring an offset of a full 17°C between his own measurements and the readings generated by the internal temperature sensor on the chip.

@k-plan Do you agree with 75'c? a good middle ground? We will only apply this to RPi3 devices in DietPi as default.

k-plan, 2016-05-27:
@Fourdee 

> Do you agree with 75'c? a good middle ground? 

**ONLY** if you have mounted a really good (copper) **heatsink** with good **thermally conductive paste** and not the crappy self-adhesive pads ...

>We will only apply this to RPi3 devices in DietPi as default.

Will you as well modify "_reducing the life of your device_" warning in  `cpu` ?

- 70°C will be moderate - but with barely effects.

- 80°C will push it to the limits of a RPi2 (don't know, if this has a thermal throttling)

Hmm, really complicated ...

Fourdee, 2016-05-27:
@k-plan 

> What does Rasbian do by default? 85°C ? (Without any heatsink?)

No idea, not installed the official Raspbian for a while lol ;)

> 70°C will be moderate - but with barely effects.

Yep i agree, should allow for a reduction in the onscreen warnings, whilst preventing sustained heat damage to the device.
I think 70'c should be for RPi3 default, it needs it. In my tests, RPi1/2 wont reach 65'c unless massively overclocked.

> Will you as well modify "reducing the life of your device" warning in cpu ?

Yep, will do, good idea :+1:

k-plan, 2016-05-28:
@Fourdee 
after sleep a night over this question, I personally think 80°C will be a good value.
My test are not based on synthetic benchmarks, its a multi-core packer, in step with actual practice.
But yes, not really the  core purpose of a SBC.

- 70° C - not enough performance (throttling down to 600 MHz - RED) and yellow warning comes up quick.
- 75° C - the same - no barely effects. Yellow and RED warning take some more time.
- 80° C - yes, yellow warning comes up, but no RED. Maximum are temp=77.4'C frequency=960000000, 	sufficient performance and frequency. 7,5° C offset from the official operating temperature limit of 85° C. 

If you don't have a good heatsink on cpu, it will a good deal worse.

When warranty time is expired, you can easy lower the default value. I don't like the thought of broil users cpu ...

I believe, @rhkean  as well has a RPi-3. Would like to hear what he is thinking of it and which experience he has.

k-plan, 2016-05-28:
### new: 80°C thermal limit test:

* new: cheap alu heatsink with cheap white thermally conductive paste

* compress backup directory with Midnight Commander as tar.bz (set cpu cores on heavy load)

* Start Temp.: 41.3° C

*  Runtime: after 15 min thermal throttling take place. (ligth grey/yellow warning comes up)

```
temp=75.8'C
frequency(45)=1130000000
temp=76.3'C
frequency(45)=1139000000
temp=75.8'C
frequency(45)=1153000000
temp=76.3'C
frequency(45)=1147000000
temp=75.2'C
frequency(45)=1129000000
temp=75.8'C
frequency(45)=1102000000
temp=76.3'C
frequency(45)=1115000000
temp=75.8'C
frequency(45)=1118000000
temp=75.8'C
frequency(45)=1149000000
temp=75.8'C
frequency(45)=1155000000
temp=75.8'C
frequency(45)=1140000000
temp=76.3'C
frequency(45)=1120000000
temp=75.8'C
frequency(45)=1132000000
temp=75.8'C
frequency(45)=1121000000
temp=76.3'C
frequency(45)=1114000000
temp=75.8'C
frequency(45)=1122000000
temp=75.8'C
frequency(45)=1126000000
temp=76.3'C
frequency(45)=1114000000
temp=75.2'C
frequency(45)=1124000000
temp=76.3'C
frequency(45)=1142000000
temp=75.2'C
frequency(45)=1142000000
temp=75.8'C
frequency(45)=1120000000
temp=75.8'C
frequency(45)=1137000000
temp=75.2'C
frequency(45)=1138000000
```

* max. Temp.:  76.3° C - min. Freq.: 1100 MHz - load average: 1.76

rhkean, 2016-05-29:
I only have an RPi 2.  Sorry

xenfomation, 2016-05-29:
Same here... all the glory... no pay ;-)

Fourdee, 2016-06-02:
@k-plan 

So from what I've can read, the internal temp has roughly a 10'c offset to the actual external temp. So, 85'c is the official thermal limit of the SoC. Therefor, 75'c thermal limit should be acceptable. 

I'am still not personally comfortable with > 65'c. But lets go for **75'c limit on the RPi 3**, it clearly needs it.

@xenfomation 
> Same here... all the glory... no pay ;-)

Join the club lol

Fourdee, 2016-06-02:
@k-plan 

Warning for high thermal limit temps:
```
							if (( $OPTION > 65 )); then
								whiptail --title "Warning: $OPTION 'c" --msgbox "Higher operating temperatures will reduce the life of your ARM SoC. Heat also dissipates through the PCB into other components, decreasing the lifespan of the whole device. Use at your own risk.\n\nDietPi recommends 65'c as a safe value (75'c for RPi 3).\n\nMore info: https://github.com/Fourdee/DietPi/issues/356" 14 70
							fi
```

Fourdee, 2016-06-02:
Done, patched for all RPi3 users. Will mark as closed.

closed 2016-06-02 by Fourdee