MichaIng/DietPi #357: DietPi-Software | GNUStep Desktop Environment

xenfomation, 2016-05-28:
While I am working on #151 and others, I thought I would add this as well.
Tested and good!
[image]

Fourdee, 2016-05-28:
@xenfomation 
Nice :+1: Although, GNUStep is a new one for me. Looks lighter than LXDE?

xenfomation, 2016-05-29:
It is lighter than LXDE and while I had looked at CDE, this was the closest Optimized Desktop I could think of.  GNUStep is a project based on Steve Job's NEXT project -- [https://en.wikipedia.org/wiki/GNUstep](https://en.wikipedia.org/wiki/GNUstep).

Before RedHat and other distros went mainstream, etc -- these were the first Desktops I recall being ported.  AIX still uses CDE, which felt more natural on a Sun machine, but indeed -- this is a wicked light version that almost had me thinking about adding a "Novelty" section, but I digress.

I am actually test that now with the following updates I will be sending a pull request for:

#363 - Stub | ViFM to Software Additional #363 DONE
#362 - DietPi-Banner | Proposal for stats #362  (N/A)
#353 - haproxy fails to start #353  (HIGHEST)
#361 - cli grouping of software?  (N/A)
#357 - DietPi-Software | GNUStep Desktop Environment #357  (IN PROGRESS)
#347 - No Machine & stock configs  DONE
#325 - DietPi Idea | Sourceable script with pre-defined color codes #325  (MEDIUM)
#309 - xenserver docs (HIGH)
#247 - DietPi-Software | Add Friendica: A Personal Social Networking Platform #247  (LOW)
#246 - NFS (LOW / CONFESSED PERSONAL PREFERENCE)
#245 - Full Disk Encryption #245 (LOW)
#271 - General | Add support for 2+ USB drives #271 (LOW)
#195 - DietPi-Software | Add QuiteRSS installation option #195  DONE
#191 - DietPi-Software | Add xrdp installation option #191  DONE
#151 - DietPi-Software | Add MATE desktop installation option #151 DONE 
#41  - [Wish List] Convert DietPi-Software into SQL database #41  (IN PROGRESS)

Thanks, @Fourdee !

xenfomation, 2016-05-29:
Re-tested #357 and #151 from test branch/v118:

- All looks good for both
- Added GNUStep under Software Optimized based on its lightweight purpose
- Working on VNC Server X dependency option: if VNC Server is selected, currently it will default to LXDE unless MATE or GNUStep are selected
- Installation includes: 

`x-window-system-core wmaker gnustep gnustep-devel gnustep-games libc-dbg upower policykit-1 iceweasel p7zip-full --no-install-recommends
`

[image]

Fourdee, 2016-06-04:
Additional implementation and testing for reference: https://github.com/Fourdee/DietPi/pull/364

Fourdee, 2016-06-08:
Documentation to finish: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=60#p2072

Fourdee, 2016-06-08:
Done.

Fourdee, 2016-06-08:
@xenfomation 
Is there a way to disable applications that are not installed, being listed?
[image]

xenfomation, 2016-06-13:
Sure!  i will make a config without those entries

Fourdee, 2016-06-30:
Taking this out of milestone. @xenfomation please let me know if you can still complete the config?

closed 2016-08-21 by Fourdee