MichaIng/DietPi #358: DietPi-Software | Add Minecraft Server

Fourdee, 2016-05-28:
> http://dietpi.com/phpbb/viewtopic.php?f=12&t=445

xenfomation, 2016-05-28:
My youngest son just went bonkers as he walked by me a few moments ago.  He says "Please and Thank You!" because I used to mod MineCraft ;-)  Family dogs, weird characters, ability to break blocks easier, etc. Looks like I'll have to code for him (or teach him a trade!)

Fourdee, 2016-06-04:
ARM64 compile didnt go well:
```
Scanning dependencies of target Cuberite
[ 86%] [ 87%] [ 87%] [ 87%] Building CXX object src/CMakeFiles/Cuberite.dir/Biom                                                                 eDef.cpp.o
Building CXX object src/CMakeFiles/Cuberite.dir/BlockID.cpp.o
Building CXX object src/CMakeFiles/Cuberite.dir/BlockInfo.cpp.o
Building CXX object src/CMakeFiles/Cuberite.dir/BlockArea.cpp.o
[ 87%] Building CXX object src/CMakeFiles/Cuberite.dir/BrewingRecipes.cpp.o
[ 88%] Building CXX object src/CMakeFiles/Cuberite.dir/Broadcaster.cpp.o
[ 88%] Building CXX object src/CMakeFiles/Cuberite.dir/BoundingBox.cpp.o
[ 88%] Building CXX object src/CMakeFiles/Cuberite.dir/ByteBuffer.cpp.o
/root/cuberite/src/ByteBuffer.cpp:44:3: error: #error Cannot determine endiannes                                                                 s of this platform
  #error Cannot determine endianness of this platform
   ^
[ 88%] Building CXX object src/CMakeFiles/Cuberite.dir/ChatColor.cpp.o
[ 88%] Building CXX object src/CMakeFiles/Cuberite.dir/Chunk.cpp.o
[ 89%] Building CXX object src/CMakeFiles/Cuberite.dir/ChunkData.cpp.o
src/CMakeFiles/Cuberite.dir/build.make:215: recipe for target 'src/CMakeFiles/Cu                                                                 berite.dir/ByteBuffer.cpp.o' failed
make[2]: *** [src/CMakeFiles/Cuberite.dir/ByteBuffer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:953: recipe for target 'src/CMakeFiles/Cuberite.dir/all' fa                                                                 iled
make[1]: *** [src/CMakeFiles/Cuberite.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2
```

Fourdee, 2016-06-04:
Cuberite is done.

Will take a look at Bukkit/Spigot next.

Cuberite is c++, Spigot is Java, this made me chuckle:
[image]

Zone22, 2016-06-05:
@Fourdee 

Might want to take a look at [MineOS](https://minecraft.codeemo.com/mineoswiki/index.php?title=Main_Page)

It's very easy to deploy. I have been using for the past 7 years. It's lightweight and handles nearly all minecraft modding platforms.

Fourdee, 2016-06-07:
https://minecraft.codeemo.com/mineoswiki/index.php?title=MineOS-node_(apt-get)

Zone22, 2016-06-07:
Here's the main [website](https://minecraft.codeemo.com/) which has a short description of features and as screenshots.

In the event that anyone needs support adapting MineOS to DietPi the author Hexparrot en.Will Dizon response usually in about 24 hours [here](https://discourse.codeemo.com/). Hexparrot is extremely open and helpful so you're free to ask anything.

Fourdee, 2016-06-07:
Notes:

Worked straight after install, but after a reboot:
```
root@DietPi:/usr/games/minecraft# supervisorctl start mineos
mineos: ERROR (no such file)
```

Fourdee, 2016-06-07:
Odroid C2, arm64:
[image]

Fourdee, 2016-06-07:
Note:
DietPi install will use OracleJava instead of OpenJRE for maximum performance.

Zone22, 2016-06-07:
@Fourdee 
You want to recommend as you do with web servers such as Owncloud to run it off an external medium instead of the DS card. Running a minecraft server off of a SD/thumb drive would lead to very poor performance and SD/thumb failure. I would go as far to say a ssd/hard drive is required for hosting minecraft.

Fourdee, 2016-06-07:
@Zone22 
Thanks for the info. Any ideas where the read/writes are to? If not, I'll try and contact Hexparrot.

Issues with the RPi install (NODEJS from apt-get). Lower version of NODEJS than expected, web service also fails to run:
```
pm WARN engine angular-translate@2.11.0: wanted: {"node":">=4.3"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine code@1.5.0: wanted: {"node":">=0.10.32"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine lab@5.18.1: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine hawk@3.1.3: wanted: {"node":">=0.10.32"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine address@1.0.0: wanted: {"node":">= 0.12.0"} (current: {"node":"0.10.29","npm":"1.4.21"})
```

Looks like I'l need to compile NODEJS for all devices, not just arm64 currently.

Zone22, 2016-06-07:
@Fourdee 
The majority of the the of the read/writes takes place in the following directory.
'/var/games/minecraft'
That directory will contain all servers instances created along with server files, archives, backups and so on. I would redirect the folder 'minecraft' to an external medium.

Extra information if you need it.
/var/games/minecraft/servers/_myserver_, where _myserver_ is anything end-user chooses. The resulting structure should look like:
   .
   |-archive
   |---_myserver_
   |-backup
   |---_myserver_
   |-import
   |-profiles
   |---vanilla172
   |-servers
   |---_myserver_
   |-----logs
   |-----world
   |-------DIM-1
   |-------DIM1
   |-------data
   |-------players
   |-------region

Fourdee, 2016-06-08:
@Zone22 
I'll probably move ```/var/games/minecraft``` to ```/mnt/dietpi_userdata/mineos```. If user has a dedicated USB drive, it will be stored there automatically via symlink.

Heres a working install script for node: https://github.com/taaem/nodejs-linux-installer
I had to add support for arm64 and made a pull request: https://github.com/taaem/nodejs-linux-installer/pull/8

Fourdee, 2016-06-08:
Completed. Available for all devices.

doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=2069#p2069

@Zone22 
Tested on C2 and RPi3, however, I dont own a copy of Minecraft to fully test it. Would you be able to test the installation with a copy of Minecraft, using the DietPi testing branch (https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md)?

Zone22, 2016-06-08:
@Fourdee 
Testing on Pine64 as it's only available hardware I have access to at the moment.
There are three issues.
One. I can't start a server instance.
[image]
MineOS.log
```
{"level":"info","message":"[Test] Discovered server","timestamp":"2016-06-08T15:39:57.242Z"}
{"level":"error","message":"[Test] Create tail on logs/latest.log failed","timestamp":"2016-06-08T15:39:57.333Z"}
{"level":"info","message":"[Test] Watching for file generation: logs/latest.log","timestamp":"2016-06-08T15:39:57.336Z"}
{"level":"error","message":"[Test] Create tail on server.log failed","timestamp":"2016-06-08T15:39:57.426Z"}
{"level":"info","message":"[Test] Watching for file generation: server.log","timestamp":"2016-06-08T15:39:57.427Z"}
{"level":"error","message":"[Test] Create tail on proxy.log.0 failed","timestamp":"2016-06-08T15:39:57.434Z"}
{"level":"info","message":"[Test] Watching for file generation: proxy.log.0","timestamp":"2016-06-08T15:39:57.435Z"}
{"level":"info","message":"Starting up server, using commit: c2cff9d removed pocketmine profiles\n","timestamp":"2016-06-08T15:39:58.306Z"}
{"level":"info","message":"[Test] eula.txt detected: ACCEPTED (eula=true)","timestamp":"2016-06-08T15:39:58.448Z"}
{"level":"info","message":"[Test] autostart = false","timestamp":"2016-06-08T15:40:02.889Z"}
{"level":"error","message":"[Test] Aborted server startup; condition not met: true","timestamp":"2016-06-08T15:40:02.894Z"}
{"level":"info","message":"[WEBUI] root connected from 192.168.1.122","timestamp":"2016-06-08T15:41:59.386Z"}
{"level":"info","message":"[WEBUI] Downloading official profiles.","timestamp":"2016-06-08T15:42:01.020Z"}
{"level":"info","message":"[Test] root (192.168.1.122) joined server namespace","timestamp":"2016-06-08T15:42:02.002Z"}
{"level":"info","message":"[Test] 192.168.1.122 issued command : \"start\"","timestamp":"2016-06-08T15:42:10.483Z"}
{"level":"info","message":"[Test] received request \"start\"","timestamp":"2016-06-08T15:42:10.537Z"}
{"command":"start","uuid":"903cb650-2d8f-11e6-978b-959104acd134","time_initiated":1465400530486,"success":false,"err":127,"time_resolved":1465400530736,"level":"error","message":"[Test] command \"start\" errored out:","timestamp":"2016-06-08T15:42:10.747Z"}
```
Two. I believe the wrong folder was redirected to the external Medium. 
Currently...
[image]
Should be...
[image]
Three. A standard user needs to be created during the initial install as a web UI should never be managed through root due to file permission issues. - [Source](https://minecraft.codeemo.com/mineoswiki/index.php?title=MineOS-node_%28apt-get%29) I would recommend the default user of 'mc' With a password of 'dietpi'

Fourdee, 2016-06-08:
@Zone22 
Thanks for the test

> Testing on Pine64 as it's only available hardware I have access to at the moment
One. I can't start a server instance.

I just did a fresh install on C2 to verify. Working fine. 
Did you accept the EULA? Required to start server.
Failing that, must be an issue limited to the Pine64 image. I'am still waiting for mine to turn up, so not much I can do my end.

> Two. I believe the wrong folder was redirected to the external Medium. 

Well spotted :+1: . I'll move and symlink /var/games/minecraft aswell.

> Three. A standard user needs to be created during the initial install as a web UI should never be managed through root due to file permission issues. - Source I would recommend the default user of 'mc' With a password of 'dietpi'

I'll look into it.

Fourdee, 2016-06-08:
```/var/games/minecraft``` is now symlinked to ```/mnt/dietpi_userdata/mineos/serverdata```

And, user is now ```mineos```

Fourdee, 2016-06-11:
I'll mark this as closed.
MineOS and Cuberite is enough Minecraft servers for now :) If users want to see more choices, please make a git ticket for the request.

closed 2016-06-11 by Fourdee