MichaIng/DietPi #362: DietPi-Banner | Proposal for stats

xenfomation, 2016-05-29:
Use case is:
- I use various locales, specifically en:US|UK|DE
- I use various dev machines with static IPs
- Just a proposal for how to go about adding useful login/head information?
- Displays IPv4/6 information along with hostname, locale, and uptime -s

[image]

Attached is the altered code as this is - IMHO - not valid for any commit, but for review of @Fourdee to consider.

Cheers!
[dietpi-banner.txt](https://github.com/Fourdee/DietPi/files/288080/dietpi-banner.txt)

Fourdee, 2016-05-29:
Yep, i think we can add some of these in. Although the banner would need cleaning up (move stuff around). I'd probably go for:
- Uptime
- IPv4
- IPv6

The items i think might be useful, but not necessary:
- Hostname

The items i personally feel shouldnt be there:
- Locale. Maybe we could scrape the current locale and add that to ```dietpi-config``` language menu instead?

Fourdee, 2016-05-29:
Just an idea, maybe we could do a ```dietpi-sysinfo``` script. Similar to cpu  ```dietpi-cpuinfo```? Instead of filling up the banner?

xenfomation, 2016-05-29:
I completely agree, @Fourdee -- I hated filling up the screen, especially when using SSH/Terminal access.  Obviously I am cooking with NASA rocket fuel this weekend, so I think I can definitely cook something up -- even something with a webhook so one could poke, say:

http://myhost.mydomain:1234
(prompts for authentication and that is the only stateful bit!)

Prototypes will be sent your way after I look at the backlog I have :-)

Fourdee, 2016-07-23:
I want to keep the banner lightweight and clean as possible. We currently have the IP address in there. Adding more stats means more processing and more mess.

We also have dietpi-cloudshell stats for those users who want it: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p204

Marking as closed. Please reopen if required.

closed 2016-07-23 by Fourdee