MichaIng/DietPi #363: Stub | ViFM to Software Additional

xenfomation, 2016-05-29:
Per work on #295 I had left out ViFM as a command-line file manager.
In testing my backlog, I have created this ticket to track the original ticket and additional software that was intended to be added.

xenfomation, 2016-05-29:
Per work on #151 - testing addition of ViFM for CLI-based file manager
[image]

xenfomation, 2016-05-29:
@Fourdee -- for tickets we have arranged the commit -> pull request, do I need to assign them back to you for testing, etc?  Just want to ensure I am following protocol

Cheers!

Fourdee, 2016-05-29:
@xenfomation 
Just send a pull request, make a note of whats been changed/added, then put a @fourdee in there. I'll then take a look :+1:

Fourdee, 2016-06-04:
Additional implementation and testing for reference: https://github.com/Fourdee/DietPi/pull/364

Fourdee, 2016-06-08:
Done. Marking as closed.

closed 2016-06-08 by Fourdee