MichaIng/DietPi #368: General | Support multiple WiFi SSID's and auto connect to one during boot.

Fourdee, 2016-05-30:
> http://dietpi.com/phpbb/viewtopic.php?f=9&t=451#p1974

Considerations:
- Allow for multiple wifi credentials in dietpi.txt
- Create a script during boot that does a ```iw scan```, match against our dietpi.txt list,  then update ```/etc/network/interfaces``` with the required info. Finally connect.

9H5G, 2017-03-12:
I'll be very happy when this is made available :) Please? :)

tatoosh, 2017-04-14:
must have feature ;)

jvdneste, 2018-04-09:
Wouldn't even need scanning, just failover. Connect first, if fail, connect second, etc. Though scanning would perform better, it's just an optimization.

Fourdee, 2018-07-02:
Note to self:
- Does wpa_supplicant.conf support multiple SSID entries and auto connect?

okoestner, 2018-08-14:
There should be some kind of file like dietpi.txt that will be used during boot to connect to any of the WiFis configured in that specific file. Not only during initial setup of the RasPi.

MichaIng, 2018-08-27:
@Fourdee 
Multiple entries in [wpa_supplicant.conf](https://linux.die.net/man/5/wpa_supplicant.conf) are possible.

We could allow adding `AUTO_SETUP_NET_WIFI2_SSID` and related other entries to `dietpi.txt` and add all of them from top to bottom to `wpa_supplicant.conf`, stopping the loop if `AUTO_SETUP_NET_WIFI${i}_SSID` is not set/available anymore. However this strictly needs DHCP client to be used for WiFi interface, since we cannot configure different interfaces (within /etc/network/interfaces) to use the same `wpa_supplicant.conf` but different networks inside (all of the arrays have the same name).
But, hmm we could configure different interfaces to use different `wpa_supplicant${i}.conf` files actually 🤔, if this works with just one WiFi adapter? Just the question what happens if two SSIDs are available at the same time. Two interfaces trying to configure the same adapter? ...

Maybe going with a single interface for now is simpler:
- Allowing to add multiple WiFi SSIDs + credentials to `dietpi.txt` with ascending variable numbers.
- We loop through available variables and add them as separate `network={` entries to `wpa_supplicant.conf`.
- We then rely on "`wpa_supplicant will automatically select the best network based on the order of network blocks in the configuration file, network security level (WPA/WPA2 is preferred), and signal strength.`": https://linux.die.net/man/5/wpa_supplicant.conf

To not allow messing up with `dietpi-config`, we need to change the WiFi adapter menu:
- Add new menu entry to add an additional SSID with credentials. By default scan and select, otherwise allow manual setup as already handled now.
- Add one as well for SSID removal, at least if it is not the last available configured SSID.
- Perhaps show and allow to adjust really all entries (key-mgr, identity, cert, eap, ...), but pre-fill the fields with "auto" which in case we do not save to dietpi.txt/wpa_supplicant.conf then?
- If multiple SSIDs are found, show a cascaded WiFi menu to select SSID first. Hide STATIC/DHCP switch then, instead force DHCP to avoid the need to separate interfaces.

Fourdee, 2018-09-15:
Notes:
- PREP copy /var/lib/dietpi/dietpi-wifi.db to /boot/
- Check automated WiFi setup
- Disable multiple SSID's when static
- Option to remove existing entries in dietpi-config
- More testing

Fourdee, 2018-09-16:
Note to self:
- https://dietpi.com/phpbb/viewtopic.php?f=8&t=9&p=9#p9

Optional (Wifi): section:
- Does not require updating, until we update any image. Existing images will still use the older 1 SSID system.

Fourdee, 2018-09-16:
Automated works, however, ```dietpi-config``` afterwards results in:
[image]

Fourdee, 2018-09-18:
Notes:
```
# Wifi
allow-hotplug wlan0
iface wlan0 inet static
address 192.168.42.1
netmask 255.255.255.0
#gateway 192.168.0.1
wireless-power off
#dns-nameservers 8.8.8.8 8.8.4.4

# IP tables
up iptables-restore < /etc/iptables.ipv4.nat
```

Fourdee, 2018-09-18:
🈯️ All testing passed https://github.com/Fourdee/DietPi/issues/368#issuecomment-421574150

The only thing I cannot achieve, due to lack of available hardware for setting up a test network, is:
- Testing WPA-EAP with automated installation
- Implement WPA-EAP options in ```dietpi-config```

Completed.

closed 2018-09-18 by Fourdee