MichaIng/DietPi #371: Update all online documentation referencing user data locations to /mnt/dietpi_userdata

Fourdee, 2016-06-01:
> https://github.com/Fourdee/DietPi/issues/334#issuecomment-222503304

```/root```
```/mnt/usb1_1```

Is now replaced with ```/mnt/dietpi_userdata```

Nice big job for me lol.

k-plan, 2016-06-01:
@Fourdee 

> Nice big job for me lol.

another "nice job" will be to update  this: [DietPi-Config | Feature rich config tool for your device](http://dietpi.com/phpbb/viewtopic.php?f=8&t=63)

Sorry ...  :smile:

Fourdee, 2016-06-01:
@k-plan 
:+1: 
Updated :dancer: http://dietpi.com/phpbb/viewtopic.php?f=8&t=63

k-plan, 2016-06-02:
@Fourdee 

really good and very quick realisation. :smile: 
:100:  :+1: 

Thank you very much.

xenfomation, 2016-06-03:
Noted for #271

Fourdee, 2016-06-03:
Think i got all the references updated to ```/mnt/dietpi_userdata``` on forum docs. Will leave this open just in case @k-plan would kindly check :+1: :+1:  *wink wink*

k-plan, 2016-06-03:
@Fourdee 

> Will leave this open just in case @k-plan wants to kindly check 👍 👍 wink wink hehe

-----

http://dietpi.com/phpbb/viewtopic.php?p=55#p55

- In the address bar at the top type ftp://root:`raspberry`@192.168.0.100 press enter.

Accessing ProFTP with a FTP Client:
username = root
password = The same as your root login password. Default is `dietpi`
address = Your ip address (eg: 192.168.0.100)

------

http://dietpi.com/phpbb/viewtopic.php?p=68#p68

DietPi will automatically save/update log file data to disk every hour `/root/logfile_storage`.

-----------

http://dietpi.com/phpbb/viewtopic.php?p=188#p188

```
cd /root
./weaved_setup.bin

```
??

----------

http://dietpi.com/phpbb/viewtopic.php?p=1066#p1066

```
cd /root/wiringPi*
ls
```
??

-------

Fourdee, 2016-06-04:
@k-plan 

>  ftp://root:raspberry@192.168.0.100 

Good spot :+1: 

> DietPi will automatically save/update log file data to disk every hour /root/logfile_storage

Yep, thats right. DietPi-Ramlog mode 2

> cd /root
./weaved_setup.bin

Yep, required so user can enter their login creds for their weaved account.

> cd /root/wiringPi*
ls

Yep, this is where WiringPi is installed. It contains samples, examples the user can run. As well as docs.

k-plan, 2016-06-04:
@Fourdee 

sorry, so fare no time to test on a clean install v.119, but  ...

> 
```
/root
/mnt/usb1_1
```
Is now replaced with `/mnt/dietpi_userdata`

 . . .

>> DietPi will automatically save/update log file data to disk every hour `/root/logfile_storage`

> Yep, thats right. DietPi-Ramlog mode 2

In v119 now it will be:  `/mnt/dietpi_userdata/logfile_storage` ?

> 
```
cd /root
./weaved_setup.bin
```

will be:

```
cd /mnt/dietpi_userdata
./weaved_setup.bin
```

and:

```
cd /mnt/dietpi_userdata/wiringPi*
ls
```

... what will be the question. But I see, I have to test it be myself ... 😉 

😄

Fourdee, 2016-06-04:
> In v119 now it will be: /mnt/dietpi_userdata/logfile_storage ?

Nope. It will stay as /root/logfile_storage

> cd /mnt/dietpi_userdata/wiringPi*
ls

User data is classed as:
- bittorrent downloads
- owncloud storage, 
- Fileservers default location (music/video folders)
- etc etc

k-plan, 2016-06-04:
@Fourdee 

>User data is classed as:

- bittorrent downloads
- owncloud storage,
- Fileservers default location (music/video folders)
- etc etc

Aha ... that's what is the missing description. 😄 

So it means more: `/mnt/dietpi_user_application_data` 

Suggestion: so please write this description somewhere in http://dietpi.com/phpbb/viewforum.php?f=8

Worry that I will not be the only one, who don't understand it instantly.

Fourdee, 2016-06-05:
@k-plan 

> Suggestion: so please write this description somewhere in http://dietpi.com/phpbb/viewforum.php?f=8
Worry that I will not be the only one, who don't understand it instantly.

Yep, it needs to be explained a little better in the docs :+1:

Fourdee, 2016-06-06:
Further improvements required:
> http://dietpi.com/phpbb/viewtopic.php?f=9&t=470#p2049

Fourdee, 2016-06-07:
first run setup question updated:
[image]

Fourdee, 2016-06-11:
Ok, so, here goes :dancer: 

#### What is user data?
In DietPi, we class user data as:
- Data storage for applications. Some examples:
 - Ownclound data store.
 - BitTorrent downloads.
- The location where your fileserver choice will point to, if you install one (eg: Samba Server, Proftpd).
- The location where you can upload and store your media content, for other applications to use (eg: Kodi, Emby, Plex etc).

#### Where is my user data?
As of DietPi v120. For all future software you install in ```dietpi-software```, you can access all your user data with ```/mnt/dietpi_userdata```. Regardless of where the data is physically stored, a symlink will automatically be created for you if needed.

#### Can I transfer my user data to another location (eg: USB drive?)
Yes. 
Simply run ```dietpi-software``` and enter the "User data" menu option. Once there, simply select the location you wish to use, or select Manual for a custom location. You can also setup a dedicated USB drive, if one is not currently installed.
DietPi will also move your existing user data to your chosen location.

Fourdee, 2016-06-11:
@k-plan

Hope this covers user data. If not, please let me know :+1: 

http://dietpi.com/phpbb/viewtopic.php?f=8&t=478&p=2087

[image]

Fourdee, 2016-06-11:
@k-plan 
Also updated the new user data screen in dietpi-software with link:
[image]

Fourdee, 2016-06-11:
I'll mark this as completed for now. 

I would appreciate your feedback @k-plan for any improvements, if you have them.

closed 2016-06-11 by Fourdee