MichaIng/DietPi #372: Error: Obtained server version value from github is invalid. Please try running dietpi-update again.

squirrelboot, 2016-06-01:
Hello,

I am having the same issue as [here](https://github.com/Fourdee/DietPi/issues/331). 
I used the fix [here](https://github.com/Fourdee/DietPi/issues/301). 
However, **this did not work**. 

Outputs:
cat /DietPi/dietpi.txt 
Was blank.

cat /DietPi/dietpi/.version
115

cat /DietPi/dietpi/dietpi-update
http://pastebin.com/CEYfauMQ

Fourdee, 2016-06-02:
> cat /DietPi/dietpi.txt 
Was blank.

@squirrelboot 

dietpi.txt should not be blank. The file also contains the ```gitbranch``` var that ```dietpi-update``` will use. So lets get that file back.
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/master/dietpi.txt -O /DietPi/dietpi.txt
```

Now try running the update again.

xenfomation, 2016-06-03:
I have tested this myself, but without the same results:

[image]

[image]

WolfganP, 2016-08-10:
Just for statistics, it happened to me as well while on v126. dietpi.txt got corrupted somehow, the wget command above solved the issue and dietpi-update run without issues after.

Fourdee, 2016-08-11:
@WolfganP 
Thanks for the info :+1: 

Did you by any chance save the corrupted dietpi.txt? It may give us some clues as to what caused it to corrupt.