MichaIng/DietPi #373: DietPi-Software | ViFM  will not be marked as installed after installation

k-plan, 2016-06-04:
- install ViFM via `dietpi-software`

```
root@RPi-3-test:~# vifm -v
Version: 0.7.8
Git commit hash:
Compiled at: Nov  6 2014 07:07:04
```
- after reboot, it will not be marked as installed.
[image]

- but you can uninstall ViFM via `dietpi-uninstall`
[image]

Fourdee, 2016-06-04:
@k-plan 

Good spot :+1:  fixed: https://github.com/Fourdee/DietPi/commit/565dc91463f956bc1a1220f5b244007fab6e89ba

xenfomation, 2016-06-05:
Fast work, sir!  Thanks @k-plan & @Fourdee

k-plan, 2016-06-05:
@Fourdee 

> Good spot 👍 fixed: 565dc91

Testing:

```
 ───────────────────────────────────────
 DietPi     | 16:00 | Sun 05/06/16
 ───────────────────────────────────────
 V120       | RPi 3 Model B (armv7l)
 ───────────────────────────────────────
```

[image]
[image]

Works! 😃  👍

Fourdee, 2016-06-08:
Thanks @k-plan :+1: 

Marking this as closed :)

closed 2016-06-08 by Fourdee