MichaIng/DietPi #374: dietpi.txt  |  Update Optimized and Additional Linux Software for installation

k-plan, 2016-06-04:
@Fourdee 

please update 

`#DietPi-Optimized Software to automatically install | Requires AUTO_Install_Index=1`

and 

`#Linux Additional Software to automatically install | Requires AUTO_Install_Index=1`

in `dietpi.txt` , so we can do  AUTO_Install all the new Software since v115.

Fourdee, 2016-06-04:
@k-plan 

I keep the dietpi.txt file upto date on github. We no longer patch existing installation dietpi.txt automation entries.

Simply grab the latest dietpi.txt from https://github.com/Fourdee/DietPi/blob/master/dietpi.txt. It contains all the current and available program automation options.
Write image, copy to /boot/, enable your options, job done.

k-plan, 2016-06-04:
@Fourdee 

```
#Linux Additional Software to automatically install | Requires AUTO_Install_Index=1
#   1=install
AUTO_DietpiSoftware_Install_VIMFULL=0
AUTO_DietpiSoftware_Install_SSHCLIENT=0
# - This will be automatically mounted to /mnt/samba. Use DietPi-Config settings section in this file to enter your details (SambaClient_*)
AUTO_DietpiSoftware_Install_SMBCLIENT=0
# - This will be automatically mounted to /mnt/ftp_client. Use DietPi-Config settings section in this file to enter your details (curlftpfs_*)
AUTO_DietpiSoftware_Install_CURLFTPFS=0
AUTO_DietpiSoftware_Install_GITCLIENT=0
AUTO_DietpiSoftware_Install_BUILDESSENTIAL=0
AUTO_DietpiSoftware_Install_ALSABASE=0
AUTO_DietpiSoftware_Install_XSERVERXORG=0
AUTO_DietpiSoftware_Install_FFMPEG=0
AUTO_DietpiSoftware_Install_ORACLEJAVA=0
AUTO_DietpiSoftware_Install_NODEJS=0
AUTO_DietpiSoftware_Install_LOGGING_RSYSLOG=0
```

No mc, tcpdump, iperf, MTR-Tiny ... and so on

As far as I can remember, you will do a re-image for all devices in v.120. Will be nice to see it in	there .

> Simply grab the latest dietpi.txt from https://github.com/Fourdee/DietPi/blob/master/dietpi.txt. It contains all the current and available program automation options.
Write image, copy to /boot/, enable your options, job done.

Nice hint, thank you very much. 👍 

But didn't help in the moment for mc, iperf, tcpdump, mtry-tiny.

Fourdee, 2016-06-04:
@k-plan 

> mc, iperf, tcpdump, mtry-tiny

Cheers, I'll take a look

xenfomation, 2016-06-04:
@Fourdee  @k-plan 

I may be jumping in where I shouldn't have been, but I remember one of my first tasks was to help add these:

mc, iperf, tcpdump, mtr, vim, vim-timy, etc

I just wanted to see if these are appearing in the builds or if I need to revisit dietpi-software as I have the following as of build 118/119 (committed, etc):

### Network Utilities```

 [ ]            ────Network Tools─────────────────────────────▒   │                
                │    [ ] iftop      Displays bandwidth usage information.         ▒   │                
                │    [ ] IPTraf     Interactive Colorful IP LAN Monitor.          ▒   │                
                │    [*] Iperf      Internet Protocol bandwidth measuring tool.   ▒   │                
                │    [*] MTR-Tiny   Full screen ncurses traceroute tool.          ▒   │                
                │    [*] nLoad      Realtime console network usage monitor.       ▒   │                
                │    [ ] tcpdump    Command-line network traffic analyzer.        ▒   │    
```

### MC Editors Etc
```
 [ ]            ────Editors────────────────────────────────── ▒   │                
                │    [ ] Emacs      GNU Emacs editor.                             ▒   │                
                │    [ ] Jed        Editor for programmers.                       ▒   │                
                │    [*] Vim        Vi Enhanced Text Editor.                      ▒   │                
                │    [ ] Vim-Tiny   Compact release of Vim.                       ▒   │                
```

Point is -- to both of you, if I missed adding any utilities -- please let me know and I will adjust accordingly.  

The original list of software (CLI or Otherwise) was 60+ tiny, useful things we all used.  I do remember in talking with @rhkean that I tripped down to "test Vim-Tiny" as I was getting familiar with the dietpi-software codes and flags.

Let me know, @Fourdee, if I need to go back in and add, check flags, or clean-up code for:

### Network tools 
telnet
iperf (really iperf3)
traceroute & tracepath
whois / nslookup
nmap
rfkill
mtr/mtr-tiny

Utilites:
mkisofs
makepassword
openssl utils
hdparm
dtrx (compression utility)

... much more, but then again -- just seeing if I missed anything.

xenfomation, 2016-06-04:
I have no idea how I closed this ticket?  AUGH.

SO -- 

@Fourdee @k-plan

I may be jumping in where I shouldn't have been, but I remember one of my first tasks was to help add these:

mc, iperf, tcpdump, mtr, vim, vim-timy, etc

I just wanted to see if these are appearing in the builds or if I need to revisit dietpi-software as I have the following as of build 118/119 (committed, etc):

###Network Utilities
```
[ ] ────Network Tools─────────────────────────────▒ │

│ [ ] iftop Displays bandwidth usage information. ▒ │

│ [ ] IPTraf Interactive Colorful IP LAN Monitor. ▒ │

│ [] Iperf Internet Protocol bandwidth measuring tool. ▒ │

│ [] MTR-Tiny Full screen ncurses traceroute tool. ▒ │

│ [*] nLoad Realtime console network usage monitor. ▒ │

│ [ ] tcpdump Command-line network traffic analyzer. ▒ │
```

### MC Editors Etc
```

[ ] ────Editors────────────────────────────────── ▒ │

│ [ ] Emacs GNU Emacs editor. ▒ │

│ [ ] Jed Editor for programmers. ▒ │

│ [*] Vim Vi Enhanced Text Editor. ▒ │

│ [ ] Vim-Tiny Compact release of Vim. ▒ │

```

Point is -- to both of you, if I missed adding any utilities -- please let me know and I will adjust accordingly.  

The original list of software (CLI or Otherwise) was 60+ tiny, useful things we all used.  I do remember in talking with @rhkean that I tripped down to "test Vim-Tiny" as I was getting familiar with the dietpi-software codes and flags.

Let me know, @Fourdee, if I need to go back in and add, check flags, or clean-up code for:

### Network tools 
telnet
iperf (really iperf3)
traceroute & tracepath
whois / nslookup
nmap
rfkill
mtr/mtr-tiny

Utilites:
mkisofs
makepassword
openssl utils
hdparm
dtrx (compression utility)

... much more, but then again -- just seeing if I missed anything.

Fourdee, 2016-06-05:
Ok, updated the missing dietpi.txt automation entries ^^.

@xenfomation 

> Let me know, @Fourdee, if I need to go back in and add, check flags, or clean-up code for:

hdparm is installed and configured on DietPi by default.

As for the rest, you can check the current list here:
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L7882-L7915. Then you can choose if you want to add any more in.

k-plan, 2016-06-05:
@Fourdee 

>  Ok, updated the missing dietpi.txt automation entries ^^.

### Testing:
 
```
root@Rpi_3-TEST:~# cat /boot/dietpi.txt
 . . .
 #Automated Software Installations:
#       0=Disabled
#       1=DietPi-Automation (use software lists below to select the software you require)
AUTO_Install_Index=1
 . . .
#Linux Additional Software to automatically install | Requires AUTO_Install_Index=1
#   1=install
AUTO_DietpiSoftware_Install_VIMFULL=0
AUTO_DietpiSoftware_Install_SSHCLIENT=0
# - This will be automatically mounted to /mnt/samba. Use DietPi-Config settings section in this file to enter your details (SambaClient_*)
AUTO_DietpiSoftware_Install_SMBCLIENT=0
# - This will be automatically mounted to /mnt/ftp_client. Use DietPi-Config settings section in this file to enter your details (curlftpfs_*)
AUTO_DietpiSoftware_Install_CURLFTPFS=0
AUTO_DietpiSoftware_Install_GITCLIENT=0
AUTO_DietpiSoftware_Install_BUILDESSENTIAL=0
AUTO_DietpiSoftware_Install_ALSABASE=0
AUTO_DietpiSoftware_Install_XSERVERXORG=0
AUTO_DietpiSoftware_Install_FFMPEG=0
AUTO_DietpiSoftware_Install_ORACLEJAVA=0
AUTO_DietpiSoftware_Install_NODEJS=0
AUTO_DietpiSoftware_Install_VIFM=0
AUTO_DietpiSoftware_Install_QUITERSS=0
AUTO_DietpiSoftware_Install_GNUEMACS=0
AUTO_DietpiSoftware_Install_JED=1
AUTO_DietpiSoftware_Install_VIMTINY=0
AUTO_DietpiSoftware_Install_MC=1
AUTO_DietpiSoftware_Install_IFTOP=0
AUTO_DietpiSoftware_Install_IPTRAF=0
AUTO_DietpiSoftware_Install_IPERF=1
AUTO_DietpiSoftware_Install_MTRTINY=1
AUTO_DietpiSoftware_Install_NLOAD=0
AUTO_DietpiSoftware_Install_TCPDUMP=1
AUTO_DietpiSoftware_Install_LOGGING_RSYSLOG=0
```

```
root@Rpi_3-TEST:~# cat /root/DietPi-Automation.log
cat: /root/DietPi-Automation.log: No such file or directory
```

```
root@Rpi_3-TEST:~# iperf
-bash: iperf: command not found

root@Rpi_3-TEST:~# mc
-bash: mc: command not found

root@Rpi_3-TEST:~# tcpdump -help
tcpdump version 4.6.2
libpcap version 1.6.2
OpenSSL 1.0.1k 8 Jan 2015

root@Rpi_3-TEST:~# mtr --v
mtr 0.85

```
 
[image]
[image]
[image]

---

@xenfomation 

> Point is -- to both of you, if I missed adding any utilities -- please let me know and I will adjust accordingly

I don`t miss anything. Only the option to automated installation your integrated tools in `dietpi.txt`.

Fourdee, 2016-06-07:
@k-plan 
Thanks for the test results :+1:  I'll take a look.

Fourdee, 2016-06-11:
@k-plan 
MC should be MIDNIGHTCOMMANDER.
iperf is missing the installation code in dietpi-software.

Fourdee, 2016-06-11:
@k-plan 
Fixed. I'll mark this as closed, please reopen if needed.

k-plan, 2016-06-12:
@Fourdee 

only tested with mc, iperf, tcpdump and mtr.

[image]

[image]

```
root@RPi-UniFi:~# mtr --v
mtr 0.85

```
Sorry, but mtr-tiny is missing in `dietpi-uninstall`

Fourdee, 2016-06-12:
@k-plan 
> Sorry, but mtr-tiny is missing in dietpi-uninstall

Thanks, i'll take a look. :+1:

Fourdee, 2016-06-13:
Fixed:
[image]

closed 2016-06-13 by Fourdee