MichaIng/DietPi #375: Netdata on clean standard install not working

Fourdee, 2016-06-04:
> http://dietpi.com/phpbb/viewtopic.php?f=11&t=466

Fourdee, 2016-06-04:
Tested on C2. Running fine. Will get more details from user

k-plan, 2016-06-05:
@Fourdee 

Tested on RPi 3 Model B (armv7l). Running, but in log:

```
root@Rpi_3-TEST:~# logread
 . . .
Jun  5 22:28:32 Rpi_3-TEST user.err netdata: Cannot open file '/etc/netdata/netdata.conf'
Jun  5 22:28:32 Rpi_3-TEST user.info apps.plugin: Found NETDATA_HOST_PREFIX=''
Jun  5 22:28:32 Rpi_3-TEST user.info apps.plugin: Found NETDATA_CONFIG_DIR='/etc/netdata'
Jun  5 22:28:32 Rpi_3-TEST user.info apps.plugin: starting...
Jun  5 22:28:33 Rpi_3-TEST user.err apps.plugin: Cannot process /proc/8102/stat
```

Fourdee, 2016-06-27:
No response from user. I will assume resolved as unable to replicate.

closed 2016-06-27 by Fourdee