MichaIng/DietPi #376: Diet-Pi Network | RPi3 build-in WiFi always try to connect while disabled

k-plan, 2016-06-05:
@Fourdee 

Description:

- RPi3 buid-in WiFi adapter always try to connect to WiFi AP while adapter is disabled in `dietpi-config`

[image]

[image]

[image]

```
root@Rpi_3-TEST:~# dmesg | grep brcmf
[   58.064918] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[   58.067106] brcmfmac: power management disabled
[   58.099530] brcmfmac: power management disabled
[   58.262574] brcmfmac: power management disabled
[   58.442766] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[   58.442781] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[   67.853994] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[   67.863335] brcmfmac: power management disabled
[   67.902551] brcmfmac: power management disabled
[   68.069905] brcmfmac: power management disabled
[   68.285730] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[   68.285743] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[   75.414342] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[   75.418999] brcmfmac: power management disabled
[   75.456260] brcmfmac: power management disabled
[   75.569395] brcmfmac: power management disabled
[   75.783101] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[   75.783113] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[   82.745149] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[   82.747959] brcmfmac: power management disabled
[   82.780690] brcmfmac: power management disabled
[   82.930449] brcmfmac: power management disabled
[   83.145856] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[   83.145869] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[   93.464449] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[   93.468034] brcmfmac: power management disabled
[   93.500888] brcmfmac: power management disabled
[   93.640873] brcmfmac: power management disabled
[   93.834717] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[   93.834730] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[   99.554723] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[   99.557294] brcmfmac: power management disabled
[   99.590130] brcmfmac: power management disabled
[   99.730556] brcmfmac: power management disabled
[   99.913281] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[   99.913295] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[  106.644471] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[  106.648424] brcmfmac: power management disabled
[  106.688807] brcmfmac: power management disabled
[  106.789396] brcmfmac: power management disabled
[  107.013247] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[  107.013260] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[  118.934864] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[  118.937167] brcmfmac: power management disabled
[  118.970211] brcmfmac: power management disabled
[  119.153963] brcmfmac: power management disabled
[  119.173322] brcmfmac: brcmf_sdio_hdparse: seq 6: sequence number error, expect 5
[  119.173380] brcmfmac: brcmf_sdio_hdparse: seq 5: sequence number error, expect 7
[  119.177752] brcmfmac: brcmf_sdio_hdparse: seq 7: sequence number error, expect 6
[  119.393403] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[  119.393417] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[  128.025598] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[  128.028773] brcmfmac: power management disabled
[  128.068585] brcmfmac: power management disabled
[  128.199455] brcmfmac: power management disabled
[  128.443344] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[... 277 lines cut ...]
[  551.184658] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[  551.188004] brcmfmac: power management disabled
[  551.233399] brcmfmac: power management disabled
[  551.350029] brcmfmac: power management disabled
[  551.582981] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[  551.582996] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[  561.744737] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[  561.747437] brcmfmac: power management disabled
[  561.780245] brcmfmac: power management disabled
[  561.901296] brcmfmac: power management disabled
[  562.153482] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[  562.153496] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[  572.415827] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[  572.420660] brcmfmac: power management disabled
[  572.461647] brcmfmac: power management disabled
[  572.569736] brcmfmac: power management disabled
[  572.783168] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[  572.783183] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[  583.254616] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[  583.257989] brcmfmac: power management disabled
[  583.290869] brcmfmac: power management disabled
[  583.410031] brcmfmac: power management disabled
[  583.614547] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[  583.614561] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
```

Fourdee, 2016-06-13:
@k-plan 
I think there is a way to disable WiFi using a dtoverlay in config.txt. May of been just for bluetooth, i'll try and find out.

The other method seems to be blacklisting the modules: http://raspberrypi.stackexchange.com/questions/43720/disable-wifi-wlan0-on-pi-3

I'll see which works

Fourdee, 2016-06-13:
Disable WiFi:
```
cat << _EOF_ > /etc/modprobe.d/disable_rpi3_wifi_bt.conf
blacklist brcmfmac
blacklist brcmutil
_EOF_
```
[image]

although, it doesn't physically disable the adapter hardware. Pretty sure theres no way to achieve this (yet).

Fourdee, 2016-06-13:
Good, bluetooth still works when wifi modules disabled https://github.com/Fourdee/DietPi/issues/376#issuecomment-225567091:
[image]

Fourdee, 2016-06-13:
disable bluetooth:
```
systemctl stop bluetooth
systemctl stop hciuart

systemctl disable bluetooth
systemctl disable hciuart

modprobe -rf bnep
modprobe -rf hci_uart
modprobe -rf btbcm
modprobe -rf bluetooth

```

enable:
```
modprobe bnep
modprobe hci_uart
modprobe btbcm
modprobe bluetooth

systemctl enable bluetooth
systemctl enable hciuart

systemctl start bluetooth
systemctl start hciuart
```

Fourdee, 2016-06-13:
Ok done. DietPi will also control the wifi modules in dietpi-config, if you have a RPi3.

Added option for bluetooth control while I was there, in advanced menu:
[image]

@k-plan 
I've tested both, but please let me know if its resolved the issue.

k-plan, 2016-06-14:
@Fourdee 

> I've tested both, but please let me know if its resolved the issue.

tested now 12h, it works.

```
root@RPi-UniFi:~# dmesg | grep brcmf
root@RPi-UniFi:~#
```

Bluetooth enable and disable works as well. (only tested with wlan off)

- Disable Bluetooth - works without reboot (on-the-fly)
- Enable Bluetooth - works only after rebooting the device (reboot device message can/will be helpful)

Nice work Fourdee! :+1:

Fourdee, 2016-06-14:
@k-plan 
Thanks for the tests :+1: 

> Bluetooth enable and disable works as well. (only tested with wlan off)

Just tried on RPi3 with Wlan only connection and disabled bluetooth, works a treat :)

Fourdee, 2016-06-14:
> bluetooth enable,  works only after rebooting the device
@k-plan 
Bizzare, works straight away for me:
[image]

Might be the one of the services, if you can replicate it again, can you please try to run
```
systemctl status hciuart.service
systemctl status bluetooth.service
```

k-plan, 2016-06-15:
- Bluetooth disabled:

```
root@RPi-UniFi:~# bluetoothctl
[bluetooth]#
```

-----

- Bluetooth enabled via `dietpi-config`:

```
Synchronizing state for bluetooth.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d bluetooth defaults
insserv: warning: current start runlevel(s) (empty) of script `bluetooth' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `bluetooth' overrides LSB defaults (0 1 6).
Executing /usr/sbin/update-rc.d bluetooth enable
Created symlink from /etc/systemd/system/dbus-org.bluez.service to /lib/systemd/system/bluetooth.service.
Created symlink from /etc/systemd/system/multi-user.target.wants/hciuart.service to /lib/systemd/system/hciuart.service.
```
[image]

```
root@RPi-UniFi:~# bluetoothctl
[bluetooth]#
```

```
root@RPi-UniFi:~# systemctl status hciuart.service
? hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled)
   Active: failed (Result: exit-code) since Wed 2016-06-15 19:47:04 CEST; 2min 31s ago
  Process: 5777 ExecStart=/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow - (code=exited, status=1/FAILURE)

Jun 15 19:47:04 RPi-UniFi hciattach[5777]: Initialization timed out.
Jun 15 19:47:04 RPi-UniFi hciattach[5777]: bcm43xx_init
Jun 15 19:47:04 RPi-UniFi systemd[1]: hciuart.service: control process exited, code=exi...s=1
Jun 15 19:47:04 RPi-UniFi systemd[1]: Failed to start Configure Bluetooth Modems connec...RT.
Jun 15 19:47:04 RPi-UniFi systemd[1]: Unit hciuart.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.
```

```
root@RPi-UniFi:~# systemctl status hciuart.service -l
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled)
   Active: failed (Result: exit-code) since Wed 2016-06-15 19:47:04 CEST; 10min ago
  Process: 5777 ExecStart=/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow - (code=exited, status=1/FAILURE)

Jun 15 19:47:04 RPi-UniFi hciattach[5777]: Initialization timed out.
Jun 15 19:47:04 RPi-UniFi hciattach[5777]: bcm43xx_init
Jun 15 19:47:04 RPi-UniFi systemd[1]: hciuart.service: control process exited, code=exited status=1
Jun 15 19:47:04 RPi-UniFi systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
Jun 15 19:47:04 RPi-UniFi systemd[1]: Unit hciuart.service entered failed state.

```

```
root@RPi-UniFi:~# systemctl status bluetooth.service
? bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled)
   Active: active (running) since Wed 2016-06-15 19:46:34 CEST; 3min 50s ago
     Docs: man:bluetoothd(8)
 Main PID: 5768 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           +-5768 /usr/lib/bluetooth/bluetoothd

Jun 15 19:46:34 RPi-UniFi systemd[1]: Started Bluetooth service.

```

```
root@RPi-UniFi:~# lsmod
Module                  Size  Used by
hci_uart               17943  0
btbcm                   5929  1 hci_uart
bnep                   10340  2
bluetooth             326105  7 bnep,btbcm,hci_uart
cfg80211              427855  0
rfkill                 16037  3 cfg80211,bluetooth
sg                     18319  0
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
evdev                  11396  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
fuse                   83461  1
ipv6                  347530  47
```

```
root@RPi-UniFi:~# dmesg
 . . .
[86475.194697] Bluetooth: Core ver 2.21
[86475.194751] NET: Registered protocol family 31
[86475.194756] Bluetooth: HCI device and connection manager initialized
[86475.194770] Bluetooth: HCI socket layer initialized
[86475.194778] Bluetooth: L2CAP socket layer initialized
[86475.194796] Bluetooth: SCO socket layer initialized
[86475.195592] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[86475.195600] Bluetooth: BNEP filters: protocol multicast
[86475.195610] Bluetooth: BNEP socket layer initialized
[86475.202134] Bluetooth: HCI UART driver ver 2.3
[86475.202147] Bluetooth: HCI UART protocol H4 registered
[86475.202152] Bluetooth: HCI UART protocol Three-wire (H5) registered
[86475.202240] Bluetooth: HCI UART protocol BCM registered

```

k-plan, 2016-06-18:
@Fourdee 

is it possible or reasonable to disable/blacklist  modul `cfg80211 ` as well, then we disable build-in wifi adapter?

Fourdee, 2016-06-18:
@k-plan 

Yep, makes sense, will free up 427.85KB. I'll take a look.

Fourdee, 2016-06-18:
@k-plan 
Its in. Also not RPi 3 specific anymore, all devices will have cfg80211 and bluetooth modules controlled by DietPi.

```
root@DietPi:~# lsmod
Module                  Size  Used by
brcmfmac              186343  0
brcmutil                5661  1 brcmfmac
cfg80211              427855  1 brcmfmac
rfkill                 16037  1 cfg80211
sg                     18319  0
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
fuse                   83461  1
ipv6                  347530  20
root@DietPi:~# /DietPi/dietpi/func/dietpi-set_wifi_bt wifi 0
root@DietPi:~# lsmod
Module                  Size  Used by
sg                     18319  0
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
fuse                   83461  1
ipv6                  347530  20
root@DietPi:~# /DietPi/dietpi/func/dietpi-set_wifi_bt wifi 1
root@DietPi:~# lsmod
Module                  Size  Used by
brcmfmac              186343  0
brcmutil                5661  1 brcmfmac
cfg80211              427855  1 brcmfmac
rfkill                 16037  1 cfg80211
sg                     18319  0
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
fuse                   83461  1
ipv6                  347530  20
```

k-plan, 2016-06-19:
@Fourdee 

- updating to the last v.120 testing build.

- disable and re-enable `Bluetooth` and `WiFi` via `dietpi-config`

```
synchronizing state for bluetooth.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d bluetooth defaults
insserv: warning: current start runlevel(s) (empty) of script `bluetooth' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `bluetooth' overrides LSB defaults (0 1 6).
Executing /usr/sbin/update-rc.d bluetooth enable
Created symlink from /etc/systemd/system/dbus-org.bluez.service to /lib/systemd/system/bluetooth.service.
Created symlink from /etc/systemd/system/multi-user.target.wants/hciuart.service to /lib/systemd/system/hciuart.service.
```

[image]

[image]

[image]

- TESTING:

```
root@RPi-UniFi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
  	. . .
  
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
 	. . .
       
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
     . . .
     
     
root@RPi-UniFi:~# bluetoothctl
[bluetooth]# show
No default controller available
[bluetooth]#

root@RPi-UniFi:~# lsmod
Module                  Size  Used by
hci_uart               17943  0
btbcm                   5929  1 hci_uart
bnep                   10340  2
bluetooth             326105  7 bnep,btbcm,hci_uart
brcmfmac              186343  0
brcmutil                5661  1 brcmfmac
cfg80211              427855  1 brcmfmac
rfkill                 16037  4 cfg80211,bluetooth
sg                     18319  0
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
evdev                  11396  0
fuse                   83461  1
ipv6                  347530  49

root@RPi-UniFi:~# systemctl status hciuart.service
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled)
   Active: failed (Result: exit-code) since Sun 2016-06-19 13:37:16 CEST; 6min ago
  Process: 7083 ExecStart=/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow - (code=exited, status=1/FAILURE)

Jun 19 13:37:16 RPi-UniFi hciattach[7083]: Initialization timed out.
Jun 19 13:37:16 RPi-UniFi hciattach[7083]: bcm43xx_init
Jun 19 13:37:16 RPi-UniFi systemd[1]: hciuart.service: control process exited, code=exited status=1
Jun 19 13:37:16 RPi-UniFi systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
[... 58 lines cut ...]
rfkill                 16037  1 cfg80211
sg                     18319  0
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
evdev                  11396  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
fuse                   83461  1
ipv6                  347530  32

root@RPi-UniFi:~# logread | grep blue
Jun 19 13:52:50 RPi-UniFi daemon.notice dbus[452]: [system] Activating via systemd: service name='org.bluez' unit='dbus-org.bluez.service'
Jun 19 13:52:50 RPi-UniFi daemon.info bluetoothd[1354]: Bluetooth daemon 5.23
Jun 19 13:52:50 RPi-UniFi daemon.notice dbus[452]: [system] Successfully activated service 'org.bluez'
Jun 19 13:52:50 RPi-UniFi daemon.err bluetoothd[1354]: Failed to access management interface
Jun 19 13:52:50 RPi-UniFi daemon.err bluetoothd[1354]: Adapter handling initialization failed
```

[image]

[image]

[image]

- 	SUMMARY after reboot:  wifi - not working, `dietpi-config` : all setting lost, `Scan and Connect` doesn't work  / bluetooth - not working

Fourdee, 2016-06-19:
@k-plan 
Thanks, i'll take a look :+1:

Fourdee, 2016-06-20:
Enabled Bluetooth via ```dietpi-config```:
```
root@DietPi:~# bluetoothctl
[NEW] Controller B8:27:EB:24:F9:F1 DietPi [default]
[bluetooth]#
```

Enabled WiFi via ```dietpi-config```:
```
root@DietPi:~# bluetoothctl
[NEW] Controller B8:27:EB:24:F9:F1 DietPi [default]
[bluetooth]#

root@DietPi:~# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.0.1     0.0.0.0         UG    0      0        0 eth0
default         192.168.0.1     0.0.0.0         UG    1      0        0 wlan0
192.168.0.0     *               255.255.255.0   U     0      0        0 eth0
192.168.0.0     *               255.255.255.0   U     1      0        0 wlan0

```

#### All appear to be working ok

===

#### Reboot

===

Bluetooth:
```
root@DietPi:~# bluetoothctl
[NEW] Controller B8:27:EB:24:F9:F1 DietPi [default]
[bluetooth]# exit
[DEL] Controller B8:27:EB:24:F9:F1 DietPi [default]
```

WiFi:
[image]
```
root@DietPi:~# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.0.1     0.0.0.0         UG    0      0        0 eth0
default         192.168.0.1     0.0.0.0         UG    1      0        0 wlan0
192.168.0.0     *               255.255.255.0   U     0      0        0 eth0
192.168.0.0     *               255.255.255.0   U     1      0        0 wlan0
```

#### All appear to be working ok

===

@k-plan 
I'am stumped. I can't replicate the same issues. If you get time, I'd be extremely grateful if you could do a fresh installation using testing branch in dietpi.txt, then run the tests again. 

Some additional notes/questions:
- After enabling bluetooth, if nothing appears, try running ```bluetoothctl``` then ```power on```
- WiFi country code, was yours set, or 00?
- Did you run these tests after DietPi had finished the "1st run setup", or during?
- Any changes to dietpi.txt or config.txt prior to "1st run setup"?

Thanks @k-plan :+1:

k-plan, 2016-06-20:
@Fourdee 

> I'am stumped. I can't replicate the same issues. If you get time, I'd be extremely grateful if you could do a fresh installation using testing branch in dietpi.txt, then run the tests again.

Yes, I can/will do this, when I am back at home.

> After enabling bluetooth and running `bluetoothctl` , if nothing appears, try running` power on`

```
root@RPi-UniFi:~# bluetoothctl
[bluetooth]# power on
No default controller available
```

> WiFi country code, was yours set, or 00? If your using WiFi channel 12/13, 00 will prevent connection.

WiFi country code is set to `DE`, channel is:
```
root@RPi-UniFi:~# iwlist wlan0 freq
wlan0     13 channels in total; available frequencies :
           . . .
          Current Frequency:2.412 GHz (Channel 1)
```
[image]

> Did you run these tests after DietPi had finished the "1st run setup", or during?

YES, after 1st run setup, with automated installation of mc, iperf, mtr-tiny and tcpdump.

> Any changes to dietpi.txt or config.txt prior to "1st run setup"?

dietpi.txt => NO change /  config.txt => NO change by edit

>  I see your running static for wifi, let me test

NO, I using DHCP on wifi, but static on eth0

Testing:

- Disable `bluetooth` and `wifi`
- reboot
- enable `bluetooth` ONLY via `dietpi-config` and testing; work!

```
root@RPi-UniFi:~# bluetoothctl
[NEW] Controller B8:27:EB:6B:xx:xx RPi-UniFi [default]
[bluetooth]# exit
[DEL] Controller B8:27:EB:6B:xx:xx RPi-UniFi [default]
root@RPi-UniFi:~#
```
- enable wifi via `dietpi-config` and testing; work!

```
root@RPi-UniFi:~# iwlist wlan0 rate
wlan0     unknown bit-rate information.
          Current Bit Rate=65 Mb/s
```
- going back to  `dietpi-config` and looking for `bluetooth`
- it is disable now?!?
- re-enable `bluetooth` ONLY via `dietpi-config` (now it show:` Bluetooth       : Enabled`)  and testing;  NOT working!

[image]

```
root@RPi-UniFi:~# bluetoothctl
[bluetooth]# power on
No default controller available
```

Fourdee, 2016-06-22:
@k-plan 
Thanks again :+1: , very strange, i'll take another look and see if I can replicate it.

Fourdee, 2016-06-22:
> My Testing:
- Disable bluetooth and wifi
- reboot
- enable bluetooth ONLY via dietpi-config and testing; work
- enable wifi via dietpi-config and testing; work!
- going back to dietpi-config and looking for bluetooth
- it is disable now?!?

Bluetooth was still enabled for me. Must be a service failing.

@k-plan Any chance I could SSH into this system and debug? Or even VNC/Remote desktop, you'll be able to see what I'am doing.

k-plan, 2016-06-25:
Sorry for the late reply, a long hard week with a lot of work ...

Testing clean install with the brand new v120 - The big one! on RPi3.

Same result. Switching many times between Bluetooth on-off and WiFi on-off.
Some times Bluetooth enabling fails.
But the is no regularity distinguishable.

When switching Bluetooth off-on and the terminal windows appears noticeable longer such as switching before, in this cases `bluetoothctl` fails.
Thinking, it can be a timing/timeout problem.

 

> Any chance I could SSH into this system and debug?

Yes. The "old" or the "new" one?

> Or even VNC/Remote desktop, you'll be able to see what I'am doing.

Yes, like to prefer to see what you are doing. 😉 

To my computer desktop or the the desktop on RPi3?
VNC or NoMachine Client?
Can we chat for this session? Email?

Fourdee, 2016-06-26:
@k-plan

> Sorry for the late reply, a long hard week with a lot of work ...

No worries, hopefully you had a good weekend to make up for it ? :)

> To my computer desktop or the the desktop on RPi3?
VNC or NoMachine Client?
Can we chat for this session? Email?
 
I'll send you an email @k-plan :+1:

Fourdee, 2016-06-30:
@k-plan 
I sent you an email a few days ago, might of gone to your spam folder, please let me know if you got it :)

Fourdee, 2016-06-30:
Will take this out of milestone for now.

Fourdee, 2016-07-03:
@k-plan 
I managed to replicate the same error :+1: 
Ran some tests today to try and debug why the bluetooth hciattach service is failing "sometimes".

So instead of trying to start the hciuart service, i modified the code to run the command manually:
```
#systemctl enable hciuart
#systemctl start hciuart
/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow -
```

All we get from hciattach output is:
```
bcm43xx_init
Initialization timed out.
```
So again, nothing really to go on.

But if we run the command after it fails, works everytime:
```
root@DietPi:~# /usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow -
bcm43xx_init
Flash firmware /lib/firmware/BCM43430A1.hcd
Set Controller UART speed to 921600 bit/s
Device setup complete
```

So, it feels like something isn't ready in time and prevents hciattach from starting,

Fourdee, 2016-07-03:
Workaround, restart service at most 3 times if it fails:
```
			# + RPi 3
			if (( $HW_MODEL == 3 )); then

				systemctl enable hciuart

				#Start service again, if it fails: https://github.com/Fourdee/DietPi/issues/376#issuecomment-230149754
				loop_count=0
				max_count=3
				while (( $(systemctl start hciuart;echo $?) != 0 ))
				do

					if (( $loop_count >= $max_count )); then
						echo -e "\nERROR: hciuart failed to start after $loop_count attempts\n"
						break
					else
						echo -e "\nINFO: hciuart failed to start. Attempting to restart it...\n"
						((loop_count++))
						sleep 1
					fi

				done

				#/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow -

			fi
```

Test results:
```
root@DietPi:~# /DietPi/dietpi/func/dietpi-set_wifi_bt bluetooth 0 && /DietPi/dietpi/func/dietpi-set_wifi_bt bluetooth 1 && systemctl status hciuart -l
Synchronizing state for bluetooth.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d bluetooth defaults
Executing /usr/sbin/update-rc.d bluetooth disable
insserv: warning: current start runlevel(s) (empty) of script `bluetooth' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `bluetooth' overrides LSB defaults (0 1 6).
Removed symlink /etc/systemd/system/dbus-org.bluez.service.
Removed symlink /etc/systemd/system/multi-user.target.wants/hciuart.service.
Synchronizing state for bluetooth.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d bluetooth defaults
insserv: warning: current start runlevel(s) (empty) of script `bluetooth' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `bluetooth' overrides LSB defaults (0 1 6).
Executing /usr/sbin/update-rc.d bluetooth enable
Created symlink from /etc/systemd/system/dbus-org.bluez.service to /lib/systemd/system/bluetooth.service.
Created symlink from /etc/systemd/system/multi-user.target.wants/hciuart.service to /lib/systemd/system/hciuart.service.
Job for hciuart.service failed. See 'systemctl status hciuart.service' and 'journalctl -xn' for details.

INFO: hciuart failed to start. Attempting to restart it...

● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled)
   Active: active (running) since Sun 2016-07-03 13:21:32 BST; 17ms ago
  Process: 3433 ExecStart=/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow - (code=exited, status=0/SUCCESS)
 Main PID: 3440 (hciattach)
   CGroup: /system.slice/hciuart.service
           └─3440 /usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow -

Jul 03 13:21:32 DietPi hciattach[3433]: bcm43xx_init
Jul 03 13:21:32 DietPi hciattach[3433]: Flash firmware /lib/firmware/BCM43430A1.hcd
Jul 03 13:21:32 DietPi hciattach[3433]: Set Controller UART speed to 921600 bit/s
Jul 03 13:21:32 DietPi hciattach[3433]: Device setup complete
Jul 03 13:21:32 DietPi systemd[1]: Started Configure Bluetooth Modems connected by UART.
```

k-plan, 2016-07-03:
@Fourdee 

> I managed to replicate the same error (took a while lol) 👍 

Thanks! So I have getting self-doubt on my sense over the time ... thinking I have illusion. 😄 

Should I manual edit `nano /DietPi/dietpi/func/dietpi-set_wifi_bt` and test it?

Fourdee, 2016-07-03:
@k-plan 
I've uploaded it to testing branch, so you could run to update:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/func/dietpi-set_wifi_bt -O /DietPi/dietpi/func/dietpi-set_wifi_bt
```

Then give it a whirl in dietpi-config, or, 
```
/DietPi/dietpi/func/dietpi-set_wifi_bt bluetooth 0 && /DietPi/dietpi/func/dietpi-set_wifi_bt bluetooth 1 && systemctl status hciuart -l
```

k-plan, 2016-07-03:
```
root@RPi-3-test:~# wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/func/dietpi-set_wifi_bt -O /DietPi/dietpi/func/dietpi-set_wifi_bt
--2016-07-03 15:03:25--  https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/func/dietpi-set_wifi_bt
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.12.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.12.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3762 (3.7K) [text/plain]
Saving to: ‘/DietPi/dietpi/func/dietpi-set_wifi_bt’

/DietPi/dietpi/func/dietpi-set_wifi_bt    100%[=====================================================================================>]   3.67K  --.-KB/s   in 0s

2016-07-03 15:03:25 (12.9 MB/s) - ‘/DietPi/dietpi/func/dietpi-set_wifi_bt’ saved [3762/3762]

root@RPi-3-test:~# /DietPi/dietpi/func/dietpi-set_wifi_bt bluetooth 0 && /DietPi/dietpi/func/dietpi-set_wifi_bt bluetooth 1 && systemctl status hciuart -l
Synchronizing state for bluetooth.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d bluetooth defaults
Executing /usr/sbin/update-rc.d bluetooth disable
insserv: warning: current start runlevel(s) (empty) of script `bluetooth' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `bluetooth' overrides LSB defaults (0 1 6).
Removed symlink /etc/systemd/system/dbus-org.bluez.service.
Removed symlink /etc/systemd/system/multi-user.target.wants/hciuart.service.
Synchronizing state for bluetooth.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d bluetooth defaults
insserv: warning: current start runlevel(s) (empty) of script `bluetooth' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `bluetooth' overrides LSB defaults (0 1 6).
Executing /usr/sbin/update-rc.d bluetooth enable
Created symlink from /etc/systemd/system/dbus-org.bluez.service to /lib/systemd/system/bluetooth.service.
Created symlink from /etc/systemd/system/multi-user.target.wants/hciuart.service to /lib/systemd/system/hciuart.service.
Job for hciuart.service failed. See 'systemctl status hciuart.service' and 'journalctl -xn' for details.

INFO: hciuart failed to start. Attempting to restart it...

● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled)
   Active: active (running) since Sun 2016-07-03 15:04:23 CEST; 31ms ago
  Process: 1997 ExecStart=/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow - (code=exited, status=0/SUCCESS)
 Main PID: 2004 (hciattach)
   CGroup: /system.slice/hciuart.service
           └─2004 /usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow -

Jul 03 15:04:23 RPi-3-test hciattach[1997]: bcm43xx_init
Jul 03 15:04:23 RPi-3-test hciattach[1997]: Flash firmware /lib/firmware/BCM43430A1.hcd
Jul 03 15:04:23 RPi-3-test hciattach[1997]: Set Controller UART speed to 921600 bit/s
Jul 03 15:04:23 RPi-3-test hciattach[1997]: Device setup complete
Jul 03 15:04:23 RPi-3-test systemd[1]: Started Configure Bluetooth Modems connected by UART.
root@RPi-3-test:~# bluetoothctl
[NEW] Controller B8:27:EB:xx:xx:xx RPi-3-test [default]
[bluetooth]#
```

k-plan, 2016-07-03:
Some mods:

```
                        # + RPi 3
                        if (( $HW_MODEL == 3 )); then

                                systemctl enable hciuart

                                #Start service again, if it fails: https://github.com/Fourdee/DietPi/issues/376#issuecomment-230149754
                                loop_count=0
                                max_count=3

                                #User message:
                                echo -e "\nINFO: \033[1;31m Starting Bluetooth Services. Please be patient ... \033[0m \n "

                                while (( $(systemctl start hciuart;echo $?) != 0 ))
                                do

                                        if (( $loop_count >= $max_count )); then
                                                echo -e "\nERROR: hciuart failed to start after $loop_count attempts\n"
                                                break
                                        else
                                                echo -e "\nINFO: hciuart failed to start. Attempting to restart it...\n"
                                                ((loop_count++))
                                                sleep 1
                                        fi

                                #User message:
                                echo -e "\nINFO: \033[1;32m  Done! Bluetooth Services was started with success! \033[0m \n"
                                sleep 3

                                done

                                #/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow -

                        fi

```

Result:

[image]

😃

Fourdee, 2016-07-03:
@k-plan 
Love the extra info and colours. :+1:  I'll add that in so it matches this style:
[image]

k-plan, 2016-07-03:
@Fourdee 

Yes! 👍 

You are the men for styling - I have only hunting the bugs  .... 😆

Fourdee, 2016-07-03:
Working on a function script so we can use a "global" style, across all DietPi scripts. should clean up the code and keep everything identical :
[image]

Fourdee, 2016-07-04:
@k-plan 
[image]