MichaIng/DietPi #378: [RPI3] Installation Errors

TimVNL, 2016-06-08:
Hi All,
I am new to dietpi and I am getting the following errors on first boot

Failed to start Configure Bluetooth Modems connected by WART
Failed to start LSB: DietPi boot shutdown reboot...

Then i hangs on Starting UTMP about System Runlevel changes

What can I do to resolve this

Specs:
Raspberry Pi 3
64GB SD Card
Latest DietPi Image

Also the forum registration is broken :(

Fourdee, 2016-06-08:
@TimVNL 
Sounds like a corrupt filesystem, or possibly PSU issue. 

Can you answer the following please:
- PSU make/model/amperage (eg: Samsung phone charger 2amp)
- SD card make/model (eg: Sandisk ultra)
- Have you modified any of the files in /boot/ prior to powering on the device? (eg: config.txt)

I would recommend trying the following:
- Download and extract the dietpi image again: http://dietpi.com/downloads/images/DietPi_RPi-(Jessie).7z
- Write the image to SD card.
- Power on and test

> Also the forum registration is broken :(

Cant see any errors in the logs, we had two registrations today so it appears to be working. Can you remember what your username was and did you receive any errors?

TimVNL, 2016-06-08:
@Fourdee 

PSU: Samsung 2 amp + Cwxuan Universal 2amp
SD Card Transend Premium 300x 64GB (Will try a other one tonight)

Editted no files before boot 

Flashed the SD card with a freshly download image still no luck I am flashing the image with a Mac and a app called ApplePi Baker.

can seem to enter a conformation code to register on the forum see the screenshot 
[image]

Fourdee, 2016-06-08:
@TimVNL 
I've had many days like that, nothing wants to work lol.

Good to hear everything is now working for you :+1: