MichaIng/DietPi #380: Kodi wont start

Fourdee, 2016-06-15:
As title

Fourdee, 2016-06-15:
Looks like we are missing GPU driver:
```
root@DietPi:~# cat /var/log/Xorg.0.log | grep '(EE)'
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[  1010.299] (EE) open /dev/dri/card0: No such file or directory
[  1010.349] (EE) AIGLX: reverting to software rendering
[  1010.349] (EE) AIGLX error: dlopen of /usr/lib/aarch64-linux-gnu/dri/swrast_dri.so failed (/usr/lib/aarch64-linux-gnu/dri/swrast_dri.so: cannot open shared object file: No such file or directory)
[  1010.349] (EE) GLX: could not load software renderer
[  1011.148] (EE) Server terminated successfully (0). Closing log file.
```

rhkean, 2016-06-15:
odd.... it should be there
the mali driver is in /lib/modules/3.10.101-0-pine64-longsleep/kernel/extramodules

Fourdee, 2016-06-15:
Yep, same for 3.10.101-4:
```
root@DietPi:~/sunxi-mali# ls -lha /lib/modules/3.10.101-4-pine64-longsleep/kernel/extramodules
total 6.3M
drwxrwxr-x 2 root root 4.0K May 26 19:04 .
drwxrwxr-x 9 root root 4.0K May 26 19:04 ..
-rw-rw-r-- 1 root root 6.3M May 26 19:04 mali.ko
root@DietPi:~/sunxi-mali# uname -a
Linux DietPi 3.10.101-4-pine64-longsleep #51 SMP PREEMPT Thu May 26 18:20:37 CEST 2016 aarch64 GNU/Linux
```
Also noted here:
https://github.com/longsleep/build-pine64-image/blob/master/kernel/README.md#compile-bsp-mali-kernel-module

Fourdee, 2016-06-15:
Looked at this: https://github.com/longsleep/build-pine64-image/blob/master/simpleimage/platform-scripts/install_mate_desktop.sh#L15-L32

Not sure where they are getting fbturbo from, not available in default jessie repo, or deb-multimedia repo for arm64.

rhkean, 2016-06-15:
found it

I just booted his xenial image.  it has an additional apt source:

cat /etc/apt/sources.list.d/longsleep-ubuntu-ubuntu-pine64-flavour-makers-xenial.list

`deb http://ppa.launchpad.net/longsleep/ubuntu-pine64-flavour-makers/ubuntu xenial main
`

rhkean, 2016-06-15:
very first package....

```
root@keanbean:~/tmp# head -n 25 Packages
Package: xserver-xorg-video-fbturbo
Priority: optional
Section: x11
Installed-Size: 85
Maintainer: Ryan Finnie <ryan@finnie.org>
Architecture: arm64
Version: 0~git.20151007.f9a6ed7-5~xenial1.0
Provides: xorg-driver-video
Depends: libc6 (>= 2.17), libump, xorg-video-abi-20, xserver-xorg-core (>= 2:1.17.99.902)
Filename: pool/main/x/xserver-xorg-video-fbturbo/xserver-xorg-video-fbturbo_0~git.20151007.f9a6ed7-5~xenial1.0_arm64.deb
Size: 29310
MD5sum: 4e7dc1e72a19568bbbdda74544750a9a
SHA1: 588c9afe5b976f1fbd22794f14331ef695249a98
SHA256: 3d9aa490013e1924c47f1471d910ba2f42453aaa4bea7fc32ba2462943ff5287
Description: X.Org X server -- fbturbo display driver
Description-md5: d6352865b2982b24bff4a7bf31ad0324

```

Fourdee, 2016-06-16:
```
root@DietPi:~# apt-get install xserver-xorg-video-fbturbo --no-install-recommends
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 xserver-xorg-video-fbturbo : Depends: xorg-video-abi-20 but it is not installable
                              Depends: xserver-xorg-core (>= 2:1.17.99.902) but 2:1.16.4-1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

```
wget http://ftp.de.debian.org/debian/pool/main/x/xorg-server/xserver-xorg-core_1.18.3-1_arm64.deb
root@DietPi:~# dpkg -i xserver-xorg-core_1.18.3-1_arm64.deb
dpkg: considering removing xserver-xorg-video-modesetting in favour of xserver-xorg-core ...
dpkg: yes, will remove xserver-xorg-video-modesetting in favour of xserver-xorg-core
dpkg: regarding xserver-xorg-core_1.18.3-1_arm64.deb containing xserver-xorg-core:
 xserver-xorg-core breaks systemd (<< 226-4~)
  systemd (version 215-17+deb8u4) is present and installed.

dpkg: error processing archive xserver-xorg-core_1.18.3-1_arm64.deb (--install):
 installing xserver-xorg-core would break systemd, and
 deconfiguration is not permitted (--auto-deconfigure might help)
Errors were encountered while processing:
 xserver-xorg-core_1.18.3-1_arm64.deb
```

```
apt-get purge xserver-xorg-video-modesetting

root@DietPi:~# dpkg -i --auto-deconfigure xserver-xorg-core_1.18.3-1_arm64.deb
dpkg: considering deconfiguration of systemd, which would be broken by installation of xserver-xorg-core ...
dpkg: yes, will deconfigure systemd (broken by xserver-xorg-core)
dpkg: considering deconfiguration of xserver-xorg, which would be broken by installation of xserver-xorg-core ...
dpkg: yes, will deconfigure xserver-xorg (broken by xserver-xorg-core)
(Reading database ... 21857 files and directories currently installed.)
Preparing to unpack xserver-xorg-core_1.18.3-1_arm64.deb ...
De-configuring xserver-xorg (1:7.7+7) ...
De-configuring systemd (215-17+deb8u4) ...
Unpacking xserver-xorg-core (2:1.18.3-1) over (2:1.16.4-1) ...
Replacing files in old package xserver-xorg (1:7.7+7) ...
dpkg: dependency problems prevent configuration of xserver-xorg-core:
 xserver-xorg-core depends on xserver-common (>= 2:1.18.3-1); however:
  Version of xserver-common on system is 2:1.16.4-1.
 xserver-xorg-core depends on libdbus-1-3 (>= 1.9.14); however:
  Version of libdbus-1-3:arm64 on system is 1.8.20-0+deb8u1.

dpkg: error processing package xserver-xorg-core (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of systemd:
 xserver-xorg-core (2:1.18.3-1) breaks systemd (<< 226-4~) and is unpacked but not configured.
  Version of systemd to be configured is 215-17+deb8u4.

dpkg: error processing package systemd (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of xserver-xorg:
 xserver-xorg depends on xserver-xorg-core (>= 2:1.15.0.901); however:
  Package xserver-xorg-core is not configured yet.
 xserver-xorg-core (2:1.18.3-1) breaks xserver-xorg (<< 1:7.7+10~) and is unpacked but not configured.
  Version of xserver-xorg to be configured is 1:7.7+7.

dpkg: error processing package xserver-xorg (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 xserver-xorg-core
 systemd
 xserver-xorg
root@DietPi:~#

```

Fourdee, 2016-06-16:
Just installed the "official" Pine64 Debian image: https://www.pine64.com/downloads
> Debian Linux with Mate GUI Image [20160501] by lenny.raposo with Longsleep kernel, updated by PINE64.

No FBturbo in dpkg. I installed Kodi, runs, at 0.5 frames second, software render:
```
root@debianpine64:~# cat /var/log/Xorg.0.log | grep '(EE)'
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    23.576] (EE) open /dev/dri/card0: No such file or directory
[    23.669] (EE) AIGLX: reverting to software rendering
```

Older kernel:
```
root@debianpine64:/home/pine64user# uname -a
Linux debianpine64 3.10.65-7-pine64-longsleep #28 SMP PREEMPT Sat Apr 23 20:13:25 CEST 2016 aarch64 GNU/Linux
```

@rhkean 
Did Kodi ever work on the DietPi Pine image?

rhkean, 2016-06-16:
> @rhkean 
Did Kodi ever work on the DietPi Pine image?

no.  I noted that in my announcement.

I didn't try it with longsleep's xenial image, though...  that's the image that I found the fbturbo package reference....
https://www.stdin.xyz/downloads/people/longsleep/pine64-images/ubuntu/
forum discussion here:  http://forum.pine64.org/showthread.php?tid=376

Fourdee, 2016-06-16:
@rhkean 

> no. I noted that in my announcement under known issues.

Apologies, I didn't check lol

http://forum.pine64.org/showthread.php?tid=1380&pid=12300#pid12300
>  Allwinner is still working on a Mali DRM binary blob

http://forum.pine64.org/showthread.php?tid=1380&pid=12325#pid12325
> That's why Kodi also depends on 3D drivers in addition to the strictly required VPU drivers. More simplistic media players can just do hardware accelerated video decoding job in fullscreen or in a plain looking rectangular window without any extra fluffiness and fancy 3D effects.

I'll leave this ticket open as a feature request, but i'll move it out of the milestone. I'll also disable the Kodi install option for now.

Fourdee, 2016-06-16:
Gah:
```
apt-get install -y build-essential git make gcc automake

cd /root
git clone https://github.com/linux-sunxi/libump.git
cd libump
autoreconf -i
./configure
make -j $(nproc --all)
make install

cd /root
git clone https://github.com/ssvb/xf86-video-fbturbo.git
cd xf86-video-fbturbo/
autoreconf -vi
./configure --prefix=/usr
make -j $(nproc --all)
make install

cd /root
git clone https://github.com/linux-sunxi/libcedrus.git
cd libcedrus
make -j $(nproc --all)
make install

cd /root
git clone https://github.com/linux-sunxi/libvdpau-sunxi.git
cd libvdpau-sunxi
git checkout 906c36ed45ceb53fecd5fc72e821c11849eeb1a3 #current source code is broken.
# requires libvdpau 1.1, grab it from stretch
sed -i 's/jessie/stretch/' /etc/apt/sources.list
apt-get update
apt-get install libvdpau-dev -y
sed -i 's/stretch/jessie/' /etc/apt/sources.list
apt-get update

make -j $(nproc --all)
make install
```

```
root@DietPi:~/libvdpau-sunxi# make -j $(nproc --all)
cc -c tiled_yuv.S -o tiled_yuv.o
tiled_yuv.S: Assembler messages:
tiled_yuv.S:25: Error: unknown pseudo-op: `.syntax'
tiled_yuv.S:26: Error: unknown architecture `armv7-a'

tiled_yuv.S:27: Error: unknown pseudo-op: `.fpu'
tiled_yuv.S:28: Error: unknown pseudo-op: `.thumb'
tiled_yuv.S:46: Warning: unknown register 'r0' -- .req ignored
tiled_yuv.S:47: Warning: unknown register 'r1' -- .req ignored
tiled_yuv.S:48: Warning: unknown register 'r2' -- .req ignored
tiled_yuv.S:49: Error: operand 1 should be a SIMD vector register -- `cnt .req r3'
tiled_yuv.S:50: Warning: unknown register 'r4' -- .req ignored
tiled_yuv.S:51: Warning: unknown register 'r5' -- .req ignored
tiled_yuv.S:52: Warning: unknown register 'r6' -- .req ignored
tiled_yuv.S:53: Warning: unknown register 'r7' -- .req ignored
tiled_yuv.S:54: Warning: unknown register 'r8' -- .req ignored
tiled_yuv.S:55: Warning: unknown register 'r9' -- .req ignored
tiled_yuv.S:56: Warning: unknown register 'r12' -- .req ignored
tiled_yuv.S:57: Warning: unknown register 'lr' -- .req ignored
tiled_yuv.S:59: Error: unknown pseudo-op: `.thumb_func'
tiled_yuv.S:60: Error: unknown mnemonic `push' -- `push {r4,r5,r6,r7,r8,lr}'
tiled_yuv.S:61: Error: operand 1 should be an integer register -- `ldr HEIGHT,[sp,#24]'
tiled_yuv.S:62: Error: operand 1 should be an integer or stack pointer register -- `add NEXTLIN,r3,#31'
tiled_yuv.S:63: Error: unknown mnemonic `lsrs' -- `lsrs NTILES,r3,#5'
tiled_yuv.S:64: Error: operand 1 should be a SIMD vector register -- `bic NEXTLIN,NEXTLIN,#31'
tiled_yuv.S:65: Error: operand 1 should be an integer or stack pointer register -- `and REST,r3,#31'
tiled_yuv.S:66: Error: operand 1 should be an integer register -- `lsl NEXTLIN,NEXTLIN,#5'
[... 50 lines cut ...]
tiled_yuv.S:145: Error: operand 1 should be an integer or stack pointer register -- `add DST,PITCH'
tiled_yuv.S:146: Error: operand 1 should be an integer or stack pointer register -- `add DST2,PITCH'
tiled_yuv.S:149: Error: operand 1 should be an integer register -- `subs TLINE,#1'
tiled_yuv.S:150: Error: unknown mnemonic `itee' -- `itee ne'
tiled_yuv.S:151: Error: unknown mnemonic `addne' -- `addne SRC,NEXTLIN'
tiled_yuv.S:152: Error: unknown mnemonic `subeq' -- `subeq SRC,#992'
tiled_yuv.S:153: Error: unknown mnemonic `moveq' -- `moveq TLINE,#32'
tiled_yuv.S:155: Error: operand 1 should be an integer register -- `subs HEIGHT,#1'
tiled_yuv.S:157: Error: unknown mnemonic `pop' -- `pop {r4,r5,r6,r7,r8,r9,pc}'
tiled_yuv.S:160: Error: operand 1 should be an integer register -- `mov TMPSRC,SRC'
tiled_yuv.S:161: Error: operand 1 should be an integer register -- `tst REST,#8'
tiled_yuv.S:163: Error: unknown mnemonic `vld2.8' -- `vld2.8 {d0-d1},[TMPSRC:128]!'
tiled_yuv.S:164: Error: unknown mnemonic `vst1.8' -- `vst1.8 {d0},[DST]!'
tiled_yuv.S:165: Error: unknown mnemonic `vst1.8' -- `vst1.8 {d1},[DST2]!'
tiled_yuv.S:166: Error: operand 1 should be an integer or stack pointer register -- `add SRC,TSIZE'
tiled_yuv.S:167: Error: operand 1 should be an integer register -- `ands CNT,REST,#7'
tiled_yuv.S:169: Error: unknown mnemonic `vld2.8' -- `vld2.8 {d0[0],d1[0]},[TMPSRC]!'
tiled_yuv.S:170: Error: operand 1 should be an integer register -- `subs CNT,#1'
tiled_yuv.S:171: Error: unknown mnemonic `vst1.8' -- `vst1.8 {d0[0]},[DST]!'
tiled_yuv.S:172: Error: unknown mnemonic `vst1.8' -- `vst1.8 {d1[0]},[DST2]!'
Makefile:46: recipe for target 'tiled_yuv.o' failed
make: *** [tiled_yuv.o] Error 1
```

> tiled_yuv.S:26: Error: unknown architecture `armv7-a'

rhkean, 2016-06-16:
why does that say "armv7" ?

shouldn't that be saying "arm64"?    (I don't think this is a multi-arch kernel)

Fourdee, 2016-06-16:
@rhkean 

> shouldn't that be saying "arm64"?

Yep. Probably needs a flag, but no idea at the moment.

The good news is FBturbo + libump works. So better performance in desktops:
```
root@DietPi:~# cat /var/log/Xorg.0.log | grep fbturbo
[    16.232] (II) LoadModule: "fbturbo"
[    16.239] (II) Loading /usr/lib/xorg/modules/drivers/fbturbo_drv.so
[    16.246] (II) Module fbturbo: vendor="X.Org Foundation"
[    16.246] (II) FBTURBO: driver for framebuffer: fbturbo
[    16.246] (WW) Falling back to old probe method for fbturbo
```

I've uploaded the arm64 debs for FBturbo:
```
wget http://dietpi.com/downloads/binaries/all/libump_1-1_arm64.deb -O package.deb
dpkg -i package.deb
rm package.deb

wget http://dietpi.com/downloads/binaries/all/xf86-video-fbturbo_1-1_arm64.deb -O package.deb
dpkg -i package.deb
rm package.deb
```

Fourdee, 2016-06-20:
Updates on mali_drm driver:

http://forum.pine64.org/showthread.php?tid=587&pid=12614#pid12614
> The 3D acceleration already completed and currently under stability test and memory optimization. The exa acceleration functionality will be implement thru hardware rotate and memory copy implementation. The code will be release on early next week.

nicothyrael, 2016-07-30:
Hi!
As the Mali driver was updated, can you re-enable Kodi to be tested?
[http://forum.pine64.org/showthread.php?tid=587&page=12](url)

Thanks!

Fourdee, 2016-07-31:
@nicothyrael 
Yep :+1: 

Notes to self:
Build script with Mali DRM driver: https://github.com/longsleep/build-pine64-image/commit/81bca181179dbcf5c82ae0a35b83ddcc2c04b9fd

Kernel with Mali DRM driver: https://github.com/longsleep/linux-pine64/commit/f79eb809916be7ec7ef30c860b08bdde3c8c2dd8

Fourdee, 2016-08-05:
http://forum.pine64.org/showthread.php?tid=587&pid=17035#pid17035
```
bash <(curl -s https://raw.githubusercontent.com/longsleep/build-pine64-image/master/simpleimage/platform-scripts/pine64_update_kernel.sh) latest drm

apt-get install -y software-properties-common
add-apt-repository -y ppa:longsleep/ubuntu-pine64-flavour-makers

#Change repo target from Jessie to xenial. Longsleeps packages are only available for distro Xenial.
echo -e "deb http://ppa.launchpad.net/longsleep/ubuntu-pine64-flavour-makers/ubuntu xenial main" > /etc/apt/sources.list.d/longsleep-ubuntu-pine64-flavour-makers-jessie.list

apt-get update
```

```
root@DietPi:~# apt-get install -y xserver-xorg-video-armsoc-sunxi
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 xserver-xorg-video-armsoc-sunxi : Depends: xorg-video-abi-20 but it is not installable
                                   Depends: xserver-xorg-core (>= 2:1.17.99.902) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

Looks like its going be one of those "hacky hacky, worky worky" jobs to get it working on Jessie :-1:

nicothyrael, 2016-08-06:
This worked for me:
1. Update kernel and uboot with longsleep scripts.
2. Reboot
3. sudo apt-get install -y software-properties-common
4. sudo add-apt-repository -y ppa:longsleep/ubuntu-pine64-flavour-makers
5. sudo apt-get update
6. sudo apt-get upgrade
7. sudo apt-get dist-upgrade
8. sudo apt-get install xserver-xorg-video-armsoc-sunxi

but still get:

the following packages have unmet dependencies:
 kodi : Depends: kodi-bin (>= 2:16.1~git20160425.1001-final-0xenial) but 15.2+dfsg1-3ubuntu1 is to be installed
        Depends: libshairplay0 but it is not installable
        Recommends: libva-intel-vaapi-driver but it is not installable or
                    i965-va-driver but it is not installable
E: Unable to correct problems, you have held broken packages.

Fourdee, 2016-08-07:
Ok lets try upgrading to stretch, rather than create a endless list of dependency workarounds.

On a fresh DietPi installation:
```
cat << _EOF_ > /etc/apt/sources.list
deb http://ftp.debian.org/debian stretch main contrib non-free
deb http://ftp.debian.org/debian stretch-updates main contrib non-free
deb http://security.debian.org stretch/updates main contrib non-free
deb http://ftp.debian.org/debian stretch-proposed-updates contrib non-free main
_EOF_

apt-get update
apt-get dist-upgrade

# Yes for service restarts
# Yes to all config overwrites, excluding swapfile and cron jobs

bash <(curl -s https://raw.githubusercontent.com/longsleep/build-pine64-image/master/simpleimage/platform-scripts/pine64_update_kernel.sh) latest drm

reboot

apt-get install -y software-properties-common
add-apt-repository -y ppa:longsleep/ubuntu-pine64-flavour-makers

#Change repo target to xenial. Longsleeps packages are only available for distro Xenial.
echo -e "deb http://ppa.launchpad.net/longsleep/ubuntu-pine64-flavour-makers/ubuntu xenial main" > /etc/apt/sources.list.d/longsleep-ubuntu-pine64-flavour-makers-jessie.list

apt-get update

apt-get install -y xserver-xorg-video-armsoc-sunxi alsa-base xinit
apt-get install -y kodi

xinit kodi

```
Kodi runs, but its < 1fps, screen keeps cycling signal.

```
16:57:40 T:547548053504  NOTICE: Starting Kodi from Debian (16.1 Debian package version: 16.1+dfsg1-1). Platform: Linux ARM 64-bit
16:57:40 T:547548053504  NOTICE: Using Release Kodi from Debian x64 build
16:57:40 T:547548053504  NOTICE: Kodi from Debian compiled Apr 24 2016 by GCC 5.3.1 for Linux ARM 64-bit version 4.5.1 (263425)
16:57:40 T:547548053504  NOTICE: Running on Debian GNU/Linux stretch/sid testing, kernel: Linux ARM 64-bit version 3.10.102-2-pine64drm-longsleep
16:57:40 T:547548053504  NOTICE: FFmpeg version: 3.0.1-3
16:57:40 T:547548053504  NOTICE: Host CPU: AArch64 Processor rev 4 (aarch64), 4 cores available
16:57:40 T:547548053504  NOTICE: special://xbmc/ is mapped to: /usr/share/kodi
16:57:40 T:547548053504  NOTICE: special://xbmcbin/ is mapped to: /usr/lib/aarch64-linux-gnu/kodi
16:57:40 T:547548053504  NOTICE: special://masterprofile/ is mapped to: /root/.kodi/userdata
16:57:40 T:547548053504  NOTICE: special://home/ is mapped to: /root/.kodi
16:57:40 T:547548053504  NOTICE: special://temp/ is mapped to: /root/.kodi/temp
16:57:40 T:547548053504  NOTICE: The executable running is: /usr/lib/aarch64-linux-gnu/kodi/kodi.bin
16:57:40 T:547548053504  NOTICE: Local hostname: DietPi
16:57:40 T:547548053504  NOTICE: Log File is located: /root/.kodi/temp/kodi.log
16:57:40 T:547548053504  NOTICE: -----------------------------------------------------------------------
16:57:40 T:547548053504   ERROR: DBus: Error org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.UPower was not provided by any .service files
16:57:40 T:547548053504  NOTICE: load settings...
16:57:41 T:547548053504   ERROR: PulseAudio: Failed to connect context
16:57:41 T:547548053504  NOTICE: PulseAudio might not be running. Context was not created.
16:57:41 T:547548053504  NOTICE: Found 1 Lists of Devices
16:57:41 T:547548053504  NOTICE: Enumerated ALSA devices:
16:57:41 T:547548053504  NOTICE:     Device 1
16:57:41 T:547548053504  NOTICE:         m_deviceName      : default
16:57:41 T:547548053504  NOTICE:         m_displayName     : Default (sndhdmi Analog)
16:57:41 T:547548053504  NOTICE:         m_displayNameExtra:
16:57:41 T:547548053504  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
16:57:41 T:547548053504  NOTICE:         m_channels        : FL,FR,BL,BR,FC,LFE,SL,SR
16:57:41 T:547548053504  NOTICE:         m_sampleRates     : 8000,11025,16000,22050,32000,44100,48000,64000,88200,96000,176400,192000
16:57:41 T:547548053504  NOTICE:         m_dataFormats     : AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S16LE
16:57:41 T:547548053504  NOTICE: No settings file to load (special://xbmc/system/advancedsettings.xml)
16:57:41 T:547548053504  NOTICE: No settings file to load (special://masterprofile/advancedsettings.xml)
16:57:41 T:547548053504  NOTICE: Default DVD Player: dvdplayer
16:57:41 T:547548053504  NOTICE: Default Video Player: dvdplayer
16:57:41 T:547548053504  NOTICE: Default Audio Player: paplayer
16:57:41 T:547548053504  NOTICE: Disabled debug logging due to GUI setting. Level 0.
16:57:41 T:547548053504  NOTICE: Log level changed to "LOG_LEVEL_NORMAL"
16:57:41 T:547548053504  NOTICE: Loading player core factory settings from special://xbmc/system/playercorefactory.xml.
16:57:41 T:547548053504  NOTICE: Loaded playercorefactory configuration
16:57:41 T:547548053504  NOTICE: Loading player core factory settings from special://masterprofile/playercorefactory.xml.
16:57:41 T:547548053504  NOTICE: special://masterprofile/playercorefactory.xml does not exist. Skipping.
16:57:41 T:547548053504  NOTICE: Running database version Addons20
16:57:41 T:547548053504  NOTICE: ADDONS: Using repository repository.xbmc.org
16:57:41 T:547548053504  NOTICE: Setup SDL
16:57:41 T:547548053504  NOTICE: Checking resolution 16
16:57:42 T:547548053504  NOTICE: Using visual 0x21
16:57:42 T:547548053504  NOTICE: GL_VENDOR = Mesa Project
16:57:42 T:547548053504  NOTICE: GL_RENDERER = Software Rasterizer
16:57:42 T:547548053504  NOTICE: GL_VERSION = 2.1 Mesa 11.2.2
16:57:42 T:547548053504  NOTICE: GL_SHADING_LANGUAGE_VERSION = 1.20
[... cut ...] ustness GL_ARB_separate_shader_objects GL_ARB_texture_compression_bptc GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_ARB_compressed_texture_pixel_storage GL_ARB_map_buffer_alignment GL_ARB_texture_storage GL_AMD_shader_trinary_minmax GL_ARB_clear_buffer_object GL_ARB_invalidate_subdata GL_ARB_program_interface_query GL_ARB_vertex_attrib_binding GL_KHR_debug GL_ARB_multi_bind GL_ARB_texture_mirror_clamp_to_edge GL_ARB_get_texture_sub_image GL_KHR_context_flush_control
```

Fourdee, 2016-08-07:
xinit startx
```
root@DietPi:~# cat /var/log/Xorg.0.log | grep '(EE)'
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    49.489] (EE) ARMSOC(0): ERROR: Pixman failed to blit from /dev/fb0 to scanout buffer
[    49.493] (EE) MODE :0
[    49.626] (EE) AIGLX error: dlopen of /usr/lib/aarch64-linux-gnu/dri/armsoc_dri.so failed (/usr/lib/aarch64-linux-gnu/dri/armsoc_dri.so: cannot open shared object file: No such file or directory)
[    49.626] (EE) AIGLX: reverting to software rendering
[    49.762] (EE) MODE :0
[    51.023] (EE) MODE :0
[    51.130] (EE) MODE :0
[    52.554] (EE) MODE :0
[    53.437] (EE) MODE :0
[    53.540] (EE) MODE :0
```
/usr/lib/aarch64-linux-gnu/dri/armsoc_dri.so

Fourdee, 2016-08-07:
Moving this out of milestone. This is going to take much longer than expected and needs input from everyone.

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

rtwld, 2023-10-30:
I am trying to run Kodi for Debian 20.x on DietPi bookworm on Pine64, but to no avail, since hw acceleration is missing.

I wonder if @MichaIng would be interested in reopening this issue? Since 2020, the [LibVA implementation for the Linux Video4Linux2 Request API](https://bootlin.com/blog/wrapping-up-the-allwinner-vpu-crowdfunded-linux-driver-work) is available for Allwinner: https://github.com/bootlin/libva-v4l2-request; and LibreElec seems to be running Kodi with hw acceleration on Pine64, too. Perhaps this would make support for Pine64 easier?