MichaIng/DietPi #382: DietPi Image/zip | Naming system

Fourdee, 2016-06-20:
As suggested by @rhkean :+1: 

7z file:
```
DietPi_${Board}-${arch}-(${distro}).7z
```

img file:
```
DietPi_${version}_${Board}-${arch}-(${distro}).img
```
Once all DietPi images have been updated to 120 and 120 is released, I'll go through all files and download links as needed.

Status:
- :u5408: | pine64
- :u5408: | odroid c2
- :u5408: | odroid c1
- :u5408: | odroid xu4
- :u5408: | rpi jessie
- :u5408: | rpi wheezy
- :u5408: | bpi m2+
- :u5408: | orange pi pc
- :u5408: | vmware
- :u5408: | xenserver
- :u5408: | virtualbox

k-plan, 2016-06-20:
Hmm,

Don't like` DietPi_${Board}-${arch}-(${distro}).7z`.
I think, I know, why you like to do this.

Will favor` DietPi_${dietpiversion}-${Board}-${arch}-(${distro}).7z` and you link on http://dietpi.com/download to ` DietPi_${Board}-${arch}-(${distro})-{latest}.7z` and have a symbolic link to the latest version. So you can leave old version on-line in `http://dietpi.com/downloads/images/`, please.
Don't like to download and unpack only to see if there is a new image version.

Sorry, but only my two penny worth ...

rhkean, 2016-06-20:
my only concern was that the x64 images have the arch after the board name, but the arm64 images have it after the distro

Fourdee, 2016-06-20:
> I think, I know, why you like to do this.

Easier to create the images lol :)

> Will favorDietPi_${dietpiversion}-${Board}-${arch}-(${distro}).7z and you link on http://dietpi.com/download to DietPi_${Board}-${arch}-(${distro})-{latest}.7z and have a symbolic link to the latest version.

I agree with the addition of version numbers for .7z files. Then we can just give users the link (as you mentioned) above to download the latest version.

> So you can leave old version on-line in http://dietpi.com/downloads/images/

Although a good idea, I'd like to ensure the user always downloads the "latest" image where possible. We dont want users downloading v90 and having to update 30+ DietPi versions. The shorter the installation and update for the user, the better.
There is also the support/time issue. If we provider older DietPi image downloads, aswell as the current ones, we would need to fully support all image releases, as well as the entire dietpi-software catalog, for eternity.
I'd love to be able to do that, but we need to be realistic, a prime example is the OPi PC image that I recently changed to ARMbian due to the security exploit. If we provide the previous Loboris image that contains the exploit, we deserve a slap in the face :)

So from where I look at it:
- I/we aim to support the most recent DietPi image download and the patch versions after that.
- I/we aim to try and support previous/outdated images where possible. But, as we are limited on time and members and the endless possibilities and combinations of what users will install with dietpi-software, we need to be realistic about what we can actually support.

k-plan, 2016-06-21:
Wow, one suggestion and so keenly reactions.  :smile:  

Sorry, no hard feelings for just anybody.

@rhkean 

> Hmm,
Don't likeDietPi_${Board}-${arch}-(${distro}).7z.

>my only concern was that the x64 images have the arch after the board name, but the arm64 images have it after the distro

Yes, my first phrase was a little bit short and harsh. I personally only miss the version-no. in packed file name.
But you are right with the architecture mark, it was inconsistent.

@Fourdee 

> I agree with the addition of version numbers for .7z files. Then we can just give users the link (as you mentioned) above to download the latest version.

Nice, thank you. :+1:  :smiley: 

And I respect your reasons, to not provide outdated images any more. It's more then okay for me.

rhkean, 2016-06-21:
@k-plan 
No hard feelings.... I didn't take anything harshly at all. :smile: 

I've been so swamped at work that I'm sure my comments have probably also come across as short and/or harsh.  And, if so, I apologize. 

All just open discussion as far as I can tell.  :+1:

Fourdee, 2016-06-21:
@k-plan @rhkean 
"Love is in the air" :dancer: :dancers: lol

All jokes aside. All feedback is good feedback. Helps make DietPi better, so :+1:  :+1:  to both of you :)

Fourdee, 2016-06-22:
@k-plan 

New .7z file naming + images.
[image]

rhkean, 2016-06-22:
I don't have a preference.  I only used aarch64 to be consistent with #341 

Will you include links for "latest"?
DietPi_latest_PineA64-arm64-(Jessie).7z  --> DietPi_v120_PineA64-arm64-(Jessie).7z

Fourdee, 2016-06-22:
@rhkean 
Prior to v120 release, i've uploaded them to: http://dietpi.com/downloads/testing/

> DietPi_latest_PineA64-arm64-(Jessie).7z --> DietPi_v120_PineA64-arm64-(Jessie).7z

Yep, i'll maintain the version URL links on webpage: http://dietpi.com/download
The above URL is now "the main link/place for users to download DietPi".

rhkean, 2016-06-22:
so, will I be able to do this in a script?
```
strTestingPath=""
if (( ${bTesting} )); then
    strTestingPath="/testing"
fi
strDownloadBaseURL="http://dietpi.com/downloads${strTestingPath}"
strDownloadFile="DietPi_latest_${strTargetBoard}-${arch}-(Jessie).7z"
strDownloadURL="${strDownloadBaseURL}/${strDownloadFile}"
wget "${strDownloadURL}"
```

or should I add a piece of code that pulls the version number from github?

Fourdee, 2016-06-23:
@rhkean 

> will I be able to do this in a script?

I usually clear out the testing folder. So once v120 is released, i'll move those images to dietpi.com/downloads/images section.

> or should I add a piece of code that pulls the version number from github?

We dont always create a updated image for each DietPi version due to the time involved to complete them. Sometimes we update 1 image (eg: due to OrangePi PC root exploit) if needed.

So in other words, your best bet would be to scrape the contents of this folder (once the new images with version codes are there): http://dietpi.com/downloads/images/ . Then either remove ```vXXX``` from the scrape, or, wildcard ``` DietPi_v*_Device-Arch-(distro).7z```

rhkean, 2016-06-23:
ok, I disagree with putting the version number in the 7z filename, then.  The reason for suggesting the naming convention was to facilitate automation as noted above:  DietPi_${Board}-${Arch}-(Jessie).7z

adding the version number into the .7z filename prevents automation, since wget doesn't support wildcards in the URL.

Addionally, since the download will only provide the latest image, and the .img filename has the version number in it, there is no reason to include it in the .7z filename.

Fourdee, 2016-06-24:
@rhkean 
Seems Curl doesn't support wildcards either, at least not natively http://apple.stackexchange.com/questions/60360/use-curl-to-download-images-from-website-using-wildcard.

I think @k-plan wanted to know the image version, before downloading the file and having to extract it.

I'am going to have to agree with @rhkean on this one. Adding the version codes into .7z means manually updating the dietpi.com page download links, each time a new image is released. I'd rather not have to do this each time.

So, .7z will not contain the version number.

Fourdee, 2016-06-24:
Ok, i'll get virtualbox image updated.

@xenfomation 
Any chance you would be up for updating the Xenserver image?

Fourdee, 2016-06-24:
Done: http://dietpi.com/downloads/testing/

closed 2016-06-24 by Fourdee