MichaIng/DietPi #388: DietPi-Software | Mate Desktop on RPi3

k-plan, 2016-06-25:
- `DietPi- ...` links on Mate Desktop didn't work:

[image]

solution: 

`root@RPi-3-test:~# apt-get install xterm`

[image]

----

- File manager link on Mate Desktop didn't work:

[image]

solution: 

create a link to `Caja` - the official File manager on Mate-Desktop

Fourdee, 2016-06-26:
@k-plan 
Thanks for the report and solutions, i'll take a look :+1:

Fourdee, 2016-06-27:
Great work @k-plan, all issues now resolved and patched in v121 :+1: 

Also resolved following by setting execute permissions during install:
```
chmod +x /usr/share/applications/*
```
[image]

Fourdee, 2016-06-28:
@k-plan 
As I've completed my tests and everything appears resolved, I'll mark this as completed, but please reopen if required. Good job :+1:

closed 2016-06-28 by Fourdee