MichaIng/DietPi #389: DietPi-Image | v.120

k-plan, 2016-06-25:
- Download:  http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Jessie).7z

- unpacked: `DietPi_v120_RPi-armv6-(Jessie).img`

- Write on sd-card and starting

- first thing which will be preformed is a update from v.119 => v.120 

```
root@RPi-3-test:~# cat /DietPi/dietpi/.version
119
```

As far as I can see in the image files for Odroid C1 v.120  and Odroide C2 v.120 is is the same ....

Fourdee, 2016-06-25:
@k-plan 

Yep this is expected. 
When we finalize the image, we drop the version number by 1. This allows us to create the image at a specific time during v120 devolpment, then run any final patches during v120 update.

If we are being clinical, i agree, the images should be called v119. But as our update system runs during 1st run, they are technically v119.5 :).

k-plan, 2016-06-25:
> Technically speaking, the images are v119.5 

😄

closed 2016-06-25 by k-plan