MichaIng/DietPi #390: DietPi-Update And UserData Problem

yunussandikci, 2016-06-25:
1) I can use dietpi-update it gives this error. I am absouletly know that my internet connection is good
[image]
2) im taking user data error when i click the Go button on dietpi-software at first boot. 
Solve = Changing user data location to usb hard drive and changing back it to Flash(SD/EMMC)
Sorry for my bad english and im really loving you distro : ) Keep Going :+1:

Fourdee, 2016-06-26:
@yunussandikci 
Thanks for the report, looks like a possible issue with EOL.

Would please to send us a bug report? Please see here for instructions: http://dietpi.com/phpbb/viewtopic.php?f=8&t=499&p=2184#p2184

Fourdee, 2016-06-26:
Possible link to another user reporting similar issues: https://twitter.com/DietPi_/status/747019385932492800

yunussandikci, 2016-06-26:
@Fourdee 
My reference code is 4e2c7da6-a3f7-4a75-bce1-231c7b6151c9-0
Thanks :)

Fourdee, 2016-06-26:
> Note : Dont forget to look for that i mentioned "2" bugs :)

Noted and thanks for doing the bug report  :+1: . I will look into this tomorrow.

Fourdee, 2016-06-27:
For reference, possible same issue: https://github.com/Fourdee/DietPi/issues/393

Fourdee, 2016-06-27:
Looks like dietpi.txt has dos line endings (bottom right):
[image]

I've checked DietPi_v120_RPi-armv6-(Jessie).img. The dietpi.txt file contained inside is unix line endings.

@yunussandikci Where did you download the compressed image, and, once decompressed, which version is in the image file?
eg: DietPi_v120_RPi-armv6-(Jessie).img
Thanks.

Fourdee, 2016-06-27:
@yunussandikci 

Please run the following command to convert line endings:
```
sed -i $'s/\r$//' /DietPi/dietpi.txt
dietpi-update
```

Fourdee, 2016-06-27:
#### Notes to self:

Flip incorrectly detects binary files:
```
root@DietPi:~# flip -uv /DietPi/dietpi/*
/DietPi/dietpi/boot
flip: /DietPi/dietpi/conf: can't open: Is a directory
/DietPi/dietpi/dietpi-apt-get_update
/DietPi/dietpi/dietpi-autostart
/DietPi/dietpi/dietpi-backup
/DietPi/dietpi/dietpi-banner: binary file, not converted
/DietPi/dietpi/dietpi-bugreport
/DietPi/dietpi/dietpi-cleaner: binary file, not converted
```

Fourdee, 2016-06-27:
@yunussandikci 
Some questions:
- Where did you download the compressed image (url)?
- Once decompressed, which version is in the image file? (eg: DietPi_v120_RPi-armv6-(Jessie).img)
- Which program did you use to write the image to SD card? And, OS of that system.
- Did you modify the /boot/dietpi.txt file prior to booting (eg: setting swapfile size)

Thanks.

yunussandikci, 2016-06-27:
1) I downloaded image from this url "http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Jessie).7z"
2) DietPi_v120_RPi-armv6-(Jessie).img
3) Win32  Disk Imager 0.9.5 with Windows 10 Pro 64Bit
4) I modified only this lines with Notepad++ ;

Wifi_Enabled=1
Wifi_SSID=YunusErk
Wifi_KEY=password
Use_Static=1
Static_IP=192.168.0.100
Static_Mask=255.255.255.0
Static_Gateway=192.168.0.1
Static_DNS=8.8.8.8

yunussandikci, 2016-06-27:
I dont really know much about  "what is dos line endings" or something you mentioned :(  I made some changenings in dietpi.txt because i dont have ethernet cable, im doing my installation with ssh over wifi. What should i do when i want to make fresh installation,
Should i change change my eol conversation setting to UNİX/OSX on notepad++ ? am i right? was it the problem ?

Fourdee, 2016-06-27:
@yunussandikci 
Thanks. Yes it looks like the modifications to dietpi.txt somehow converted the line endings to dos.

I'll code something into v121 that will automatically convert that file and a few others. Should prevent this from occurring in the future.

For now, you can set EOL in notepad++ here:
[image]

yunussandikci, 2016-06-27:
I'm sorry about my ignorance about that :/ And really thank you for effort. Best RPi Distro Ever! :+1:

closed 2016-06-27 by yunussandikci

Fourdee, 2016-06-27:
@yunussandikci 

> I'm sorry about my ignorance about that :/ 

Not at all! I didn't know what line endings were when I started DietPi lol. Had a few similar issues myself, and took me a while to figure it all came down to line endings.

I've updated DietPi to automatically apply unix line endings to dietpi.txt and core user files during boot. It wont take effect until the user has updated DietPi to v121, or, when we next update the images. So, in a case like this ticket, users can run ```sed -i $'s/\r$//' /DietPi/dietpi.txt``` to proceed.
```
#echo -e $(date +%N) >> /root/test
sed -i $'s/\r$//' "$FILEPATH_RAM"/uEnv.txt &> /dev/null
sed -i $'s/\r$//' "$FILEPATH_RAM"/boot.ini &> /dev/null
sed -i $'s/\r$//' "$FILEPATH_RAM"/config.txt &> /dev/null
sed -i $'s/\r$//' "$FILEPATH_RAM"/dietpi.txt
#echo -e $(date +%N) >> /root/test
```
Adds 20~ms (probably <10ms due to file write) to bootime, but its worth it. Line endings are a nightmare and not always obvious to debug.
```
root@DietPi:~# cat test
295602000
315157000
```