MichaIng/DietPi #391: DietPi-Software | Add installation option for Gogs

Fourdee, 2016-06-26:
As title: https://gogs.io/

Fourdee, 2016-06-26:
docs: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=70#p2187

Fourdee, 2016-06-26:
Completed, available from v121.

closed 2016-06-26 by Fourdee

Zone22, 2016-09-02:
Is it available on arm64?

Fourdee, 2016-09-04:
@Zone22 
Yep its disabled for ARM64: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L744
Gogs do not currently offer the ARM64 binary as a download: https://dl.gogs.io/

DJCrashdummy, 2017-07-12:
there is an open issue #686 for adding [Gitea](http://gitea.io/) (a community managed fork of Gogs) which [provides arm64-binaries (and even more)](https://dl.gitea.io/gitea/master/) besides a quicker development and some other improvements.

i guess adding Gitea (maybe replacing Gogs) and solving #686 will bring some real benefit to DietPi.