MichaIng/DietPi #392: DietPi-Software | Webmin

Fourdee, 2016-06-26:
http://www.webmin.com/

> http://dietpi.com/phpbb/viewtopic.php?f=12&t=498#p2181

joaofl, 2016-08-30:
Its a great tool. Have tested on my odroid xu4 running Dietpi. Tested quiet a lot of tools: admin pages for ProFTP Server, Apache, MySQL, bandwidth monitor (only tested on eth0. Requires rsyslog package). Everything I've tested is working so far.

Very easy to install and configure: 

Depends on:
apt-get install perl libnet-ssleay-perl openssl libauthen-pam-perl libpam-runtime libio-pty-perl apt-show-versions python

source (/etc/apt/sources.list): deb http://download.webmin.com/download/repository sarge contrib 
cd /root
wget http://www.webmin.com/jcameron-key.asc
apt-key add jcameron-key.asc 

apt-get update
apt-get install webmin

from: http://www.webmin.com/deb.html

mknrls, 2016-09-26:
will this be added sometime soon or still being tested?

Fourdee, 2016-09-27:
@mknrls 

> will this be added sometime soon or still being tested?

Not started yet. I'll take a look at @joaofl notes .

Fourdee, 2016-09-27:
No ARM64 packages in repo: http://download.webmin.com/download/repository/dists/sarge/contrib/

Will use ```wget http://www.webmin.com/download/deb/webmin-current.deb -O package.deb``` instead

Zone22, 2016-09-27:
Webmin Is pretty robust but I've had to find alternatives in the past. In part because Webmin and some major security flaws that are unresolved and It's bloated. That being said to each their own.

If an alternative is needed check out [Ajenti](http://ajenti.org/) it's very lightweight and runs on Debian, Ubuntu, CentOS/RHEL, FreeBSD (experimental), Raspbian. The icing on top of the cake is the the web UI is a nice modern polished look.

Fourdee, 2016-09-28:
@Zone22 

> If an alternative is needed check out Ajenti.

Looks interesting 👍  Please create a Git ticket and i'll look into it.

closed 2016-09-28 by Fourdee