MichaIng/DietPi #393: Update from v.115 to v.119 don't work

MCosta2016, 2016-06-27:


Fourdee, 2016-06-27:
@MCosta2016 

Hi Marco, thanks for posting here (Twitter is a nightmare with 140 char max lol). :+1: 

If you can get the bug report sent and reply with reference code, when you can, i'd be grateful :)
Link to guide for sending the bug report: http://dietpi.com/phpbb/viewtopic.php?f=8&t=499#p2184

MCosta2016, 2016-06-27:
ad58354b-6571-438d-89fe-670466dcfa40-1 this is the reference code i get from dietpi: i hope i do everithing you need to help me! Thanks.

Fourdee, 2016-06-27:
@MCosta2016 

Thanks for the report. :+1: 
It looks like the same issue as this git ticket: https://github.com/Fourdee/DietPi/issues/390

Please can you try the following commands:
```
sed -i $'s/\r$//' /DietPi/dietpi.txt
dietpi-update
```

Also, i'd appreciate it if you could answer the following questions:
- Did you make any changes to the ```/boot/dietpi.txt``` file before booting for the 1st time?
- Which OS did you write the DietPi image on (eg: Windows 10)?

MCosta2016, 2016-06-28:
Good morning everibody, i' m enjoy because all works well.
After i try the command below, i get immediately update to v.120
Then, i retry ea fresh installation without modify dietpi.txt and all works fine.
So, in order to replay your questio i use win32disk imager to create bootable sd with S:O windows 7 ultimate updated.
But i think that the problem was create by using wordpad to modify dietpi.txt.
thanks a lot, now... now i hava another big quantity of problem to resolve, but  ... one for time.
For the moment ... THANK YOUUUUUU
Marco.

MCosta2016, 2016-06-28:
For example, at reboot appear...
[image]

MCosta2016, 2016-06-28:
ohhhhh plex server don't work 
It is related with the error attached before ?
Transission start
....

Fourdee, 2016-06-28:
@MCosta2016 

> But i think that the problem was create by using wordpad to modify dietpi.txt.

Yep. I've added some code to prevent this occurring from v121 onwards: https://github.com/Fourdee/DietPi/issues/390#issuecomment-228753085

> ohhhhh plex server don't work 

Plex server requires en.US-UTF8 locale to be installed and the default locale. There should be a prompt when installing Plex to apply those changes. 
Please check the active locale with ```locale```. You can change it in dietpi-config > language and regional > locale, then reboot.

Fourdee, 2016-06-28:
@MCosta2016 
https://cloud.githubusercontent.com/assets/20165457/16412688/e5c4a116-3d2d-11e6-9f32-176466b8c520.jpg
This can safely be ignored, but i'll take a look. Are you running desktop via VNC server?

MCosta2016, 2016-06-28:
Are you running desktop via VNC server?  --- Yes

MCosta2016, 2016-06-28:
ok, i replace locale and now it is ok!
Good
Now i have to install internet key lte 4g e392 huawei for internet connection.
then i need to create a bridge between key and eth0 to permit at the entire lan ( pc and AP) to connect to the web
But ... there is a but, i need a data control block from 08.00 am to 00.00 pm to not exceed 1gb by day, free for 00.10 am to 7.50 am.
So i think to use        http://netramon.sourceforge.net/ita/        to got it
What do you think ?
So i have 50/ 60 mb7s downloading speed, 30/40 uploading speed with a low cost.. ..can you help me..? Thanks in advance.

MCosta2016, 2016-06-28:
I would like to use this method...

So for all out there that need to get it working on trisquel (in my case a Huwaei E392 LTE Usb Modem (from TMN Portuguese provider)) here's how:
type in terminal as root (or sudo the commands):
lsusb -v | less
Get the vendor and product ID from the result of the above command.
Like: Bus 003 Device 003: ID 12d1:1505 Huawei Technologies Co., Ltd. E398 LTE/UMTS/GSM Modem/Networkcard
Note the two references after ID (12d1 and 1505)
Now append this line in /etc/modules:
usbserial vendor=0x12d1 product=0x1505
(to edit type "nano /etc/modules" as root, add the lines, hit ctrl+o to write the file (save) and ctrl+x to close nano.)
Reboot your computer and your done.
Set up a new mobile broadband connection and your online! ;-)

.. but where i can setup new mobile connection ?
Thanks

Fourdee, 2016-06-28:
@MCosta2016 

> .. but where i can setup new mobile connection ?

This is not a feature we support in DietPi at the moment (I don't own a mobile/sim based device to test/implement it). So you would need to check online guides for setting this up.

As DietPi soley uses the ```/etc/network/interfaces``` file for lightweight networking, I think "Installation with PPP and /etc/network/interfaces" would be a good place to start: https://wiki.debian.org/Modem/3G#Installation_with_PPP_and_.2Fetc.2Fnetwork.2Finterfaces. This is for Wheezy, but it shouldn't be much different in Jessie.

MCosta2016, 2016-06-28:
Ok, i try to understand, but i'm in difficult because i have the translation correct to made before..
Don't worry, but what do you think about...http://netramon.sourceforge.net/ita/
I have to use gedit to install .deb package ?
many thanks

Fourdee, 2016-06-30:
@MCosta2016 

> I have to use gedit to install .deb package ?

To install deb packages:
```dpkg -i package.deb```

Fourdee, 2016-06-30:
As the original issue has now been resolved, i'll mark this as closed.

closed 2016-06-30 by Fourdee

MCosta2016, 2016-07-01:
OK Thank you.