MichaIng/DietPi #394: bluetoothctl | Broken during update from v119 to v120

GaM3r2Xtreme, 2016-06-27:
I've been using the interactive command bluetoothctl in order to have an expect script automate a connection with a Wii U Pro Controller. This worked great on a RPi 1 B+ paired with a bluetooth dongle, but once I ran an update to the v120 firmware, this command no longer responds when executed manually. It seems like the program runs, but any commands given can not be seen or passed.

I have noticed in the change log for v120 that bluetooth is disabled by default, but enabling it through dietpi-config has not fixed the issue. An apt-get upgrade has been ran in order to make sure bluetoothctl is at the latest version (currently 5.23).

I have a RPi 3 still on v119 with bluetoothctl v5.23 which works fine. I will try to update the RPi 3 to v120  of DietPi and see if the issue persist.

This issue occurs on a fresh install from the v115 DietPi .img file after all updates have been completed. I will try to flash using the latest v120 DietPi .img file to see if this issue persist.

I should also note that this issue occurs in both an SSH environment (using Dropbear) and an HDMI display.

Fourdee, 2016-06-27:
> I have noticed in the change log for v120 that bluetooth is disabled by default, but enabling it through dietpi-config has not fixed the issue.

@GaM3r2Xtreme 

Thanks for the report.

On the RPi3 running v120, Can you please paste the results of the following:
```
lsmod
cat /etc/modules
cat /etc/modprobe.d/*
```

Can you try running the following commands please aswell:
```
/DietPi/dietpi/func/dietpi-set_wifi_bt bluetooth 0
/DietPi/dietpi/func/dietpi-set_wifi_bt bluetooth 1
bluetoothctl
```

Then ```power on```

GaM3r2Xtreme, 2016-06-28:
Alright, so I've gone ahead and flashed a card using the latest .img file and popped it into the RPi 3. DietPi v120 on RPi 3 works as intended for the bluetoothctl command. This is after bluetooth is enabled within dietpi-config. I have attached the output you requested coming from v120 on the RPi 3 as [RPi3_output.txt](https://github.com/Fourdee/DietPi/files/336318/RPi3_output.txt) (github keeps causing issues with the text format).

Looks like flashing with the latest DietPi_v120_RPi-armv6-(Jessie).img file for the RPi 1 did the trick too. The only difference I notice is that the text color changes do not appear on HDMI output while a PuTTY connection does (typically bluetoothctl is output as blue text). Attached in [RPi1_output.txt](https://github.com/Fourdee/DietPi/files/336321/RPi1_output.txt) is the output of the given commands on the RPi 1 running DietPi v120.

I can only conclude by using DietPi_v115_RPi-(Jessie).img to update up to DietPi v119, then running the dietpi-update in order to move from v119 to v120 caused bluetoothctl to break on the RPi 1. Sadly, I can no longer give any output from this scenario due to the card re-flash executed above.

If it is at all possible to pause an update at v119 while using the older .img file, I wouldn't mind running a few test starting from this old .img and working back up to the issue. Just give me a heads up if this is needed.

As far as I'm concerned, using the latest .img will solve this issue. Thanks for the help.

Fourdee, 2016-06-28:
@GaM3r2Xtreme 

> I can only conclude by using DietPi_v115_RPi-(Jessie).img to update up to DietPi v119, then running the dietpi-update in order to move from v119 to v120 caused bluetoothctl to break on the RPi 1. Sadly, I can no longer give any output from this scenario due to the card re-flash executed above.

Interesting, thanks for results.

There is a way of updating from v115 to v119, but it will require manual input. I'll leave this ticket open and see if I can replicate the issue.

> As far as I'm concerned, using the latest .img will solve this issue. Thanks for the help.

Excellent :+1:

Fourdee, 2016-06-28:
v115 > v119 test. Appears working:
```
root@DietPi:~# bluetoothctl
[NEW] Controller B8:27:EB:24:F9:F1 DietPi [default]
[bluetooth]#
```

v119 > v120:
```
root@DietPi:~# bluetoothctl
[NEW] Controller B8:27:EB:24:F9:F1 DietPi [default]
[bluetooth]#
```
Status in dietpi-config = Enabled

I can't seem to replicate the issue on a fresh image, but thats not to say it doesn't exist. It could be some weird and wonderful installation combination or settings while you were on v119, that may of interfered with the patch system. 

Either way, as you have resolved the issue with latest image install, I'll mark this as closed for now. But if anyone experiences a similar issue, hopefully we can run some commands and try to debug it.

closed 2016-06-28 by Fourdee