MichaIng/DietPi #396: DietPi doesn't connect to WiFi when network goes offline for a moment

artuross, 2016-06-27:
Hey,

I noticed that DietPi won't connect to WiFi if the network is not available at the time of boot.

I just installed this distro on my Raspberry Pi 3, there is no software installed, only basic configuration. I have configured WiFi (through Ethernet) and it has connected to it (I saw traffic on my router). Then I rebooted Pi, also worked. Then I rebooted router, and there was no IP lease on DHCP. I double checked that by connecting through Ethernet, WiFi was not connected (config was saved though).

Can you please help me fix that? I assume this is wrong/incomplete configuration, but my knowledge on this topic is very minimal.

Fourdee, 2016-06-28:
@artuross 

Hi, did you use dietpi-config to change network settings?

Can you paste the results of the following please (hide any wifi keys):
```
cat /etc/network/interfaces
ifconfig -a
route
dpkg -l | grep -ci -m1 netplug
lsmod
```

artuross, 2016-06-28:
Hey, yes, I did use dietpi-config.

I just noticed that 2 of 3 leased IP from DHCP, I guess `ntp` might have run and it forced networking to connect to WiFi.

Still, I'd love to have them connecting to network as soon as there is one (or checking for available network every few minutes).

cat /etc/network/interfaces
```
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet dhcp
metric 0
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 8.8.8.8 8.8.4.4

# Wifi
allow-hotplug wlan0
iface wlan0 inet dhcp
metric 1
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
wireless-essid DOM
wireless-key password
wireless-mode Managed
wireless-power off
wpa-ssid DOM
wpa-psk password
#dns-nameservers 8.8.8.8 8.8.4.4
```

ifconfig -a
```
eth0      Link encap:Ethernet  HWaddr b8:27:eb:3f:70:f3
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:9456 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9456 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:766976 (749.0 KiB)  TX bytes:766976 (749.0 KiB)

wlan0     Link encap:Ethernet  HWaddr b8:27:eb:6a:25:a6
          inet addr:192.168.1.7  Bcast:192.168.1.31  Mask:255.255.255.224
          inet6 addr: fe80::ba27:ebff:fe6a:25a6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:74857 errors:0 dropped:10 overruns:0 frame:0
          TX packets:336 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:16480513 (15.7 MiB)  TX bytes:55825 (54.5 KiB)
```

route
```
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.1.1     0.0.0.0         UG    1      0        0 wlan0
192.168.1.0     *               255.255.255.224 U     0      0        0 wlan0
```

dpkg -l | grep -ci -m1 netplug
```
1
```

lsmod
```
Module                  Size  Used by
brcmfmac              186343  0
brcmutil                5661  1 brcmfmac
cfg80211              427855  1 brcmfmac
rfkill                 16037  2 cfg80211
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
fuse                   83461  1
ipv6                  347530  22
```

Fourdee, 2016-06-29:
@artuross 

Thanks for the results. All looks in order.

> Then I rebooted router, and there was no IP lease on DHCP
Still, I'd love to have them connecting to network as soon as there is one (or checking for available network every few minutes).

So if I'am reading this right, WiFi is not reestablishing connection, when:
- the router (SSID) drops and comes back on again.
- SSID wasn't available during boot, but is later on.

Let me know if this is correct and i'll see if we can sort something.

artuross, 2016-06-29:
Yes, this is correct, connection is not established for both of pointed scenarios.

Fourdee, 2016-07-01:
> Notes to self: 
ping gateway to check AP is online and connection is valid?

@artuross 
Can you try the following from this page and see if you get the same result: http://raspberrypi.stackexchange.com/questions/4120/how-to-automatically-reconnect-wifi

> Go to /etc/ifplugd/action.d/ and rename the ifupdown file to ifupdown.original
Then do: cp /etc/wpa_supplicant/ifupdown.sh ./ifupdown
Finally: sudo reboot
That's all. Test this by turning off/on your AP; you should see that your Raspberry Pi properly reconnects.

artuross, 2016-07-01:
Works, thanks a lot.

closed 2016-07-01 by artuross

9H5G, 2017-03-09:
This does not work for me.  //etc/ifplugd/ contains action_wpa only