MichaIng/DietPi #398: DietPi-Config | Add display options for Pine A64

Fourdee, 2016-06-28:
> http://forum.pine64.org/showthread.php?tid=1068&pid=13760#pid13760

Fourdee, 2016-06-28:
@rhkean 
Would appreciate your input on this if you can :+1:. Is changing display resolution even possible at the moment?

rhkean, 2016-06-28:
@Fourdee 
I believe that I saw some threads on the pine forum regarding this, but I didn't get a chance to follow up on it, so I'm not 100% sure

Fourdee, 2016-07-01:
uEnv.txt
```
optargs=disp.screen0_output_mode=720p60
```

List of available options: https://github.com/longsleep/sunxi-disp-tool#available-hdmi-output-names

Fourdee, 2016-07-01:
Gonna leave the 720p/1080p's out, use the refresh rate options instead.
[image]

Fourdee, 2016-07-01:
Done, available for v123. Tested 1080i60 720p50 720p60 and 1080p60. I'll test 4K in a few years, when they are affordable and the world catches up with the extra bandwidth lol

closed 2016-07-01 by Fourdee