MichaIng/DietPi #399: DietPi-Software | Kodi will not run.

Fourdee, 2016-06-28:
> http://dietpi.com/phpbb/viewtopic.php?f=11&t=504&p=2205#p2205

Fourdee, 2016-06-29:
Due to shift from fbdev to X11 with recent mali driver release. Many thanks to Meveric for his help on this one:
> Changes for X11 installing xf86-video-mali and libump-odroid aside of mali450-odroid and a new set of aml-libs-odroid. 
xorg.conf for /etc/X11

rhkean, 2016-06-29:
so, is it fixable?

Fourdee, 2016-06-29:
@rhkean Yep, just testing now

Fourdee, 2016-06-29:
notes:
- :u55b6: boot.ini ```setenv vout "dvi"``` now disables hdmi audio. So this must now be commented out.
- :u6307: Kodi 32bit works from console.
- :u55b6: Kodi 24bit from console visually appears slower and stutters. 32bit is alot smoother.
- :u5408: LXDE 24bit fails on xorg.
- :u5408: LXDE 32bit desktop fine. Kodi stutters and flashes in menus. Desktop visible in video playback. [image]

Fourdee, 2016-06-29:
- 32bit is now default. Required for optimal performance in Kodi and allows for working Desktops.
- ```setenv vout "dvi"``` is now commented out by default to allow HDMI audio.

Fourdee, 2016-06-29:
The above commit should cover patching. But i'll test it tomorrow from 120 master with Kodi+LXDE installed > 121 testing.

Reason for boot.ini overwrite:
Patch would be risky due to far too many additions and changes and I honestly don't have the time to implement it. Also contains the new "nographics" (headless) option which i'll add to dietpi-config at a later date.

Fourdee, 2016-06-30:
Patch tests completed. Resolved for v121.
Marking as closed :)

closed 2016-06-30 by Fourdee