MichaIng/DietPi #402: [DietPi Update Error] - Obtained server value from github is invalid!

bhaveshgohel, 2016-06-30:
Hello @Fourdee ,

Today when i update my Raspberry Pi I get the bellow error message. I tried again but same error.

[image]

I'll be happy to provide more info if needed. :smiley:

Fourdee, 2016-06-30:
@bhaveshgohel 
Thanks for the report.

> I'll be happy to provide more info if needed.

Excellent :+1:  
I mainly need to see ```/DietPi/dietpi.txt```, but, its probably best to send us a bug report as this will upload the majority of DietPi settings including dietpi.txt: http://dietpi.com/phpbb/viewtopic.php?f=8&t=499

bhaveshgohel, 2016-06-30:
@Fourdee : Bug Report Sent! :+1: 

Where can i send you the "Reference CODE"?

Fourdee, 2016-06-30:
@bhaveshgohel 
On here will be good :+1:

bhaveshgohel, 2016-07-13:
@Fourdee : I'm planning to backup this OS and completely install the latest one. So if anything is there i can switch between that OS and we can debug it. :)

Fourdee, 2016-07-23:
Assuming issue has not occurred on fresh install. Marking as closed, please reopen if required.

closed 2016-07-23 by Fourdee