MichaIng/DietPi #405: DietPi Update 120 to 121 on Odroid-XU4 bricks the system because of C2 kernel

Tzimovo, 2016-06-30:
Hi,
unfortunately the latest update bricks my Odroid-XU4 because for some reason the update installs at least the Odroid-C2 UBoot config (in boot.txt ODROIDC2-UBOOT-CONFIG). Not sure if the kernel is also from the C2, at least the system does no longer start at all and the heartbeat led does not show any activity.
Is there anything I can try to prevent this?
Cheers and keep up the good work on this excellent distro!
Uwe Schurig

rhkean, 2016-06-30:
Please change the title.  You can't [brick](https://en.wikipedia.org/wiki/Brick_(electronics)) an arm sbc.

Based on your description, it appears that the `boot_c2.ini` file got copied over your `boot.ini`

pull the xu4 file from [here](https://raw.githubusercontent.com/Fourdee/DietPi/master/boot_xu4.ini) and save it to the FAT partition of the uSD card as `boot.ini`

That should get you booted up.  In the meantime, we'll look into correcting the problem.

rhkean, 2016-06-30:
@Fourdee 
I wasn't following closely as you were addressing #399.  Was that an Odroid-C2 specific fix?

It looks like the v121 patch file needs a `$HW_MODEL` test [HERE](https://github.com/Fourdee/DietPi/blob/master/dietpi/patch_file#L957-993).

Fourdee, 2016-06-30:
@Tzimovo
Many thanks for bringing this to our attention :+1: 

@rhkean  
> It looks like the v121 patch file needs a $HW_MODEL test HERE.

Yep, major f*** up on my part. So apologies to those users this has affected.

I've released v122 which will stop this from occurring. Existing installations i'll address here in a minute.

Fourdee, 2016-06-30:
####If your system has been affected by this patch, please do the following:
- Power off system and plug SD/EMMC media into a USB capable reader/writer.
- Download one of the the following file for your device:
 - XU4 = https://raw.githubusercontent.com/Fourdee/DietPi/master/boot_xu4.ini
 - C1 = https://raw.githubusercontent.com/Fourdee/DietPi/master/boot_c1.ini
- Rename the file to ```boot.ini``` then place that file into ```/boot/boot.ini```.
- Plug SD/EMMC back into device and power on.

If you still have issues, please post here.

Tzimovo, 2016-06-30:
@Fourdee Gonna try this out now. 
@rhkean According to your linked article it would even fit HARD BRICKED because not even the heartbeat was there, fan not moving. But well, the changed description fits too. :P

Fourdee, 2016-06-30:
> HARD BRICKED

Bricked or not, its irreverent. This should never of happened and I can only apologise to those it has affected. 
I'll be changing the way we do releases from now on to prevent this from occurring in the future.

> Gonna try this out now. 

Let us know if you have any issues. :+1:

rhkean, 2016-06-30:
just to explain.... mainly because the term _bricked_ carries with it the connotation of _not fixable_, and it's probably best to not have that in the title of a bug report

it's not considered bricked, because there is no firmware to corrupt.  u-boot based systems load everything from disk... including the firmware.  If the disk gets corrupt or goes bad, it's not considered bricked.  It's considered corrupted or damaged.  Replace the disk and reboot.  (ex.  you wouldn't consider a Windows computer with a bad C: drive to be bricked... it's just got a bad hard drive.  Replace the drive, reinstall Windows and reboot)

Soft bricked devices typically boot partially before crashing... meaning that they get to the 2nd boot stage (or further) before the firmware crashes.  So, if you can interrupt the boot process at an early enough stage, you can repair it.

Hard bricked devices typically require a JTAG to interrupt the boot process at (or before) the 1st stage.  This is a physical interruption where a specific signal is applied to a specific pin on the CPU and puts the CPU into a special diagnostic state.  Since many (if not most) production boards (phone, tablets, etc) don't include JTAG circuitry, the cost to get a professional to trace out the JTAG pin-out, build a harness to implement a JTAG diagnostic boot, and repair the damaged firmware, renders it a not worth-while expense.... hence the term, brick.... meaning this rectangular electronic device isn't worth much more than a brick now...

Since most, if not all, SBC boot exclusively from a removable flash block device, they aren't considered bricked.... just swap out the removable device with a functioning one (or rewrite the image) and you're ready to boot again.

Tzimovo, 2016-06-30:
I think you can safely close this one. The XU4 is happily up and running again! 

As a sidenote: the automatic installer did seem to work fine too - apart from me having to log in once and start it - which could have been because of the interruption dueue of the incorrect ini file. I'll verify that again tomorrow with another SD card.

So the XU4 is definitely "un-bricked" again in the very words of the wikipedia article. :) :) 

Anyways a big thumbs up for the both of you for a reaction time that definitely beats that of most major manufacturers and a proper fix!

Fourdee, 2016-07-01:
@Tzimovo 
> The XU4 is happily up and running again!

Excellent. Thanks for getting back to us :+1: 

Closing ticket. Please reopen if required.

closed 2016-07-01 by Fourdee

k-plan, 2016-07-03:
@Fourdee 

Update v.122 break my display output ...

- write `DietPi_v120_OdroidC2-arm64-(Jessie).img` on sd-card
- edit `boot.ini` file, so it will work with my LCD monitor
  
```
#Modifcations to /boot/* files, will not be saved when powered on.
#Please ensure you edit from the DietPi-RamDisk location /DietPi/*

ODROIDC2-UBOOT-CONFIG

#Resolution + Refresh rate
#setenv m "720p60hz"
# setenv m "sxga"           ## (SXGA) work only on C1+, but NOT on C2  => P19 Monitor
setenv m "1280x1024p60hz"   ## (SXGA) work on C2, but NOT on C1+  => P19 Monitor

# HDMI BPP Mode | 24bit to ensure Kodi and Xserver both work, whilst we wait for X11 drivers
setenv m_bpp "24"

# HDMI DVI/VGA modes
# Uncomment only a single Line! The line with setenv written.
# At least one mode must be selected.
setenv vout "dvi"  ## DVI-to-HDMI cable (SXGA) work on C2 and C1+  => P19 Monitor
# setenv vout "vga"  ## DVI-to-HDMI cable (SXGA) work only on C2        => P19 Monitor

 . . . .
```
- booting up with the new image on sd-card
- auto updating from v.119 to v.122 running on first boot
- auto reboot
- darkness on my LCD monitor (no signal ...)

Can't see any more, if C2 is booting or `DietPi-Automation` is running and working ...

Connect via ssh, but in this way I can't see what is running. So I break the automated installation process...