MichaIng/DietPi #407: Xfce | Update Request

yunussandikci, 2016-07-01:
Hi again. i just installed v121 and xfce. But it is little ugly and old. i tried some themes on apparence but nothings changed(i think it's a bug) and i figured out it was the version 4.10 that relased 4 years ago. Can we update it to 4.12? and how can we change theme? Thanks, nice distro :)

yunussandikci, 2016-07-02:
Why dont you tell anything :(

Fourdee, 2016-07-03:
@yunussandikci 
> Why dont you tell anything :(

Because I have a real life outside of DietPi that needs maintenance. lol ;)

Ok, so, 4.12 is not available in the Jessie (stable) repo.

Few options:
- Upgrade your installation from Jessie (stable) to Stretch (testing). Untested and I cant guarantee DietPi will still function, let alone the device.
- Compile 4.12 from source.
- Find another Repo that offers 4.12 arm binaries. Possibly Ubuntu, but expect dependency issues that would need to be resolved.

k-plan, 2016-07-03:
sorry, but what is the meaning of "_**diet**_" in "_**DietPi**_"?

For me, it's not to have the possibility for 4 desktop environment like:
- [LXDE - Highly optimised lightweight desktop without the "bloatware".](http://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p42)
- [Mate - Popular Gnome 2 desktop.](http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=60#p2073)
- [GnuStep - Alternative/unique desktop.](http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=60#p2072)
- [XFCE Desktop - Lightweight desktop environment.](http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=70#p2203)

No harm meant, just me 2 cent.

Fourdee, 2016-07-05:
Sorry @yunussandikci , but i'am going to mark this as closed. I just don't have the time to compile/test, and support XFCE 4.12 at the moment. I know 4.10 is "old", but it is classed as "stable".

If anyone comes across a 4.12 repo with ARM binaries, or is willing to compile this for ARM devices, i'am all ears and please reopen the ticket.