MichaIng/DietPi #408: replace php5-mysql with php5-mysqlnd when installing mariadb

rhkean, 2016-07-03:
phpmyadmin displays this error message when using php5-mysql client to access a MariaDB database:
`Your PHP MySQL library version 5.5.44 differs from your MySQL server version 10.0.20. This may cause unpredictable behavior`

per this [post](http://dba.stackexchange.com/questions/109072/your-php-mysql-library-version-5-5-44-differs-from-your-mysql-server-version-10), it can be resolved by using the native driver client instead.

I have tested and the error goes away.  In theory, we could use the native driver for both MySQL and MariaDB, and it should make for a smaller install footprint, because it looks like the native driver package is 300kb compared to 150kb, but it doesn't require the 3mb libmysqlclient18 package.

Fourdee, 2016-07-05:
@rhkean 

Just want to confirm php5 packages to install:
MYSQL = php5-mysql
MARIADB = php5-mysqlnd
?

rhkean, 2016-07-05:
I have tested that combination... and yes.

I was going to test php5-mysqlnd with MYSQL also, because I think it will shrink the footprint, but I didn't get a chance to this weekend.

Fourdee, 2016-07-08:
Tested ok with Lighttpd, mariadb, php5-mysqlnd, phpmyadmin and owncloud. Marking as closed:
```
root@DietPi:~# dpkg -l | grep php5 | grep mysql
ii  php5-mysqlnd                   5.6.22+dfsg-0+deb8u1      arm64        MySQL module for php5 (Native Driver)
```

closed 2016-07-08 by Fourdee