MichaIng/DietPi #409: DietPi-Cloudshell | Support for Odroid C1 / C1+  / C0   (and RPi ?)

k-plan, 2016-07-03:
Hi,

is it possible to integrate all needed drivers (e.g. [notro/fbtft](https://github.com/notro/fbtft) ...) and configurations in DietPi, so it is possible to use DietPi-Cloudshell with Odroid C1 / C1+  / C0 and [Hardkernel 3.2inch TFT+Touchscreen Shield](http://www.hardkernel.com/main/products/prdt_info.php?g_code=G142060963922) by default?
Enable/disabling this feature support by `dietpi-config` will be very nice.

Hardkernel 3.2'' Display is a: "Wave Share SpotPear 3.2inch RPi LCD V4"

[image]

http://www.waveshare.com/3.2inch-rpi-lcd-b.htm

Info and Downloads: http://www.waveshare.com/wiki/3.2inch_RPi_LCD_(B)

- [Schematic](http://www.waveshare.com/wiki/File:3.2inch-RPi-LCD-B-Schematic.pdf)
- [User manual](http://www.waveshare.com/w/upload/3/3e/RPi-LCD-User-Manual.pdf)

Maybe it is possible to support RPi as well.

- [Driver for Raspbian](http://www.waveshare.com/wiki/3.2inch_RPi_LCD_(B)#Driver)

Thanks in advance.

Fourdee, 2016-07-05:
@k-plan 
Excellent idea. Same resolution as cloudshell screen 320x240 aswell.

My only concern is these screens are not HDMI. I read somewhere that some of these screens will only display during X session and not framebuffer. Not sure if thats true or not, but having one of these devices to impliment/test would be fun. 

Only downside is this requires a kernel driver, so i'd need to compile those kernels and maintain them for each device we add. For the C1/C2 i'll speak to meveric and see if hes willing to add the driver to his kernel.

> edit: - if you will do, I will spend you 20,- €, so you will be able to buy it - amazon.uk

Thats very kind of you. Do you have one of these screens at the moment? If so, we could do a quick test to see if it will work on framebuffer. Just let me know and i'll compile the kernel.
Found a RPI forum thread regarding this screen: https://www.raspberrypi.org/forums/viewtopic.php?f=100&t=98069

k-plan, 2016-07-05:
@Fourdee 

>Do you have one of these screens at the moment? 

Yes.

>My only concern is these screens are not HDMI.

Me too .... :smile: 

> I read somewhere that some of these screens will only display during X session and not framebuffer. 

Have tested it with Odroid C1+
- work out-of-box with Hardkernels Ubuntu 14.04 image
- can't get it on work with [Auto start console mode](http://odroid.com/dokuwiki/doku.php?id=en:c1_tft_autoconsole) 
- doesn't work with DietPi on C1+ 

> If so, we could do a quick test to see if it will work on framebuffer. 

Let*s do so. Any suggestions? On RPi B+ or Odroid C1+?

Thanks!

Fourdee, 2016-07-05:
> Let*s do so. Any suggestions? On RPi B+ or Odroid C1+?

Ok, found the driver for RPi: http://www.waveshare.com/w/upload/2/24/LCD-show-160520.tar.gz

I'll go through it and get back to you with steps needed to "hopefully" get it working :)

Fourdee, 2016-07-08:
@k-plan , this is for the 3.2 screen:

First, backup your system. So you can restore as needed.
```
dietpi-backup 1
```

Add these to ```/DietPi/config.txt```
```
start_x=1  # This is for camera, so not sure why its in here, but lets test it
gpu_mem=128
dtparam=spi=on
dtoverlay=waveshare32b:rotate=270
dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,swapxy=0,pmax=255,xohms=60,xmin=200,xmax=3900,ymin=200,ymax=3900
dtparam=i2c_arm=on
dtoverlay=w1-gpio-pullup,gpiopin=4,extpullup=1
```

Overwrite ```/boot/cmdline.txt``` with:
```
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
```

Put this file into ```/boot/overlays``` http://dietpi.com/downloads/testing/waveshare35a-overlay.dtb

k-plan, 2016-07-08:
@Fourdee 

Testing on RPi3, starting from the scratch with `DietPi_v120_RPi-armv6-(Jessie).img` on another sd-card:

```
root@RPi-3_Display:~# cat /DietPi/config.txt

#Modifcations to /boot/* files, will not be saved when powered on.
#Please ensure you edit from the DietPi-RamDisk location /DietPi/*

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# Set to 1 if your display has a black border of unused pixels visible
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
framebuffer_width=1280
framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#-------RPi Camera Module-------
start_x=0
disable_camera_led=0

#-------GPU memory Splits-------
gpu_mem_256=16
gpu_mem_512=16
gpu_mem_1024=16

#-------Max USB Current -------
max_usb_current=1

#-------Disable Ram temp refresh every 500ms---------------
disable_pvt=1

#-------Disable Pi Splash screen on boot-------------
disable_splash=1

#-------Remove dedicated pll for PWM audio -------
avoid_pwm_pll=1

#-------SoundCard-------
dtparam=audio=off

#-------i2c-------------
dtparam=i2c_arm=off
dtparam=i2c1=off
i2c_arm_baudrate=100000

#-------Overclock-------
temp_limit=75
initial_turbo=20
force_turbo=0

#over_voltage=0
#arm_freq=1200
#core_freq=400
#sdram_freq=400

#arm_freq_min=700
#core_freq_min=250
#sdram_freq_min=400

#----- WaveShare 3.2 Display ------
start_x=1  # This is for camera, so not sure why its in here, but lets test it
gpu_mem=128
dtparam=spi=on
dtoverlay=waveshare32b:rotate=270
dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,swapxy=0,pmax=255,xohms=60,xmin=200,xmax=3900,ymin=200,ymax=3900
dtparam=i2c_arm=on
dtoverlay=w1-gpio-pullup,gpiopin=4,extpullup=1

#Note To Self, NEVER enable L2 cache, breaks most X based applications that were not compiled with L2 cache enabled.
```

```
root@RPi-3_Display:~# cat /boot/cmdline.txt

dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=noop rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
```

```
root@RPi-3_Display:~# ls -lah /boot/overlays | grep wave

-rwxr-xr-x 1 root root 2.1K Jul  8  2016 waveshare32b-overlay.dtb
```

Result:

[image]

Fourdee, 2016-07-08:
@k-plan 

Thanks for the results, although, white screen of brilliance lol

Looks like I missed something, i'll take another look and let you know. 
Very tempted to buy the screen, but, i've got a BananaPi Pro sample board arriving soon. Yes its free, but it isn't, minimal UK customs import tax of £8 and "handling fee" of £7-8 :-1:

k-plan, 2016-07-08:
@Fourdee 

> Yes its free, but it isn't, minimal UK customs import tax of £8 and "handling fee" of £7-8 👎

Sorry, but [image]

> Very tempted to buy the screen, but,  ...

Don't trouble your head about it. 
Until it possibly work, I will do this : 
[image]

http://dietpi.com/phpbb/viewtopic.php?f=9&t=524

😃

Fourdee, 2016-07-09:
@k-plan 

> For a customs tariff number declared "developer sample" ? Sorry, but

Yep, minimal fee of any imported package from outside EU. Theres also a 20% VAT charge on the declared total price of the items, if its over £15 lol. 

Looks epic and excellent guide :+1: , I've got to give this a go.

k-plan, 2016-07-09:
@Fourdee 

> Looks like I missed something, i'll take another look and let you know. 

Yes, it seem so.

Do another test on RPI2 B+ staring from scratch.

Use `root@RPi2_Display:~# wget http://www.waveshare.com/w/upload/2/24/LCD-show-160520.tar.gz`
and copy file over by hand : 

```
root@RPi2_Display:~# cat LCD-show/LCD32-show
## sudo mkdir /etc/X11/xorg.conf.d
sudo cp ./waveshare32b-overlay.dtb /boot/overlays/waveshare32b.dtbo
## sudo cp -rf ./etc/X11/xorg.conf.d/99-calibration.conf-32  /etc/X11/xorg.conf.d/99-calibration.conf
## sudo cp -rf ./usr/share/X11/xorg.conf.d/99-fbturbo.conf  /usr/share/X11/xorg.conf.d/
sudo cp ./cmdline.txt /boot/
sudo cp ./inittab /etc/
sudo cp ./boot/config-32.txt /DietPi/config.txt
## sudo reboot
```
and edit 

```
root@RPi2_Display:~# cat /etc/modules
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

snd-bcm2835
i2c-bcm2708
i2c-dev
fbtft_device name=flexfb gpios=dc:22,reset:27 speed=48000000
flexfb width=320 height=240 buswidth=8 init=-1,0xCB,0x39,0x2C,0x00,0x34,0x02,-1,0xCF,0x00,0XC1,0X30,-1,0xE8,0x85,0x00,0x78,-1,0xEA,0x00,0x00,-1,0xED,0x64,0x03,0$
ads7846_device model=7846 cs=1 gpio_pendown=17  speed=1000000 keep_vref_on=1 swap_xy=0 pressure_max=255 x_plate_ohms=60 x_min=200 x_max=3900 y_min=200 y_max=3900
```
No sure, if we need all that stuff ....

But, after reboot:

[image]

[image]

Issue:

- not all screen fits perfect to the display (s.u.)
- after a few minutes display go to sleep and only a reboot can wake it up ...

 
[image]

Fourdee, 2016-07-09:
@k-plan 

Great work :+1: !. Looks awesome 

> after a few minutes display go to sleep and only a reboot can wake it up ...

The screen blank issue, few resolutions:
- Run ```setterm --blank 0 --powersave off``` on the physical system, before running dietpi-cloudshell. I've been unable to get this to work over SSH. Seems it must originate from the screen you want to set powersaving for.
- use ```dietpi-autostart``` and select ```dietpi-cloudshell``` as the option. This runs the above command automatically during login.
- I think its possible to add this to ```/boot/cmdline.txt```, but, cant remember, i'll take a look

> not all screen fits perfect to the display (s.u.)

The orange bars look fine. I think its an issue with the board description being too long to fit on same line. I'll take a look and try to move the board description to the next line.

Fourdee, 2016-07-09:
@k-plan 
Yep: http://forum.odroid.com/viewtopic.php?f=138&t=19948&p=138189&hilit=blank#p138189

Try adding ``` consoleblank=0``` to ```/boot/cmdline.txt```

k-plan, 2016-07-10:
@Fourdee 

> - use dietpi-autostart and select dietpi-cloudshell as the option. This runs the above command automatically during boot and auto login.

Yes, this will works on RPi B+ 😄 

💡  - I like the idea that the display is going off after some time. Don't need it, if I am not at home and it will save power consume as well. When I need it back, taping on screen or push on of the three buttons to wake it up, will be nice. If you have added a keyboard to the RPi, a keystroke will wake display up (tested). But so you need more power consuming hardware.

---------------------

Do some testing on RPi3 with 3.2 display, but can not get it on work with DietPi v122. 😢 

In principle it will work on RPi3. Take the WaveSpear [3.2inch Rpi LCD (B) Raspbian Image --- Raspberry Pi 3 Model B](https://drive.google.com/open?id=0B8Xjwnqc7XAMdDhlWGtMdGJfVDA) and make a "`fake DietPi`" out of it. This will then work:

[image]

[image]

```
root@RPi3_ws-display:/home/pi# cat /proc/version
Linux version 4.1.19-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #858 SMP Tue Mar 15 15:56:00 GMT 2016

root@RPi3_ws-display:/home/pi# lsmod
Module                  Size  Used by
fb_ili9340              3369  1
fbtft                  27484  1 fb_ili9340
syscopyarea             2773  1 fbtft
sysfillrect             3299  1 fbtft
sysimgblt               1921  1 fbtft
fb_sys_fops             1157  1 fbtft
joydev                  9024  0
ads7846                11138  0
hwmon                   2684  1 ads7846
evdev                  10278  3
spi_bcm2835             7074  0
bcm2835_rng             1763  0
bcm2835_gpiomem         2860  0
w1_gpio                 3401  0
wire                   24703  1 w1_gpio
cn                      4258  1 wire
uio_pdrv_genirq         2944  0
uio                     7753  1 uio_pdrv_genirq
i2c_dev                 5671  0
i2c_bcm2708             4920  0
snd_bcm2835            19802  0
snd_pcm                73474  1 snd_bcm2835
snd_timer              18848  1 snd_pcm
snd                    50779  3 snd_bcm2835,snd_timer,snd_pcm
fuse                   80694  1
ipv6                  338660  40
```

Yes, I know, RPi's a the most popular devices for DietPi, but I am for miles away, to get it on work with Odroid C1+  ...  😢

Fourdee, 2016-07-13:
@k-plan 

Something turned up today which I dont remember ordering lol. I think i have you to thank for sending this? ? :+1: :dancer: 

[image]

Fourdee, 2016-07-13:
Cant get it to work with X11 yet.

@k-plan 
Heres installation code on a default dietpi installation:
```
#Waveshare 32 LCD screen -----------------------------------------------------------------------
#RPI

wget http://www.waveshare.com/w/upload/2/24/LCD-show-160520.tar.gz -O package.tar
tar xzvf package.tar
rm package.tar

cd LCD*
cp waveshare32b-overlay.dtb /boot/overlays/waveshare32b.dtbo

# cp cmdline.txt /boot/
							#consoleblank=0 untested
sed -i 's/rootwait /rootwait consoleblank=0 fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo /' /boot/cmdline.txt # Change this to check for entries, if dont exist, add to end of existing line.

# cp boot/config-32.txt /boot/config.txt
# https://www.diffchecker.com/nnrq4lxy
# - exists in config.txt
sed -i "/dtparam=i2c_arm=/c\dtparam=i2c_arm=on" /DietPi/config.txt

# - These dont exist in our config.txt
cat << _EOF_ >> /DietPi/config.txt

#Waveshare 32 LCD
dtparam=spi=on
dtoverlay=waveshare32b:rotate=270
dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,swapxy=0,pmax=255,xohms=60,xmin=200,xmax=3900,ymin=200,ymax=3900
dtoverlay=w1-gpio-pullup,gpiopin=4,extpullup=1

_EOF_

# cp inittab /etc/ #Wheezy only

#X11 support
dpkg -i xinput-calibrator_0.7.5-1_armhf.deb

# - requires FBturbo driver
apt-get install -y xserver-xorg-video-fbturbo
mkdir -p /etc/X11/xorg.conf.d
cp -rf etc/X11/xorg.conf.d/99-calibration.conf-32  /etc/X11/xorg.conf.d/99-calibration.conf
cp -rf usr/share/X11/xorg.conf.d/99-fbturbo.conf  /usr/share/X11/xorg.conf.d/

#Waveshare 32 LCD screen -----------------------------------------------------------------------
```

Fourdee, 2016-07-13:
@k-plan 
Resolved device being displayed on 2 lines.
[image]

k-plan, 2016-07-14:
@Fourdee 
> I think i have you to thank for sending this? ?

Sorry, yes - guilty.  :smile: 

> Something turned up today which I dont remember ordering

Sorry, but can't order it from amazon.uk, because of a missing credit card.
Comes it from here? [SB Components Ltd.](https://shop.sb-components.co.uk/collections/waveshare/products/3-2-inch-resistive-touch-screen-tft-xpt2046-lcd-lcm-custom-raspbian-system-co)
It is the only shop I can find in GB, who has it in stock and was able to pay it from here and delivered to a different address in GB.

Is it a real "Wave Share SpotPear 3.2inch RPi LCD V4"?
There are so many cheap Chinese clone on the market like this   
[image]  
[image]

and the shop site don't show a picture of the rear side.

Hope, you will now be able, to test all the needed stuff by yourself.
Have fun with your new  3.2inch  LCD display. 
Hopefully you don't have to pay anything more, no HMRC or handling fee.
If you will need it, I can send you the invoice incl. taxes and shipping.

Fourdee, 2016-07-14:
@k-plan 

> Sorry, yes - guilty.  :smile:

Cant thank you enough, legend, love it :)
I owe you one. I'l have to get you a bottle/beer one day :)

The plan is to add this into dietpi-config somewhere. So users can select it and have it automatically installed and configured.

I just need to resolve the X11 support, then try and get it working on the Odroids.

> It is the only shop I can find in GB, who has it in stock and was able to pay it from here and delivered to a different address in GB.

You did a great job tracking that one down, thanks again. And yes, no import tax, yay! :dancer: 

@k-plan 

> Is it a real "Wave Share SpotPear 3.2inch RPi LCD V4"?

Oh yes, its the real deal, surprisingly bright aswell. Looks awesome :D
https://goo.gl/photos/1XkezCDbpV9DgCNd7

> If you will need it, I can send you the invoice incl. declared taxes and shipping.

No need, but thanks for offering.

k-plan, 2016-07-14:
@Fourdee 

> And yes, no import tax, yay! :dancer:
Oh yes, its the real deal, surprisingly bright aswell. 

That's what I want to hear. Same device and nothing to pay for you. Very good. :+1:  :smiley: 

> Cant thank you enough, legend, love it :)
I owe you one.

No, no.
We have to thank you for maintain DietPi and implementing so many feature or supporting different hardware.

Thanks for your time, patience, staying power, nerves of steel and having a ear for everyone wishes.

Fourdee, 2016-07-14:
@k-plan 

Got X11 working on RPi fbdev, its cute! lol:
[image]
```
cat << _EOF_ > /etc/X11/xorg.conf.d/99-waveshare32_xorg.conf
Section "Device"
        Identifier      "FBDEV"
        Driver          "fbdev"
        Option          "fbdev" "/dev/fb1"

        Option          "SwapbuffersWait" "true"
EndSection

_EOF_
```

k-plan, 2016-07-14:
@Fourdee 

> Got X11 working on RPi fbdev, its cute! 

Nice! 😄 

Hmm, if you really want to run a desktop on 3.2 '' display something like ` matchbox-keyboard` and `xinput_calibrator` for using the stylus will be obligation.

[image]

I think, use it for DietPi Cloudshell on a 3.2'' display , its more than enough.

But yes, I see, its a nice new toy  and you have a lot of fun with it. 
I am pleased to see this.

Fourdee, 2016-07-15:
@k-plan 
Good news, got it working on C1. There is a C1 under there, promise:
```
modprobe spicc
modprobe fbtft_device name=odroidc_tft32 rotate=270 gpios=reset:116,dc:115 speed=32000000 cs=0
sleep 2 # wait for modules to become active. -w (wait) on modprobe, would be nice
con2fbmap 1 2
```
[image]

k-plan, 2016-07-15:
@Fourdee 

> ```
modprobe spicc
modprobe fbtft_device name=odroidc_tft32 rotate=270 gpios=reset:116,dc:115 speed=32000000 cs=0
sleep 2 # wait for modules to become active. -w (wait) on modprobe, would be nice
con2fbmap 1 2
#lets test it
echo -e "Hello there!" > /dev/tty1
```
😃 
On Odroid C1+ (promised)
![fullsizerender-4 2](https://cloud.githubusercontent.com/assets/17237601/16890378/20420c52-4aee-11e6-8de0-08fe8584fb77.jpg)

In some rare cases after `modprobe spicc` the prompt does not appears back.
After this, when now starting `dietpi-cloudshell` after some scense toggle 

![fullsizerender-3 2](https://cloud.githubusercontent.com/assets/17237601/16890444/096c2ec6-4aef-11e6-90e6-db747732e1fe.jpg)

Fourdee, 2016-07-16:
@k-plan 

Updated testing branch. On a fresh install, you can install the waveshare32 device with:
```
/DietPi/dietpi/func/dietpi-set_hardware lcdpanel waveshare32
reboot
```

I've tested RPi3 and C1 :)

And to remove it
```
/DietPi/dietpi/func/dietpi-set_hardware lcdpanel none
reboot
```

k-plan, 2016-07-16:
@Fourdee 

to reproduce this "phenomenon":

- power on your device
- login in via ssh to Dietpi and do
```
modprobe spicc
modprobe fbtft_device name=odroidc_tft32 rotate=270 gpios=reset:116,dc:115 speed=32000000 cs=0
sleep 2 # wait for modules to become active. -w (wait) on modprobe, would be nice
con2fbmap 1 2
```
-  start via ssh `dietpi-cloudshell`  on Main Screen (tty1).
- do a `reboot` or `poweroff`

seems to be a problem with first initialisation after a reboot or we had better to stop "fbdev" before rebooting. But don't know, how to do this.

Fourdee, 2016-07-16:
@k-plan 

> seems to be a problem with first initialisation after a reboot or we had better to stop "fbdev" before rebooting. But don't know, how to do this.

Shouldn't occur using this script, we write directly to modprobe.d and on Odroids, we map the console fb to display during boot.

k-plan, 2016-07-16:
@Fourdee 

Cool, running at boot-time. 💯 

Very nice and smooth installation on Odroid C1+ 

Kudos to Dan!

Fourdee, 2016-07-17:
@k-plan 

Waveshare32 (all RPI + Odroids) and Odroid-Cloudshell (XU4 only) are available in ```dietpi-config``` under Display options > LCD panel.

closed 2016-07-17 by Fourdee

WolfganP, 2016-09-18:
I found an old monochrome LCD 20x4 chars display in a drawer (Winstar WH2004A like https://www.adafruit.com/product/198), and connected successfully to my headless DietPi. 
I want to use it for system stats initially, but looking at the dietpi-cloudshell code and forum comments, I couldn't determine if it accepts simpler, character oriented displays like the one I use.
Any confirmation? If not, any idea on how to expand the display support for these kind of display?
Thx, Wolf

Fourdee, 2016-09-19:
@WolfganP 

DietPi-Cloudshell is coded to display on a 320x240 resolution screen, or greater.

The 20 char width and 4 lines would be a challenge to fit stats in there. Most of DietPi-Cloudshell would need to be rewritten to accommodate it.

As I dont own one of these 20x4 displays, implementing it would not be possible at this time. 
Also, as HAT's like the waveshare are possibly cheaper, for me, it doesn't seem logical to work on "outdated/replaced" tech.

WolfganP, 2016-09-19:
Thanks @Fourdee. Just for my reference, does Cloudshell relies heavily in the 320x240 resolution as it's managed as a graphical display, or once initialized (and set up the font size) you consider it to be a W x H char terminal (and just spit X chars' text lines)?

Fourdee, 2016-09-19:
@WolfganP 

Its coded to fit in a 320x240 resolution when using the font:
```
setfont /usr/share/consolefonts/Uni3-TerminusBold24x12.psf
```
couldn't find any bigger ones :)

Its basically a lightweight terminal program that prints to the terminal (tty1). We dont check the users current row/lines if thats what you mean?

WolfganP, 2016-09-19:
Thanks @Fourdee When I saw the screenshots of the waveshare32 display, all screens shown seems to be 25x8 chars long, so my thinking went to write a different display initialization routine, and then restrict cloudshell output to 20x4 display boundaries.
Do you think it may work?

saiakash, 2017-11-02:
[image]

saiakash, 2017-11-02:
how to hide or disable this screen when boot

Fourdee, 2017-11-13:
@saiakash 

> how to hide or disable this screen when boo

```dietpi-autostart``` > Console