MichaIng/DietPi #411: DietPi - Cloudshell v6 | Run + Skip intro doesn't work

k-plan, 2016-07-05:
@Fourdee 

> 	#////////////////////////////////////
	#
	# Info:
	# - System Stats for Cloudshell (or monitor/terminal)
	#
	# Usage:
	# dietpi-cloudshell				= Config Menu
	# dietpi-cloudshell 1			= Run
	# dietpi-cloudshell 2			= Run + Skip intro
	#////////////////////////////////////

`:~# dietpi-cloudshell 2` doesnt' work. (nothing happens)

Tested on Odroid C2 and RPi 3.

Fourdee, 2016-07-05:
@k-plan 
Good find :+1: 

I've updated the dietpi-cloudshell script with commit above to fix this.

k-plan, 2016-07-05:
@Fourdee 

wow, this fix was very quick! :smiley:  Thanks!

```
#Run DietPi-Cloudshell
elif (( $INPUT >= 1 )); then
```
edit, testing and it works on C2. :+1: