MichaIng/DietPi #414: New Odroid XU4 Debian Base Image with 4.6 kernel

rhkean, 2016-07-06:
http://linuxfactory.or.kr/dokuwiki/doku.php?id=yet_another_debian_ubuntu_release_for_odroid-xu4&&ckattempt=1

I just got word of this today and will be looking into it over the next week or so.  May steal the kernel... may rebuild the image... may do nothing....

but, I wanted to share.

Fourdee, 2016-07-06:
@rhkean 
Legend :+1: . Hopefully this will improve the 300mbit~ ethernet we have now ?

rhkean, 2016-07-06:
How do you test ethernet?  I'll check on my.  I'm running a 4.2 kernel at the moment.

Fourdee, 2016-07-06:
@rhkean 
```apt-get install iperf```

https://www.veritas.com/support/en_US/article.HOWTO64302

You'll need 2 machines. 1 acts as receiver, other acts as the sender of data.

rhkean, 2016-07-06:
what's the performance on a RPi-2?
or, I guess I can hard-wire my laptop....

Fourdee, 2016-07-06:
@rhkean 

> what's the performance on a RPi-2? 

All RPi's are 100mbit, so 10-12mb/s max.

C2 can sustain 880mbit, 110mb/s

rhkean, 2016-07-06:
> Pine A64+ is gigabit, i get around 550mbit: http://forum.pine64.org/showthread.php?tid=1068&pid=14466#pid14466

show off... I have to keep mine on a 100mbit switch to prevent it from hanging.  come to think of it... I'll have to do some perf tests with the 2nd one and see if its isolated to my first board.

> XU4 is gigabit capable, but i've only been able to achieve 280mbit 35mb/s
I've never tested the network on my XU4, because it streams high-def to 4 clients simultaneously without a stutter.  So, I just assumed it was fast enough... :)

Fourdee, 2016-07-07:
> show off... I have to keep mine on a 100mbit switch to prevent it from hanging. come to think of it... I'll have to do some perf tests with the 2nd one and see if its isolated to my first board.

Hehe, yep, no problems with the ones I've got.
it would be interesting if it is a hardware issue. Could explain why er0l is having throughput issues

rhkean, 2016-07-08:
well.... this will take me a few days... :cry: 

I was replacing the heatsink on my XU4 last night and knocked 2 capacitors off of the bottom of the board.  :sob: :sob: 

luckily, they're just part of the USB3_0 subsystem, which is the external USB3 ports.  I had to move my hard drives over to the USB2 bus until I can re-attach them.  :crying_cat_face:

Fourdee, 2016-07-08:
> I was replacing the heatsink on my XU4 last night and knocked 2 capacitors off of the bottom of the board.

Gutted :( easily done, especially with those tiny heatsink clips right next to components. I think I scratched a few on mine with the pliers lol.

Did you have any issues getting the heatsink unstuck? I had to get it upto 80'c before the sticky would detach without taking the ARM chip with it lol.

rhkean, 2016-07-08:
> Gutted :( easily done, especially with those tiny heatsink clips right next to components. I think I scratched a few on mine with the pliers lol.

fortunately, I have a good soldering iron.  :smile:   Took all my willpower not to try to fix it last night.  It was about midnight when it happened.... I would have destroyed the board for sure if I had tried to take a soldering iron to it that late at night.  :laughing: 

> Did you have any issues getting the heatsink unstuck? I had to get it upto 80'c before the sticky would detach without taking the ARM chip with it lol.

that was tricky, but I was expecting it.  So, just a little constant, prying pressure under the corner and it gave way.  I just kept adding pressure until I heard the sticky start to make that crinkling sound... then held it there until it let go.

rhkean, 2016-07-08:
What I need is a REALLY good Hi-res pic of the area to make sure that I put that tiny capacitor back where it belongs.

Fourdee, 2016-07-08:
@rhkean 

lmao.

My XU4 is doing backups tonight, but if you seriously want the image, let me know and i'll take it tomorrow.

rhkean, 2016-07-08:
I let you know after I take a look at it tonight under better lighting.  It might be a simple fix.

Also, as of today, I've got 4.6.0 and 4.7.0-rc6 kernel builds that boot and initialize the HDMI output!

rhkean, 2016-07-11:
gonna take me a bit longer on the USB3 testing... :sob: 

I knocked 2 capacitors and a ferrous blob (inductor) off the board... and loss the inductor.  

fortunately, it is only affecting the USB3 ports.  I moved the USB3 hub over to the USB2 port.... sigh...
gotta order a new board.

Fourdee, 2016-07-11:
>  sigh...
gotta order a new board.

Gutted. Not the cheapest SBC on the market to replace lol :(

rhkean, 2016-07-12:
So true...

rhkean, 2016-07-16:
> @Fourdee 
> Legend 👍 . Hopefully this will improve the 300mbit~ ethernet we have now ?

ran a couple tests tonight between 2 XU4 boards

client kernel |  server kernel | result
--- | ---- | ---
3.10.96 | 4.6.3 | 525 mbps
4.6.3 | 3.10.96 | 280 mbps
4.6.3 | 4.6.3 | 340 mbps
4.6.3 | 4.6.3 | 280 mbps
4.6.3 | 4.6.3 | 420 mbps
4.6.3 | 4.6.3 | 485 mbps

I'll try both boards against another system tomorrow

rhkean, 2016-07-16:
Just my luck...
My work laptop has a firewall installed and I don't have the rights to open the port for testing
And my wife's laptop's wired NIC is only 100mbit

Oddly, using my wife's wifi as the server and the new xu4 as client, it tested at only 330mbps and the old xu4 tested at 540mbps

rhkean, 2016-07-17:
final testing with iperf3
XU4 server, Ubuntu x86_64 client:  985mbps
XU4 client, Ubuntu x86_64 server:  starts out around 950mbps, but settles down around 340mbps

Fourdee, 2016-07-17:
@rhkean 
Thanks Rob,
I'll run the test on my XU4 + 1Gbit network today to compare. Any chance you can send the kernel/compile notes over?

rhkean, 2016-07-17:
will this do?

assumes running as root, builds in `/usr/src` directory (also patches in AUFS as a module)

save as:  `buildkernel.sh`
for a kernel with version `4.6.3-odroidxu4-1` from tobetter's repository:
`buildkernel.sh -f -r tobetter -b "odroidxu4-v4.6" -v 4.6.3 -x 1`
or 
`buildkernel.sh -h`

```
#!/bin/bash

bUseConfigGz=0
bFinalize=0

#REPOSITORY="tobetter"
#BRANCH="odroidxu4-v4.6"
#VERSION="4.6.3"

#REPOSITORY="ioft"
#BRANCH="new_master"
#VERSION="4.7.0-rc6"

CloneAufsAndPatch() {
        if [ ! -d "/usr/src/${LINUX}/fs/aufs" ]; then
                cd /usr/src
                echo "Purging old build tree for ${AUFS_BRANCH}"
                rm -rf "${AUFS_BRANCH}"

                echo "Cloning aufs.standalone from https://github.com/sfjro/aufs4-standalone -b ${AUFS_BRANCH}"
                if ! git clone --depth 1 https://github.com/sfjro/aufs4-standalone.git -b ${AUFS_BRANCH} ${AUFS_BRANCH}; then
                        echo "git clone aufs4-standalone FAILED"
                        exit 1
                fi

                cd "/usr/src/${LINUX}"

                echo "Applying aufs patches"
                echo "  applying aufs4-kbuild.patch"
                if ! patch -p1 < ../${AUFS_BRANCH}/aufs4-kbuild.patch; then
                        echo "patch aufs4-kbuild.patch FAILED"
                        exit 1
                fi

                echo "  applying aufs4-base.patch"
                if ! patch -p1 < ../${AUFS_BRANCH}/aufs4-base.patch; then
                        echo "patch aufs4-base.patch FAILED"
                        exit 1
                fi

                echo "  applying aufs4-mmap${RC}.patch"
                if ! patch -p1 < ../${AUFS_BRANCH}/aufs4-mmap.patch; then
                        echo "patch aufs4-mmap.patch FAILED"
                        exit 1
                fi

                echo "  copying aufs code to linux build tree"
                if ! cp -vR ../${AUFS_BRANCH}/{Documentation,fs} .; then
                        echo "copy ../${AUFS_BRANCH}{Documentation,fs} FAILED"
                        exit 1
                fi

                if ! cp -v ../${AUFS_BRANCH}/include/uapi/linux/aufs_type.h include/uapi/linux/ ; then
                        echo "copy aufs_type.h FAILED"
                        exit 1
                fi
        fi
}

[... 258 lines cut ...]
                AUFS_BRANCH="aufs4.x-rcN"
        fi

        BUILD_VERSION="${VERSION}-${REPOSITORY}${EXTRAVERSION}"
        if (( ${bFinalize} )); then
                BUILD_VERSION="${VERSION}-odroidxu4${EXTRAVERSION}"
        fi
        echo "Setting BUILD_VERSION: ${BUILD_VERSION}"
        LINUX="linux-${BUILD_VERSION}"
}

##################  MAIN #####################
cd "/usr/src"
ProcessOpts "$@"

PurgeAndCloneLinuxTree

CloneAufsAndPatch

ConfigureBuild

BuildLinuxKernel

InstallKernel
```

rhkean, 2016-07-17:
you'll need to edit the `boot.ini` file.  I suggest this, so you only have to change the version in the future:

```
setenv version "4.6.3-odroidxu4-1+"

setenv zimage "zImage-${version}"
setenv uinitrd "uInitrd-${version}"
setenv devicetree "odroidxu4-${version}.dtb"

# boot commands
setenv bootcmd "fatload mmc 0:1 0x40008000 ${zimage}; fatload mmc 0:1 0x42000000 ${uinitrd}; fatload mmc 0:1 0x44000000 ${devicetree}; bootz 0x40008000 0x42000000 0x44000000"
```

rhkean, 2016-07-17:
also:

`echo "11" > /etc/.dietpi_hw_model_identifier`

Fourdee, 2016-07-17:
@rhkean 
Thanks Rob :+1: 
Managed to get v123 out the door, but i'am out of time today. Will take a look start of next week :)

Fourdee, 2016-07-23:
notes:

XU4 (3.10.96+) <> C2
```
[  5]  0.0-30.0 sec  2.37 GBytes   679 Mbits/sec
[  4]  0.0-30.0 sec   535 MBytes   150 Mbits/sec
root@DietPi-XU4:~# uname -a
Linux DietPi-XU4 3.10.96+ #1 SMP PREEMPT Fri May 13 14:06:12 CEST 2016 armv7l GNU/Linux
```

Fourdee, 2016-07-23:
Had to clock her down a bit, she was a little "toasty" lol:
[image]

joaofl, 2016-08-02:
I would definitely appreciate it

rhkean, 2016-08-02:
> Had to clock her down a bit, she was a little "toasty" lol:
> make -j $(nproc --all) = 8

yup... at full speed (2GHz), I couldn't do more than -j=4 

not sure which would be faster... -j8 at 1GHz or -j4 at 2GHz

Fourdee, 2016-08-31:
I've compiled and hosted the XU4 4.7.0 Kernel

To install on DietPi:
```
#install
# - p7zip (allows for 40MB.7z, compared to 90MB.zip
apt-get install -y p7zip-full

# - Remove existing kernel packages
#	NB: When asked to abort removal, say <No>
apt-get purge -y --force-yes linux-image-armhf-odroid-xu3 linux-image-3.*
apt-get autoremove --purge -y

# - Download/install new kernel
wget http://dietpi.com/downloads/binaries/xu4/DietPi-XU4-4.7.0-28_10_2016.7z -O package.7z
7z x -aoa package.7z -o/
rm package.7z

# - Hardcode HW_MODEL
echo 11 > /etc/.dietpi_hw_model_identifier

reboot
```

joaofl, 2016-08-31:
Just tried it remotely, but it did not reboot. Maybe I need a hard reset?

A doubt is: when I run  "apt-get autoremove --purge -y" it removes the uboot. Is that supposed to happen? Or just by extracting the package I'm already replacing it?

I'm afraid I've killed it. Later today I'll unplug and plug it again to check it it's heart beats, and how to possibly bring it alive again.

Fourdee, 2016-08-31:
> Just tried it remotely, but it did not reboot. Maybe I need a hard reset?

@joaofl 

It should reboot fine.

> A doubt is: when I run "apt-get autoremove --purge -y" it removes the uboot. Is that supposed to happen? Or just by extracting the package I'm already replacing it?

> https://packages.debian.org/jessie/u-boot-tools

Its a tools package, so it shouldn't be required.

The extracted zip will replace the files and packages we removed eariler.

I'll look into it and see if i can replicate the failed reboot.

Fourdee, 2016-08-31:
@joaofl 

Yep, works fine for me. 
I ran same commands https://github.com/Fourdee/DietPi/issues/414#issuecomment-243736813. Boots to 4.7 straight after reboot.

If its not working after a hard power cycle. Try removing the USB drive, then power on again. Failing that, plug SD card into PC and send a screenshot of its contents please.

joaofl, 2016-08-31:
I will try that asap. Which content/logs would you go for?

joaofl, 2016-08-31:
@Fourdee
Follows the screenshot. It did not boot with or without the hd.
At a certain point it was replying pings, but not accepting ssh. Without the hd, none of both worked.

selection_001

joaofl, 2016-08-31:
Does that mean that kernel did not get replaced properly?

Fourdee, 2016-08-31:
@joaofl 

> Does that mean that kernel did not get replaced properly?

Possibly, was that screenshot taken on a Linux machine?

> At a certain point it was replying pings, but not accepting ssh. 

Any output/errors on HDMI?

joaofl, 2016-08-31:
Screenshot from a Linux Mint 18

joaofl, 2016-08-31:
Screenshot from a Linux Mint 18 

Here is the output from the hdmi. It fails with the dev-mmcbl1...
Maybe I had the file permissions messed up.

[image]

Fourdee, 2016-08-31:
@joaofl 

Return the XU4 to the screenshot state.
Type in your password (eg: dietpi), then run and take another shot:
```
journalctl -xb
```

Fourdee, 2016-08-31:
@joaofl 

Revert back to 3.x:
- Download http://dietpi.com/downloads/testing/XU4_3.x_joaofl.7z
- Extract it on your Linux Mint machine, anywhere.
- Plug in the XU4 SD card on your linux machine
- From the extracted files, copy ```boot``` folder to the SD cards 1st partition (fat).
- From the extracted files, copy ```lib``` folder to the SD cards 2nd partition (ext4).
If you are unsure, do NOT do any of the above, let me know and i'll try and explain it better :)

joaofl, 2016-08-31:
@Fourdee 

Is it an attempt to save my previous machine, or you want to test something?
Because if it is for restoring my stuff, no worries, I can start a new one with no harm (good chance to clean the junk).

If you want to test something, let me know and I follow the steps.

Fourdee, 2016-08-31:
@joaofl 

> Is it an attempt to save my previous machine, or you want to test something?

Bit of both. 
Be nice to find out if reverting back to 3.x resolves the issue. Could indicate an issue with SDcard support on 4.7.

Either way, fresh install never fails :+1: 

If you do get a chance:
Do a fresh install, run through the DietPi initial setup and install nothing extra, then try 4.7 again: https://github.com/Fourdee/DietPi/issues/414#issuecomment-243736813. But if not, no worries :+1:

Fourdee, 2017-03-13:
Update, Odroid are officially developing 4.9 kernel for XU4, still in testing:

> http://dietpi.com/phpbb/viewtopic.php?f=12&t=1393&p=5891&hilit=xu4+4.9#p5861
http://forum.odroid.com/viewtopic.php?f=146&t=26007

Fourdee, 2017-04-26:
https://github.com/Fourdee/DietPi/issues/899

closed 2017-04-26 by Fourdee