MichaIng/DietPi #416: DietPi-Software | SickRage

Fourdee, 2016-07-06:
> https://twitter.com/MattScherbatsky/status/750460181201481728

rhkean, 2016-07-06:
yeah, we can do that... It's written in Python and runs similarly to CouchPotato and HeadPhones.  I used to run it before I found Sonarr.

I'll work on these next week while the Mrs. is out of town.  I've already got a media-server-common that is the base to all 3.

Fourdee, 2016-07-06:
@rhkean 

> I'll work on these next week while the Mrs. is out of town. I've already got a media-server-common that is the base to all 3.

Excellent :+1:  I installed it just to take a look, havn't a clue how to use it lol

rhkean, 2016-07-06:
Yeah.  Installation is pretty straight forward.

Configuration, however will require a usenet search index server.  Most of those are not free.

I'll find one that's free to use for basic setup and then document the recommendation to get an account on a non-public index server

Fourdee, 2016-10-16:
Docs: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=3327#p3327

Fourdee, 2016-10-16:
Completed

closed 2016-10-16 by Fourdee