MichaIng/DietPi #420: DietPi Image | BananaPi Pro

Fourdee, 2016-07-11:
Many thanks to Mattia for making this possible :+1: . He sent a request to Lemaker for the sample board, and here it is: https://twitter.com/DietPi_/status/749932814439710720

k-plan, 2016-07-11:
Nice. 
Congratulations and have a lot of fun with the new device.

Fourdee, 2016-07-11:
@k-plan 
Still cost me £16 for import tax lol

I'am actually really impressed with the build quality of the board. I'd even go far as to say its up there with the RPi and Odroid's. Perfect PCB cut, soldering is spot on, none of that "leakage" on the PCB. High quality SBC that even has a SATA port, awesome :)

https://goo.gl/photos/FcyaoEgNewQvTLoN6

As for the OrangePi PC and a few others, well, lets not talk about that ;)

Fourdee, 2016-07-13:
Doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=534&p=2324#p2324

rhkean, 2016-07-13:
> @Fourdee 
> Still cost me £16 for import tax lol

I think that's crazy -- about a 50% import tax -- , and I wonder how much will it cost after all of the "Brexit" dust settles?

Fourdee, 2016-07-13:
@rhkean 

£8 goes to HMRC (import tax)
DHL/Fedex all charge £8-9 "handling fee"

Brexit, whats this all about :P. Hopefully it cant get any worse lol

Fourdee, 2016-07-13:
Issues with wifi:
http://forum.armbian.com/index.php/topic/1608-bpi-pro-wifi-seems-active-but-fails/#entry12426

k-plan, 2016-07-14:
> £8 goes to HMRC (import tax minimal fee)
DHL/Fedex all charge £8-9 "handling fee"

This problem is caused by dispatcher or lemaker. If device is correct declared with a [CN22](http://www.royalmail.com/sites/default/files/CN22_01.pdf) as "commercial sample" and have a pro forma invoice on the outside (	value of goods < 50 €  /	invoice total = 0 € ) and  the 	hint on the device "SAMPLE - NOT FOR SALE" you don't have to pay any taxes. And if so, DHL/Fedex/RoyalMail can't  impose any "handling fee".

https://www.gov.uk/government/publications/notice-372-importing-commercial-samples-free-of-duty-and-vat/notice-372-importing-commercial-samples-free-of-duty-and-vat

Has nothing to do with Brexit ....  :smile:

Fourdee, 2016-07-14:
@k-plan 

> If device is correct declared with a CN22 as "commercial sample" and have a pro forma invoice on the outside ( value of goods < 50 € / invoice total = 0 € ) and the hint on the device "SAMPLE - NOT [...]

You are a legend :+1: . I'll try to request this for next time and see what happens.

Fourdee, 2016-07-14:
SATA confirmed working. 

Although we still have outstanding issues with onboard WiFi.  I'll mark this as closed for now and create a separate git ticket for the WiFi.

closed 2016-07-14 by Fourdee