MichaIng/DietPi #422: DietPi-Software | Syncthing

Fourdee, 2016-07-11:
> http://dietpi.com/phpbb/viewtopic.php?f=12&t=531#p2305

Prebuilt deb packages: http://apt.syncthing.net/

joaofl, 2016-07-18:
/etc/init.d/syncthing script for Raspberry Pi (or any Ubuntu/Debian)

https://gist.github.com/arudmin/5a13e9105814c3f568ec

Fourdee, 2016-07-18:
@joaofl 
Bumped to v124. I keep getting asked for this one, so lets get it in :+1:

Fourdee, 2016-07-19:
notes:
:u55b6: ARMv6
:u6307: ARMv7
:u55b6: ARM64
:u55b6: x86_64

Fourdee, 2016-07-19:
This program is awesome, although, a little reading is needed to understand how to set it up.
[image]

joaofl, 2016-07-19:
More or less I would say. The basics are almost plug and play. You only
need the others device key to pair the two clients. Then, adding folders
and sharing them comes as an option at each device you add.

If I can help on something, let me know.

Fourdee, 2016-07-19:
@joaofl 
With our install, i've allowed access over LAN.
```
sed -i '/:8384<\/address>/c\        <address>0.0.0.0:8384<\/address>' "$HOME"/.config/syncthing/config.xml
```
How do you run your setup?

I created a systemd service for this also:
```
cat << _EOF_ > /etc/systemd/system/syncthing.service
[Unit]
Description=Syncthing

[Service]
Type=oneshot
ExecStart=/bin/bash -c '/usr/bin/syncthing > /var/log/syncthing.log &'
ExecStop=/usr/bin/killall -w syncthing
RemainAfterExit=yes
User=root

[Install]
WantedBy=multi-user.target
_EOF_
systemctl enable syncthing.service
systemctl daemon-reload
```

joaofl, 2016-07-19:
That is good. I would also enable enable https for the web gui, and disable the "open browser" option, otherwise it will get launched at one's desktop, if they have it.

Want me to send you a config file example?

Fourdee, 2016-07-19:
@joaofl 
> Want me to send you a config file example?

Please if you can. Remove any passwords/auth keys etc before posting.

Fourdee, 2016-07-19:
Docs: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=2363#p2363

joaofl, 2016-07-19:
@Fourdee 

Follows an example of config file.

It enables secure remote access to the Web-GUI, which is very important for headless setups.
 
[config.xml.zip](https://github.com/Fourdee/DietPi/files/371742/config.xml.zip)

Fourdee, 2016-07-19:
@joaofl 
thanks, pretty much identical. I've added ```<startBrowser>false</startBrowser>``` config.xml.

Note to self: arm64 install is not applying GUI user/password. Danger prompt remains. armv6+ fine.

joaofl, 2016-07-19:
@Fourdee 
It is identical, but to every major release, back compatibility is broken for other reasons, and for that they keep the config file version at the header: <configuration version="15">
And this would make it incompatible with 0.13.xx. But anyway, dietpi will start from 0.14, so it doesn't matter, but could create a bug, since you are working on it right now that they did this release.

joaofl, 2016-07-19:
@Fourdee 
Forgot to answer your question: "How do you run your setup? eg: Do you use web interface over LAN, external access through port forwarding?"

Yes. I use no-ip + port forwarding

Fourdee, 2016-07-19:
@joaofl 

> And this would make it incompatible with 0.13.xx. But anyway, dietpi will start from 0.14, so it doesn't matter, but could create a bug, since you are working on it right now that they did this release.

During installation, we get syncthing to generate the config. Then we only change specific values in the config.xml. So in theory, unless those variable names change in config.xml, the version should have no effect on the installation.

Fourdee, 2016-07-20:
Completed.

closed 2016-07-20 by Fourdee

joaofl, 2017-04-06:
@Fourdee 

I was thinking these days about moving my syncthing config folder to the dietpi_userdata folder. This because, in case i have that folder on a pen drive over USB3, I can get much faster indexing, db r/w operations, etc...

In this sense, what do you think of having the config folder inside dietpi_userdata? I saw that it can become a bit tricky because of the inotify and this API key, but I could give it a try.

Having all my config files inside the same folder interests me, specially for backup proposes.

Fourdee, 2017-04-06:
@joaofl 

> This because, in case i have that folder on a pen drive over USB3, I can get much faster indexing, db r/w operations, etc...
In this sense, what do you think of having the config folder inside dietpi_userdata? 

Yep, makes sense 👍 . If the DB is movable, DietPi user data directory should be the target.
We did this for MySQL recently, would be great if we can follow this up with other software.

joaofl, 2017-08-09:
Finally did it: https://github.com/Fourdee/DietPi/pull/1103

Tested and working. Don't know what do you think of running the service as dietpi user.

joaofl, 2018-11-25:
> Note to self: arm64 install is not applying GUI user/password.
@Fourdee 

After long time, while setting up my new cloud, I decided to give this issue a try. I already managed to get the password generated and hashed according to the algorithm they use. Now its a matter of inserting this line inside the setup. However, I'm not really sure about the exact right approach.

In the following 2 lines, I install a python dependency and run the python script that generates and print hashed password, which should be added to the config.xml file, inside the <gui> scope:
```
            # - Install requisites in order to generate the bcrypted default password for the webUI
            apt install -y python3-bcrypt
            python3 -c 'import bcrypt as b; print(b.hashpw("PASS_HERE".encode(), b.gensalt()).decode()); '
```
Then have something like
```
    <gui enabled="true" tls="true" debugging="false">
        <address>0.0.0.0:8384</address>
        <apikey>whatever</apikey>
        <user>dietpi</user>
        <password>$2b$hashed_pass_here</password>
        <theme>default</theme>
    </gui>

```

Now it is a meter of to putting the 'apt install' at the right place, and to get the output from the python script into the config file, along with the user name. We should also use the '$GLOBAL_PW' var somehow inside the python script.

Can you give me some directions on the best practices? Think I can implement it.

MichaIng, 2018-11-25:
@joaofl 
I hope there is a way without python-bcrypt? Since Syncthing does not rely on it, it should be. I am not keen to install whole python dependency chain just for this 😉.

@Fourdee 
> I'll also disable automatic upgrades aswell in the config.

We should re-enable, generally. Rely on internal updater handle/updating existing config correctly. If users update via dietpi-software reinstall, this might break thinks instead, since internal config/database update steps are missing.

This is something I would do in general:
- If software has an internal updater, enable it in every case, have our install code allowing this in every case. Only internal updates can respect existing install best, where manual upgrades in case always bear the risk to break existing install.
- In case of internal updater make dietpi-software skip the download & install step, only do config injection, systemd unit creation, user/permissions setup and such.
- Rely on future versions still working with our installer/uninstaller. We have not the manpower to go though all those software titles every DietPi release to bring them up-to-date. Check: https://github.com/Fourdee/DietPi/issues/1340
Most of them are out of date again, and I didn't check for a longer while... Switch to a download link that automatically points to latest release (api.github.com) where possible.

MichaIng, 2018-11-25:
@joaofl 
Ah, so this option is only for really applying the update, while update checks are anyway done regularly?

Jep then I agree to leave it disabled.