MichaIng/DietPi #424: DietPi utility aliases fail to work with sudo

Stickerpants, 2016-07-12:
This is mostly for awareness, since it's a pretty minor detail. As explained [in this post](http://askubuntu.com/a/22043), it seems like aliases fail to expand when they are not the first word in a call.

As a result, attempting to call `sudo dietpi-config` (like it instructs you to, if not run as root) gives the error `sudo: dietpi-config: command not found`. Of course, we can run the scripts direct by specifying the path to the ramdisk directly via `/DietPi/dietpi/dietpi-config`.

There's a few options to this end - what if we put `/DietPi/dietpi/` on the path? Then there's no need to deal with alias expansion.

I do recognize that this is a very minor issue - maybe the best solution here is just to strip out the messages that tell people to use `sudo` to run the scripts, and instead replace it with a message telling people to `su` instead.

Fourdee, 2016-07-12:
@Stickerpants 

> what if we put /DietPi/dietpi/ on the path? Then there's no need to deal with alias expansion.

Yep, using the full path is one resolution, although, we shouldn't rely on the the user to type it, every time they want to run a DietPi script. Its a bit long :)

> maybe the best solution here is just to strip out the messages that tell people to use sudo to run the scripts, and instead replace it with a message telling people to su instead.

@Stickerpants 
Like this?
```
sudo su
dietpi-runthis
```

Fourdee, 2016-07-12:
@Stickerpants 

Read the link, yep, so if we add this to ```/etc/bash.bashrc```, the space after sudo allows for other alias to be triggered.
```
alias sudo='sudo '
```

```sudo dietpi-config``` now works.

Stickerpants, 2016-07-12:
Yup, I do agree that aliasing `sudo` is likely the cleanest way to address this. Thank you @Fourdee!

rhkean, 2016-07-12:
I think a cleaner way might be too add `/DietPi/dietpi` to the PATH search.

This can be done in `/etc/profile` as:
`PATH="$PATH:/DietPi/DietPi"`

Just before the `export PATH` statement

Fourdee, 2016-07-13:
@rhkean 

> Then we don't need to alias any of them

Great idea :+1:.

Gutted, sudo doesn't appear to work with defined PATH:
```
root@DietPi:~# echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/DietPi/dietpi

root@DietPi:~# sudo dietpi-cpuinfo
sudo: dietpi-cpuinfo: command not found

#non-sudo works fine
root@DietPi:~# dietpi-
dietpi-apt-get_update        dietpi-letsencrypt
dietpi-autostart             dietpi-logclear
dietpi-backup                dietpi-morsecode
dietpi-banner                dietpi-obtain_hw_model
dietpi-bugreport             dietpi-process_tool
dietpi-cleaner               dietpi-ramdisk
dietpi-cloudshell            dietpi-ramlog
dietpi-config                dietpi-services
dietpi-cpuinfo               dietpi-software
dietpi-cpu_set               dietpi-survey
dietpi-cron                  dietpi-sync
dietpi-external_drive_setup  dietpi-uninstall
dietpi-funtime               dietpi-update
dietpi-launcher
```

rhkean, 2016-07-13:
> @Fourdee 
> Gutted, sudo doesn't appear to work with defined PATH:

it has to be in root's path.

> Alias also allows us to essentially control/limit what dietpi- scripts are available for the user.

yeah.... good point.... you don't want an end-user running `finalise` like I just did.  :laughing:

rhkean, 2016-07-13:
This stackoverflow discussion appears to have the sudo PATH answers, but I won't have time to read it until this evening.

http://stackoverflow.com/questions/257616/sudo-changes-path-why

Fourdee, 2016-07-13:
> yeah.... good point.... you don't want an end-user running finalise like I just did

lol.

rhkean, 2016-07-13:
my proposed fix is to 

1.  add `/DietPi/dietpi` to the secure_path in the `sudoers` file 
2.  add `/DietPi/dietpi` to the PATH in `/etc/profile`
3.  in `dietpi-services` when the `dietpi-*` files are copied to `/DietPi/dietpi` directory, perform a `chmod 744 /DietPi/dietpi/dietpi-*`  (or some variant that makes sense for limiting which scripts are executable)

this will 
1.  allow for `sudo` use
2.  prevent scripts that we don't want run from getting run accidentally
3.  let us avoid using hacks to circumvent the security feature

Fourdee, 2016-07-14:
1. Yep, maybe we can pop something in ```/etc/sudoers.d/dietpi```. I tried ```Defaults secure_path="/DietPi/dietpi"``` has no effect.

2. Yep.I think we should again put something in .d ```/etc/profile.d/dietpi```

3. I  can do this with DietPi-Ramdisk, create a list of scrips that we can ```chmod -x``` during boot.

Fourdee, 2016-07-14:
Auto completion does not work with /etc/sudoers. eg: ```sudo dietpi-```

Fourdee, 2016-07-14:
@rhkean 
Above commit is not final for sudo fix. I put the it in a few days ago and we can change as needed.

rhkean, 2016-07-14:
we'll have to actually edit the `/etc/sudoers` file as such:

```
Defaults        secure_path="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/DietPi/dietpi"
```

might need to wait for the next full image.... that file doesn't like to be edited.

Fourdee, 2016-07-14:
@rhkean 
> might need to wait for the next full image.... that file doesn't like to be edited.

Yep, so i've seen ```visudo```. I'll take a look.

Fourdee, 2016-07-14:
> # This file MUST be edited with the 'visudo' command as root.

This is non-sense. The file can be written to directly without ```visudo```. Even works with ```sudo sed```:
```
if (( ! $(cat /etc/sudoers | grep 'secure_path=' | grep -ci -m1 '/DietPi/dietpi') )); then
	sed -i 's/:\/bin\"/:\/bin:\/DietPi\/dietpi\"/' /etc/sudoers
fi
```

Stickerpants, 2016-07-14:
The reason why it says "must" is mostly a "don't shoot yourself in the foot" scenario - `visudo` syntax checks `/etc/sudoers` to ensure that everything is correct before allowing you to save and exit.

This is to prevent you from locking yourself out by breaking `sudo` - imagine if you did `sudo vim /etc/sudoers`, and then broke it without having direct access to `root`.

Fourdee, 2016-07-14:
@Stickerpants 

> The reason why it says "must" is mostly a "don't shoot yourself in the foot" scenario
This is to prevent you from locking yourself out by breaking sudo

Aha, now it makes sense. A end-user countermeasure to prevent breaking Sudo. Thanks for the info :+1: 

I'll make sure to test the pattern match patch for ```/etc/sudoers``` on all devices prior to release.

rhkean, 2016-07-14:
> @Fourdee 
> Even works with sudo sed

I'd be interested to know how you got sed to edit a file with `-r--r----- ` permissions.   :laughing:

Fourdee, 2016-07-15:
> I'd be interested to know how you got sed to edit a file with -r--r----- permissions.  :laughing:

Yep, I ran sudo as root :wink: 

```
testuser@DietPi:~$ sudo sed -i 's/:\/bin\"/:\/bin:\/DietPi\/dietpi\"/' /etc/sudoers
[sudo] password for testuser:
testuser is not in the sudoers file.  This incident will be reported.
```

Fourdee, 2016-07-15:
Ok, so there is a risk with users running ```sudo dietpi-update``` on this.
I'll revert the changes back to alias for now. 

I'll also move this ticket to another milestone (https://github.com/Fourdee/DietPi/milestone/25).

Fourdee, 2016-10-29:
This appears to be working fine currently with:
```
sudo dietpi-config
sudo dietpi-update
```

/etc/bash.bashrc
```alias sudo='sudo '```

Marking as closed.

closed 2016-10-29 by Fourdee