MichaIng/DietPi #425: DietPi-Software | OpenVPN Client with killswitch

Othou, 2016-07-12:
Anyway to get OpenVPN client in DietPi? So you could just throw in .ovpn file in and the text file with login information and it would work? And trustable killswitch would be nice! Mostly looking for running for btsync server in a place where p2p networking is forbidden. You would be awesome if you would make that work with killswitch! I would donate you some money right away if ya make that work :P And i know its possible to do yourself but i just cant finish any of those projects, i blame it on my ADD, baby. So looking for a easy-to-setup thingy here :)

Zone22, 2016-07-13:
Are you the owner of the network and are you only using it for legal p2p content?

Othou, 2016-07-13:
Well the thing is that we have been using btsync to transfer some video project files and now i'm moving to student dormitory where p2p is blocked. Not that im transfering stuff in there all the time, nope. Just every now and then. Asked for a killswitch so i would not get banned or anything. TL;DR: No. Yes.

Zone22, 2016-07-22:
What do you mean by kill switch and how would you wanted to function?

It's very unlikely that the team DietPi would make something so custom although there might be methods which you are not aware of to do exactly with what you want using the SSH terminal.

oshank, 2016-09-21:
I have exactly what you want. I was in a similar situation as you, followed this tutorial, https://gist.github.com/superjamie/ac55b6d2c080582a3e64. Made a few scripts to flush iptables and make sure that the VPN is always connected. Check out the comment towards the bottom with the script to flush and set the iptables.

Zone22, 2016-09-21:
That's an interesting tutorial. Although I would keep in mind it will slow your connection to the Internet and your internal network down due to the 100 Mb ethernet port on the raspberry pie.

oshank, 2016-09-21:
This is true, I do not get great speed. Close to 2 or 3 MB down with transmission.

closed 2017-12-30 by Fourdee